ha7ilm/openwebrx #1: Consider Affero GPL as a license

dnet, 2015-05-03:
While "plain" GPL requires changes to the source to be available if someone distributes the software as a binary (which would be weird anyway with a Python project like this), it does allow people to change the software and run it as a service but not share these modifications. [Affero GPL](https://www.gnu.org/licenses/agpl.html) [fixes this](https://www.gnu.org/licenses/why-affero-gpl.html), and should match a project like this more closely, like [searx](https://github.com/asciimoo/searx).

ha7ilm, 2015-05-04:
Hi Dnet,

You're quite right, I've already been thinking about moving to AGPL.

irl, 2015-05-08:
I would consider Affero GPL non-free. There may be changes that people would like to make that couldn't be redistributed and this the Affero license would thus prevent those changes being made. To ensure the software is as useful as possible for as many people as possible, I would recommend against changing the license. If you were to change it, I would prefer a BSD 2- or 3-clause license instead.

stef, 2015-05-08:
I think others would disagree with you on whether Affero GNU GPL is non-free. The thing about the Affero GNU GPL is that it keeps the software free at the expense of the right of 3rd parties closing up their contributions. What good really is a closed contribution anyway? That is one of the few ways we seem to be able to ensure the freedom of the software. It is this "ancient" dilemma, do we trust users more or software? Which freedom is more important? In some important infrastructural use-cases - like an IP stack, it might make sense to go PD or permissive licensing to achieve an industry wide adoption. However this piece in particular I think is an excellent example of something that the Affero GNU GPL is made for and which should be kept as free as possible. Also there's enough closed websdrs already.

irl, 2015-05-08:
I'm mainly just against the restricting of freedoms. If others choose to restrict freedoms then that's them doing bad and their contributions will happen and then be surpassed by the free software anyway. You're welcome to your opinions but as a Debian Maintainer I've seen on too many occasions license incompatibilities and overly restrictive licenses trying to act with good intentions but inadvertently causing problems for some use cases. GNU GPL is a tried and tested license. Honestly though, this isn't something I want to spend months arguing over, so this will be my last reply on this issue. I've made my point and it's up to the developer(s) at the end of the day to decide what license they want their work to be under.

stef, 2015-05-08:
you restrict freedoms either way.

stef, 2015-05-08:
the thing is, that the GNU GPL works when you distribute the sw to your users, which is for binary stuff, when however you provide the sw over the web, then there is no distribution that would kick in to ensure the users indeed have their freedom to access the source code honored. this web-loophole is exactly what the Affero GNU GPL was made for, apps that are served over the web, for which open*web*rx is a good example.

closed 2016-09-07 by dnet