ha7ilm/openwebrx #3: Consistently 10kHz off when tuning

irl, 2015-05-08:
I've noticed that when tuning, the reported value is 10kHz lower than the value I'm transmitting on with the radio.

My config is:

#DSP/RX settings
dsp_plugin="csdr"
fft_fps=9
fft_size=8192
samp_rate = 2000000
center_freq = 145000000
rf_gain = 5

Thanks,

ha7ilm, 2015-05-11:
Most likely this issue is caused by the inaccuracy of the crystal oscillator on the RTL-SDR.
It can be calibrated (for example, with your own reference signal source, or the kalibrate-rtl utility).
If you know the PPM value, you can add it to config_webrx.py:

    start_rtl_command="rtl_sdr -p PPM_value (...)

However, this inaccuracy may be dependent on actual temperature, that's why people are modding RTL-SDR sticks with TCXO-s.

How to calibrate with kalibrate-rtl:

    ~/Temp/kalibrate-rtl $ kal -s GSM900 -e 0
    Found 1 device(s):
      0:  Generic RTL2832U OEM
    
    Using device 0: Generic RTL2832U OEM
    Found Rafael Micro R820T tuner
    Exact sample rate is: 270833.002142 Hz
    kal: Scanning for GSM-900 base stations.
    GSM-900:
    	chan: 11 (937.2MHz - 8.013kHz)	power: 208040.92

8013 / 937200000 * 1000000 = 8.55 
round(8.55)=9

Then you write:

    start_rtl_command="rtl_sdr -p 9 (...)

ha7ilm, 2015-08-18:
Now we have a `ppm` option in `config_webrx.py`, so based on the previous example, you could write:

    ppm = 9

nickoe, 2016-04-30:
@ha7ilm I think this issue can also be closed, since this is not strictly a problem with openwebrx and that you can now use the ppm option.

ha7ilm, 2016-04-30:
Yeah. I've started to write an article on the wiki about this problem:
https://github.com/simonyiszk/openwebrx/wiki/Correcting-the-frequency-offset

I'll close the issue when I'll have finished.