ha7ilm/openwebrx #4: Please provide a release

irl, 2015-05-08:
I'm looking at packaging your software for Debian. We like to base our packages on releases of the software and then track new releases automatically so we know when we need to update our packages.

Could you please tag a release or make a release through the GitHub releases system when you feel your software has reached a stable state?

I'm happy enough with the state of the software currently to make a package based on the latest Git commit but I would prefer to have a release before uploading this to the Debian archives.

(I'm also looking at packaging csdr as a dependency of this.)

ha7ilm, 2015-05-10:
Hi Iain,

I am very pleased that you find OpenWebRX valuable enough to package for Debian!
Although some time ago I had a release tagged with "v0.1b", I would like to add some bugfixes before it goes to the Debian repos. I plan to add these fixes this week, then I will make a new release.

irl, 2015-05-11:
I actually tried to do this myself, but never had the time (https://github.com/irl/websdr-plus).

I'd tried to get the code from the WebSDR guy but he wouldn't even give me the binaries to set up a site.

I'm running an instance at http://webrx.radio.57north.org.uk:8073/ and I've made the local amateurs aware of this, so they're all getting some testing done. The frequencies there do all still seem to be about 10kHz off though, as I reported in #3.

ha7ilm, 2015-05-11:
> I'm running an instance at http://webrx.radio.57north.org.uk:8073/ and I've made the local amateurs aware of this, so they're all getting some testing done.

Thanks for sharing the receiver, and doing testing!

jks-prv, 2015-05-24:
Hello everyone,

Please excuse the thread hijack, but since you are all OpenWebRX users I thought I'd give you the first look at the OpenWebRX-based WRX live prototype receiver at http://www.jks.com:8073  This is a web-enabled SDR+GPS board for the BeagleBone Black. 100% open-source / open-hardware of course.

This is the very first time I'm making it publicly available so there may be significant problems, although it works for me with Mac versions of Safari, Firefox and Chrome. I cannot thank Andr�s enough for letting me know about OpenWebRX back in December. It has made a HUGE difference in the progress of my user interface work.

For those interested I have written a design review document http://www.jks.com/docs/wrx/wrx.design.review.pdf detailing the hardware and software. I would appreciate feedback of any kind before I spend the money to have a PCB fabricated. I am considering a Kickstarter campaign to get some boards built.

history: www.jks.com/wrx
github: https://github.com/jks-prv/Beagle_SDR_GPS

Kind regards,

John Seamons, ZL/KF6VO
jks@jks.com
Tauranga, New Zealand

jks-prv, 2015-05-24:
Sorry everyone. The password is 'kiwi'.

ha7ilm, 2015-05-26:
Hi All,

> Could you please tag a release or make a release through the GitHub releases system when you feel your software has reached a stable state?

@irl: I made the release! Please let me know if I can help you anything while making the Debian package. 

> I thought I'd give you the first look at the OpenWebRX-based WRX live prototype receiver

@jks-prv: That's cool! Thanks for sharing, and for making the hardware open as well!

closed 2015-05-26 by ha7ilm