ha7ilm/openwebrx #6: Cannot Email Developer

BT-Justice, 2015-06-04:
I have been trying to send an email to randras@sdr.hu but I keep getting the email returned as...

SMTP error from remote server after RCPT command:
host: mail.sdr.hu
Administrative prohibition

Please fix this.  Get an free Outlook.com email account or something please.

ha7ilm, 2015-06-04:
Hi BT-Justice, 

Thanks for reporting that, I'll look after it!
(Yesterday I've been happiply using that address, maybe something happened at my provider...)

Until that, you can contact me at:

retzlerandras@gmail.com

Best regards,

Andras

closed 2015-06-04 by ha7ilm

BT-Justice, 2015-06-04:
Thanks.  Just sent you an email.

ha7ilm, 2015-06-04:
I've tested my e-mail address (randras@sdr.hu), and it should work now!

BT-Justice, 2015-06-05:
"randras@sdr.hu":
SMTP error from remote server after RCPT command:
host: mail.sdr.hu
Administrative prohibition

ha7ilm, 2015-06-05:
Interesting. I managed to receive mail from both outlook.com and Gmail accounts yesterday, and there was no such error. 

As a workaround, can still mail me at the other address.

BT-Justice, 2015-06-05:
I use GMX.com for my email so maybe your server blocks it for some reason.