ha7ilm/openwebrx #9: getting audio underruns

ha7ilm, 2015-08-20:
Hans, DL9RDZ has reported that they were getting audio underruns running OpenWebRX on a Raspberry Pi until they removed the `csdr flowcontrol` from the DSP chain <a href="https://github.com/simonyiszk/openwebrx/blob/master/plugins/dsp/csdr/plugin.py#L53">at this line</a>. 
 If someone has a similar problem, it is worth trying. I'll try to investigate.
(In fact, I'm getting audio underruns if I do that change. )

kaback, 2015-10-11:
Hi, i do see the exact same problem. Rpi2, debian-arm. After i did this change, there are no underruns anymore. But Audio Output and Network Usage is a bit higher now.

ha7ilm, 2015-10-12:
Hi Karsten,

You could also try the current "dev" branches of OpenWebRX and csdr.
This doesn't have the flowcontrol, and also has some NEON optimizations in inline assembly.

anitracks, 2015-12-07:
I still have significant audio underruns from the master branch on a regular Ubuntu server. The dev branch failed to run properly on my server with the following output:
```
OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] Started rtl_thread: rtl_sdr -s 250000 -f 28120000 -p 0 -g 5 - | nc -vvl 127.0.0.1 8888
sh: 1: ncat: not found
[openwebrx-main] Started rtl_mus.
Listening on [127.0.0.1] (family 0, port 8888)
Found 1 device(s):

rtl_mus: Multi-User I/Q Data Server for RTL-SDR v0.22, made at HA5KFU Amateur Radio Club (http://ha5kfu.hu)
    code by Andras Retzler, HA7ILM <randras@sdr.hu>
    distributed under GNU GPL v3

2015-12-07 06:01:57,435 INFO Server is UP
Connection from [127.0.0.1] port 8888 [tcp/*] accepted (family 2, sport 45553)
2015-12-07 06:01:57,440 INFO Server listening on port: 4951
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-main] Starting HTTP server.
Sampling at 250000 S/s.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = sleep 1; nc -v localhost 4951 | csdr setbuf 11264 | csdr through | csdr convert_u8_f | csdr fft_cc 4096 27777 | csdr logpower_cf -70 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
Tuned to 28120000 Hz.
Tuner gain set to 3.70 dB.
Reading samples in async mode...
2015-12-07 06:01:59,198 INFO rtl_tcp host connection estabilished
function name given in argument 1 does not exist.

[openwebrx-spectrum] Spectrum thread started.
function name given in argument 1 does not exist.

nc: connect to localhost port 4951 (tcp) failed: Connection refused
2015-12-07 06:01:59,433 INFO client accepted: 0@127.0.0.1:50330  users now: 1
Connection to localhost 4951 port [tcp/*] succeeded!
2015-12-07 06:01:59,828 INFO client disconnected: 0@127.0.0.1
[openwebrx-check_server] >>>>>>> ERROR: spectrum_thread dsp subprocess failed
10.0.2.2 - - [07/Dec/2015 06:02:14] "GET / HTTP/1.1" 500 -
10.0.2.2 - - [07/Dec/2015 06:02:14] "GET /favicon.ico HTTP/1.1" 200 -
^C2015-12-07 06:02:26,072 INFO Ctrl+C: aborting.
Signal caught, exiting!
[openwebrx] Ctrl+C: aborting.
Traceback (most recent call last):
  File "./openwebrx.py", line 579, in <module>
    main()
  File "./openwebrx.py", line 170, in main
    httpd.serve_forever()
  File "/usr/lib/python2.7/SocketServer.py", line 236, in serve_forever
    poll_interval)
  File "/usr/lib/python2.7/SocketServer.py", line 155, in _eintr_retry
    return func(*args)
  File "./openwebrx.py", line 80, in handle_signal
    spectrum_dsp.stop()
  File "/home/drseth/projects/openwebrx/plugins/dsp/csdr/plugin.py", line 186, in stop
    os.killpg(os.getpgid(self.process.pid), signal.SIGTERM) 
OSError: [Errno 3] No such process

User cancel, exiting...
```

ha7ilm, 2015-12-07:
Hi,

I'm really sorry that I can't fix this in the master branch right now.
The dev branch should work at all.

    function name given in argument 1 does not exist.

This can be solved by updating `csdr` to the dev branch, and recompiling it.

    git clone https://github.com/simonyiszk/csdr/
    cd csdr
    git fetch
    git checkout dev
    make
    sudo make install

Gwendir, 2016-06-14:
Hi all, 

I am getting severe audio underruns on my Raspberry Pi 3. (4 users - fft_fps=9 - size=4096 - sample rate = 250000) 
Tried the tip Hans DL9RDZ gave - first post - and fiddled with many settings but unfortunately no succes, probably due to the fact that the plugin.py file has changed over time.
The link he gave does not point to the csdr flowcontrol. A bit further in the code csdr flowcontrol is commented out by using ## 

I know it can work ;-) >  http://hamsatsdr.ddns.net:8073/  
Any ideas?