ha7ilm/openwebrx #10: Is this still being maintained?

Toontje, 2015-09-22:
I think it's a cool project.

ha7ilm, 2015-09-22:
Hi Toontje,

Thanks! It is actively maintained.

Maybe it doesn't seem so as I'm usually pushing many changes at once, after a lot of testing.
In August I launched http://sdr.hu/ for listing the online receivers, and also added ADPCM compression to OpenWebRX. Right now I'm working on speeding it up on the Raspberry Pi 2, and fixing a lot of other things. I've also created a Wiki for the project on GitHub.

I'm still open to help anyone who has problems setting up a public receiver.

Toontje, 2015-09-22:
Ok, good!

I have two issues.

- When running as non-root i get the following error:
```[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin

[openwebrx-main] Started rtl thread: rtl_sdr -s 250000 -f 820000000 -p 0 -
| nc -vvl 127.0.0.1 8888

[openwebrx-main] Started rtl_mus.
```
- How does an end-user change the centre frequency? I can change the LO,
but only within the limit of the frequencies currently in display.

Ton.

ha7ilm, 2015-09-22:
> When running as non-root i get the following error

If you run it as root, everything works well?
Maybe there is a problem with `librtlsdr` USB privileges. 
It especially applies if you get an error when you try to run the following without root privileges:

    rtl_sdr - > /dev/null

You should recompile `rtl_sdr` with:

    cmake ../ -DINSTALL_UDEV_RULES=ON

It is detailed on the RTL-SDR project page: http://sdr.osmocom.org/trac/wiki/rtl-sdr

> How does an end-user change the centre frequency?

OpenWebRX was designed to work with a fixed LO, with multiple users tuning their DDC within the receiver bandwidth. In this manner it is similar to WebSDR. 

The reason behind this is that OpenWebRX is meant for setting up public receivers available on the Internet. If one user would change the receiver LO, then it would detune everything for all other users. 

As I've already got some requests to add this feature (complete receiver control) to OpenWebRX, it is likely that I will do it, but now there are more important things to work on. 

Until then, ShinySDR is the project which can be used for controlling all parameters of the receiver, but it is not feasible to set up public receivers with, only private ones.

Toontje, 2015-09-23:
rtl_sdr >/dev/null doesn't generate errors:

```ton@MacMini:/home/ton>rtl_sdr >/dev/null

rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage: -f frequency_to_tune_to [Hz]

[-s samplerate (default: 2048000 Hz)]

[-d device_index (default: 0)]

[-g gain (default: 0 for auto)]

[-b output_block_size (default: 16 * 16384)]

[-n number of samples to read (default: 0, infinite)]

[-S force sync output (default: async)]

filename (a '-' dumps samples to stdout)```

Thanks for the hint on ShinySDR.

Ton.

On Tue, Sep 22, 2015 at 9:28 PM, András Retzler <notifications@github.com>
wrote:

> When running as non-root i get the following error
>
> If you run it as root, everything works well?
> Maybe there is a problem with librtlsdr USB privileges.
> It especially applies if you get an error when you try to run the
> following without root privileges:
>
> rtl_sdr - > /dev/null
>
> You should recompile rtl_sdr with:
>
> cmake ../ -DINSTALL_UDEV_RULES=ON
>
> It is detailed on the RTL-SDR project page:
> http://sdr.osmocom.org/trac/wiki/rtl-sdr
>
> How does an end-user change the centre frequency?
>
> OpenWebRX was designed to work with a fixed LO, with multiple users tuning
> their DDC within the receiver bandwidth. In this manner it is similar to
> WebSDR.
>
> The reason behind this is that OpenWebRX is meant for setting up public
> receivers available on the Internet. If one user would change the receiver
> LO, then it would detune everything for all other users.
>
> As I've already got some requests to add this feature (complete receiver
> control) to OpenWebRX, it is likely that I will do it, but now there are
> more important things to work on.
>
> Until then, ShinySDR is the project which can be used for controlling all
> parameters of the receiver, but it is not feasible to set up public
> receivers with, only private ones.
>
> —
> Reply to this email directly or view it on GitHub
> <https://github.com/simonyiszk/openwebrx/issues/10#issuecomment-142392622>
> .
>

ha7ilm, 2015-09-23:
> rtl_sdr >/dev/null doesn't generate errors:

Instead of:

     rtl_sdr > /dev/null

I meant:

    rtl_sdr - > /dev/null

ha7ilm, 2015-09-23:
```ton@MacMini:/home/ton>rtl_sdr >/dev/null

Just curious, are you trying to run OpenWebRX on OS X, or on Ubuntu or a similar Linux distro?

Toontje, 2015-09-23:
Ubuntu 14.04 x32

```ton@MacMini:/home/ton>uname -a

Linux MacMini 3.16.0-46-generic #62~14.04.1-Ubuntu SMP Tue Aug 11 16:28:19
UTC 2015 i686 i686 i686 GNU/Linux```

```ton@MacMini:/home/ton>rtl_sdr - > /dev/null

Found 1 device(s):

  0:  Generic, RTL2832U, SN: 77771111153705700

Using device 0: Generic RTL2832U

Found Rafael Micro R820T tuner

Tuned to 100000000 Hz.

Reading samples in async mode… ```

On Wed, Sep 23, 2015 at 10:29 AM, András Retzler <notifications@github.com>
wrote:

> ```ton@MacMini:/home/ton>rtl_sdr >/dev/null
>
> Just curious, are you trying to run OpenWebRX on OS X, or on Ubuntu or a
> similar Linux distro?
>
> —
> Reply to this email directly or view it on GitHub
> <https://github.com/simonyiszk/openwebrx/issues/10#issuecomment-142528219>
> .
>

ha7ilm, 2015-09-23:
Okay, it seems fine, doesn't fail.
If you still have the error while running it as non-root, could you copy and paste the whole terminal output while running OpenWebRX? If you don't want to paste it here, you could send it to my e-mail address <a href="https://github.com/ha7ilm">available here</a>.

Toontje, 2015-09-23:
When i run it as non-root i get this:
```
OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] Started rtl thread: rtl_sdr -s 250000 -f 820000000 -p 0 - | nc -vvl 127.0.0.1 8888
[openwebrx-main] Started rtl_mus.
Listening on [127.0.0.1] (family 0, port 8888)
rtl_sdr: invalid option -- 'p'
rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	filename (a '-' dumps samples to stdout)

rtl_mus: Multi-User I/Q Data Server for RTL-SDR v0.22, made at HA5KFU Amateur Radio Club (http://ha5kfu.hu)
    code by Andras Retzler, HA7ILM <randras@sdr.hu>
    distributed under GNU GPL v3

2015-09-23 11:56:50,298 INFO Server is UP
2015-09-23 11:56:50,305 INFO Server listening on port: 4951
Connection from [127.0.0.1] port 8888 [tcp/*] accepted (family 2, sport 58217)
2015-09-23 11:56:50,306 INFO rtl_tcp host connection estabilished
2015-09-23 11:56:50,306 ERROR rtl_tcp host connection has closed, now trying to reopen
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-main] Starting HTTP server.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = sleep 1; nc -v localhost 4951 | csdr convert_u8_f | csdr flowcontrol 3000000 10 | csdr fft_cc 4096 27777 | csdr logpower_cf -70 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-spectrum] Spectrum thread started.
flowcontrol: flowcontrol_bufsize = 300000, flowcontrol_sleep = 100000
2015-09-23 11:56:52,266 INFO client accepted: 0@127.0.0.1:35755  users now: 1
Connection to localhost 4951 port [tcp/*] succeeded!
2015-09-23 11:56:52,367 ERROR with rtl_tcp host connection: [Errno 111] Connection refused
```

ha7ilm, 2015-09-23:
Thanks! As it is a common problem, I've just added the <a href="https://github.com/simonyiszk/openwebrx/wiki/Common-problems-and-their-solutions#rtl_sdr-invalid-option----p">solution</a> for this to the Wiki.

ha7ilm, 2015-09-23:
Anyway I've forgot that it only happens to you if you're not root.
Maybe you have multiple versions of `rtl-sd`r on your system, and another one is ran when you're root, or not?

ha7ilm, 2015-09-23:
That would be rather strange, but anyway you could check this:

    andris@ubuntu ~/openwebrx $ whereis rtl_sdr
    rtl_sdr: /usr/local/bin/rtl_sdr
    andris@ubuntu ~/openwebrx $ su
    ubuntu openwebrx # whereis rtl_sdr
    rtl_sdr: /usr/local/bin/rtl_sdr

I've already had a problem with different rtl_sdr versions on the same machine, one was installed from the Ubuntu repos and another was installed by me. The Ubuntu one was put at another place than the one installed with `sudo make install`, and that was the default.

Toontje, 2015-09-23:
Exactly what happened here. I has rtl_sdr installed via Ubuntu repos and they were in /usr/bin which is before /usr/local/bin in my $PATH.
All working now. But i still want to look at ShinySDR. ;-)

ha7ilm, 2017-01-19:
I hope this is solved by now.

closed 2017-01-19 by ha7ilm