ha7ilm/openwebrx #11: Problem in first set-up (no sound and no waterfall)

mehdideveloper, 2015-10-16:
Hi
I've installed all the pre-requisites, disabled the kernel module and tested that "rtl_sdr - > /dev/null" actually works. 
Python version: 2.7.3
I don't have any activity (sound or waterfall) in my browser!
Here's the output of "python openwebrx.py": 

```
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] Started rtl thread: rtl_sdr -s 250000 -f 88000000 -p 0 - | nc -vvl 127.0.0.1 8888
[openwebrx-main] Started rtl_mus.
listening on [any] 32928 ...
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM

rtl_mus: Multi-User I/Q Data Server for RTL-SDR v0.22, made at HA5KFU Amateur Radio Club (http://ha5kfu.hu)
    code by Andras Retzler, HA7ILM <randras@sdr.hu>
    distributed under GNU GPL v3

2015-10-16 15:02:53,564 INFO Server is UP
2015-10-16 15:02:53,568 INFO Server listening on port: 4951
2015-10-16 15:02:53,568 ERROR with rtl_tcp host connection: [Errno 111] Connection refused
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-main] Starting HTTP server.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = sleep 1; nc -v localhost 4951 | csdr convert_u8_f | csdr flowcontrol 3000000 10 | csdr fft_cc 4096 27777 | csdr logpower_cf -70 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-spectrum] Spectrum thread started.
[R82XX] PLL not locked!
Sampling at 250000 S/s.
Tuned to 88000000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
flowcontrol: flowcontrol_bufsize = 300000, flowcontrol_sleep = 100000
2015-10-16 15:02:55,540 INFO client accepted: 0@127.0.0.1:52586  users now: 1
```

I have also tried running with root, but there's no difference.

ha7ilm, 2015-10-16:
Hi Mehdi!

The problem seems to be here:

    [openwebrx-main] Started rtl thread: rtl_sdr -s 250000 -f 88000000 -p 0 - | nc -vvl 127.0.0.1 8888
    (...)
    listening on [any] 32928 ...

On different systems, different netcat versions exist, and sometimes they are not compatible with each other's commandline parameters. In this particular case netcat seems to start listen on port 32928 instead of 8888. I would be curious, which netcat do you have on your system (Hobbit's, GNU, ncat, etc.) and its version. Anyway, the nmap package used to have another netcat called `ncat`. Could you try replacing `nc` with `ncat` in `config_webrx.py`?

mehdideveloper, 2015-10-17:
Thank you. It solved my problem!
My nc version is "1.10-40"; I think it's GNU version (provided by default on Kali linux)
There's another occurrence of "nc" in csdr plugin's chain function. Do I need to replace that with ncat too?

ha7ilm, 2015-10-17:
The other occurence in `csdr/plugin.py` should work both with `nc` and `ncat`.
It is the problem with the `-p` switch only when trying to get netcat listening on a port: the GNU version seems to require it, but others fail if we use it.

closed 2015-10-17 by ha7ilm

wittend, 2015-10-17:
I am experiencing this problem using Beaglebone Black standard Debian.  I have nc ver. '1.10-40' as well  (apt-get from the standard repo, I think).

$ rtl_sdr -s 250000 -f 88000000 -p 0 - | nc -vvl 127.0.0.1 32928
listening on [any] 57019 ...Found 1 device(s):

  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sampling at 250000 S/s.
Tuned to 88000000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...

-- no issue apparent to me.  I'm not so clear what is happening on the other side of the connection.  But changing the port number in the config*.py doesn't seem to help.

ha7ilm, 2015-10-17:
You just have to replace `nc` with `ncat`, after setting up the `nmap` package:

    sudo apt-get install nmap