ha7ilm/openwebrx #12: openwebrx on Mac OS X

mehdideveloper, 2015-10-19:
Hi
I managed to compile fftw3 and csdr on OS X. Now (like my previous issue), when I go to the webpage, I just have an empty spectrum and no sound. (Also replaced nc with ncat in config_webrx.py)
Can you help me troubleshoot this?
Mac OS X 10.11
fftw 3.3.4 (compiled from source)

Here's the output:

```
acBook-Pro:openwebrx mehdi$ python openwebrx.py

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] Started rtl thread: rtl_sdr -s 250000 -f 145525000 -p 0 - | ncat -vvl 127.0.0.1 8888
[openwebrx-main] Started rtl_mus.
Ncat: Version 6.47 ( http://nmap.org/ncat )
Ncat: Listening on 127.0.0.1:8888

rtl_mus: Multi-User I/Q Data Server for RTL-SDR v0.22, made at HA5KFU Amateur Radio Club (http://ha5kfu.hu)
    code by Andras Retzler, HA7ILM <randras@sdr.hu>
    distributed under GNU GPL v3

2015-10-19 16:30:49,908 INFO Server is UP
Ncat: Connection from 127.0.0.1.
Ncat: Connection from 127.0.0.1:56872.
2015-10-19 16:30:49,917 INFO Server listening on port: 4951
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = sleep 1; nc -v localhost 4951 | csdr convert_u8_f | csdr flowcontrol 3000000 10 | csdr fft_cc 4096 27777 | csdr logpower_cf -70 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-spectrum] Spectrum thread started.
Found 1 device(s):
[openwebrx-main] Starting HTTP server.
function name given in argument 1 does not exist.

function name given in argument 1 does not exist.

found 0 associations
found 1 connections:
     1:	flags=82<CONNECTED,PREFERRED>
	outif lo0
	src 127.0.0.1 port 56873
	dst 127.0.0.1 port 4951
	rank info not available
	TCP aux info available

2015-10-19 16:30:51,898 INFO client accepted: 0@127.0.0.1:56873  users now: 1
function name given in argument 1 does not exist.

Connection to localhost port 4951 [tcp/pwgwims] succeeded!
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sampling at 250000 S/s.
Tuned to 145525000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
2015-10-19 16:30:55,046 INFO rtl_tcp host connection estabilished
127.0.0.1 - - [19/Oct/2015 16:31:12] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client 127.0.0.1#8aa134823b021d4229188686fd7f479b :: client added. Clients now: 1
127.0.0.1 - - [19/Oct/2015 16:31:12] "GET /sdr.js HTTP/1.1" 200 -
127.0.0.1 - - [19/Oct/2015 16:31:12] "GET /openwebrx.js HTTP/1.1" 200 -
127.0.0.1 - - [19/Oct/2015 16:31:12] "GET /openwebrx.css HTTP/1.1" 200 -
127.0.0.1 - - [19/Oct/2015 16:31:12] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client 127.0.0.1#22315c84e6367d13fd36a3c6c0860620 :: client added. Clients now: 2
127.0.0.1 - - [19/Oct/2015 16:31:12] "GET /favicon.ico HTTP/1.1" 200 -
127.0.0.1 - - [19/Oct/2015 16:31:12] "GET /sdr.js HTTP/1.1" 200 -
127.0.0.1 - - [19/Oct/2015 16:31:12] "GET /openwebrx.js HTTP/1.1" 200 -
127.0.0.1 - - [19/Oct/2015 16:31:12] "GET /openwebrx.css HTTP/1.1" 200 -
127.0.0.1 - - [19/Oct/2015 16:31:12] "GET /favicon.ico HTTP/1.1" 200 -
[openwebrx-dsp-plugin:csdr] Command = nc -v localhost 4951 | csdr convert_u8_f | csdr flowcontrol 3000000 10 | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_4317379184_shift | csdr fir_decimate_cc 22 0.0068178 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_4317379184_bpf 0.0281615770483 HAMMING | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.03065759637 | csdr deemphasis_nfm_ff 11025 | csdr fastagc_ff | csdr convert_f_i16 | csdr encode_ima_adpcm_i16_u8
fir_decimate_cc: taps_length = 587
found 0 associations
found 1 connections:
     1:	flags=82<CONNECTED,PREFERRED>
	outif lo0
	src 127.0.0.1 port 56885
	dst 127.0.0.1 port 4951
	rank info not available
	TCP aux info available

2015-10-19 16:31:12,449 INFO client accepted: 1@127.0.0.1:56885  users now: 2
csdr: fifo control mode on
function name given in argument 1 does not exist.

csdr: fifo control mode on
fractional_decimator_ff: window = HAMMING
fractional_decimator_ff: taps_length = 133
deemphasis_nfm_ff: invalid sample rate (this function works only with specific sample rates).

shift_addition_cc: reinitialized to -0
shift_addition_cc: reinitialized to -0
function name given in argument 1 does not exist.

Connection to localhost port 4951 [tcp/pwgwims] succeeded!
bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 143) + (input_size = 370) - 1
(overlap_length = 142) = taps_length - 1
bandpass_fir_fft_cc: filter initialized, low_cut = -0.35202, high_cut = 0.35202
```

Mehdi

ha7ilm, 2015-10-19:
These are the problematic messages:

     function name given in argument 1 does not exist.

...which used to happen if an old version of *csdr* is ran, which doesn't have specific functions, that the newer version of OpenWebRX requires. (Functions are given in the first parameter, like *deemphasis_nfm_ff*.) Anyway, I'm not sure this is the case over here.

Maybe you could try this one?

https://github.com/sgentle/csdr/tree/osx

ha7ilm, 2015-10-19:
Strange. I even tried to build *csdr* with *clang* on Linux, and it worked.
(Had to do these changes to build with clang:)

    -PARAMS_LOOPVECT = -O3 -ffast-math -fdump-tree-vect-details -dumpbase dumpvect
    +PARAMS_LOOPVECT = -O3 
    -       c99 $(PARAMS_LOOPVECT) $(PARAMS_SIMD) $(LIBSOURCES) $(PARAMS_LIBS) $(PARAMS_MISC) -fpic -shared -o libcsdr.so
    +       clang $(PARAMS_LOOPVECT) $(PARAMS_SIMD) $(LIBSOURCES) $(PARAMS_LIBS) $(PARAMS_MISC) -fpic -shared -o libcsdr.so
    -       c99 $(PARAMS_LOOPVECT) $(PARAMS_SIMD) csdr.c $(PARAMS_LIBS) -L. -lcsdr $(PARAMS_MISC) -o csdr
    +       clang $(PARAMS_LOOPVECT) $(PARAMS_SIMD) csdr.c $(PARAMS_LIBS) -L. -lcsdr $(PARAMS_MISC) -o csdr

At this point I can't help really much as I don't have a Mac yet.

mehdideveloper, 2015-10-19:
Thank you. It finally worked.
I had to delete the "libcsdr.so" from my previous installation.
I also added "-L/usr/local/lib" and "-m64" to PARAMS_LIBS in Makefile

closed 2015-10-19 by mehdideveloper