ha7ilm/openwebrx #14: Fedora 22 Support

oliverkrystal, 2015-12-22:
python openwebrx.py 

    OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
    _________________________________________________________________________________________________

    Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

    [openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
    [openwebrx-main] Started rtl thread: rtl_sdr -s 250000 -f 146730000 -p 0 - | nc -vvl 127.0.0.1 8888
    [openwebrx-main] Started rtl_mus.
    Ncat: Version 6.47 ( http://nmap.org/ncat )
    Ncat: bind to 127.0.0.1:8888: Address already in use. QUITTING.
    Found 1 device(s):
    0:  Realtek, RTL2838UHIDIR, SN: 00000001

    Using device 0: Generic RTL2832U OEM
    usb_claim_interface error -6
    Failed to open rtlsdr device #0.

    rtl_mus: Multi-User I/Q Data Server for RTL-SDR v0.22, made at HA5KFU Amateur Radio Club (http://ha5kfu.hu)
	code by Andras Retzler, HA7ILM <randras@sdr.hu>
	distributed under GNU GPL v3

    2015-12-22 04:36:20,673 INFO Server is UP
    client accepted!
    2015-12-22 04:36:20,689 INFO Server listening on port: 4951
    2015-12-22 04:36:20,690 INFO rtl_tcp host connection estabilished
    [openwebrx-main] Starting watchdog threads.
    [openwebrx-main] Starting spectrum thread.
    [openwebrx-spectrum] Spectrum thread initialized successfully.
    [openwebrx-dsp-plugin:csdr] Command = sleep 1; nc -v localhost 4951 | csdr convert_u8_f | csdr flowcontrol 3000000 10 | csdr fft_cc 4096 27777 | csdr logpower_cf -70 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
    [openwebrx-spectrum] Spectrum thread started.
    [openwebrx-main] Starting HTTP server.
    flowcontrol: flowcontrol_bufsize = 300000, flowcontrol_sleep = 100000
    Ncat: Version 6.47 ( http://nmap.org/ncat )
    Ncat: Connection refused.

I'm using Fedora 22 and the nmap provided netcat from the repos.  I can open the page from a remote browser on the lan (as intended) but the `connection refused` occurs before connecting with a browser.  When I open chrome, it states `WebSocket has closed unexpectedly. Please reload the page.`

Any suggestions on what to try?  I have already attempted the nc -p\ncat edits to configure_webrx.py and the csdr/plugin.py.

ha7ilm, 2015-12-23:
Hi!

First of all, Merry Christmas!

Your problem seems to be related to the following output lines:

    Using device 0: Generic RTL2832U OEM
    usb_claim_interface error -6
    Failed to open rtlsdr device #0.

It seems that your RTL-SDR dongle cannot be accessed.

OpenWebRX should work OK for you, if the following command doesn't give an error message:

    rtl_sdr - >/dev/null

* You may have a permission problem related to udev rules, especially if the following command runs OK:

    sudo rtl_sdr - >/dev/null

Please follow the installation procedure for RTL-SDR from the following link: http://ha5kfu.sch.bme.hu/openwebrx-quick-setup

* You may have to correctly blacklist the `dvb_usb_rtl28xxu` kernel module, especially if the following temporary fix helps:

    sudo rmmod dvb_usb_rtl28xxu; rtl_sdr - >/dev/null

The instructions for this on the link above are not guaranteed to work on Fedora, so you will have to find out how to do it.

oliverkrystal, 2015-12-24:
Merry Christmas to you as well.  The errors may have been related to how I've been trying to get this to work.  A fresh reboot, and I get this:

./openwebrx.py 

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] Started rtl thread: rtl_sdr -s 250000 -f 146730000 -p 0 - | nc -vvl 127.0.0.1 8888
[openwebrx-main] Started rtl_mus.
Ncat: Version 6.47 ( http://nmap.org/ncat )
Ncat: Listening on 127.0.0.1:8888
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM

rtl_mus: Multi-User I/Q Data Server for RTL-SDR v0.22, made at HA5KFU Amateur Radio Club (http://ha5kfu.hu)
    code by Andras Retzler, HA7ILM <randras@sdr.hu>
    distributed under GNU GPL v3

2015-12-24 04:40:34,183 INFO Server is UP
Found Rafael Micro R820T tuner
Ncat: Connection from 127.0.0.1.
Ncat: Connection from 127.0.0.1:43990.
2015-12-24 04:40:34,212 INFO Server listening on port: 4951
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sampling at 250000 S/s.
Tuned to 146730000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = sleep 1; nc -v localhost 4951 | csdr convert_u8_f | csdr flowcontrol 3000000 10 | csdr fft_cc 4096 27777 | csdr logpower_cf -70 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-spectrum] Spectrum thread started.
2015-12-24 04:40:35,109 INFO rtl_tcp host connection estabilished
[openwebrx-main] Starting HTTP server.
Ncat: Version 6.47 ( http://nmap.org/ncat )
Ncat: Connection refused.
flowcontrol: flowcontrol_bufsize = 300000, flowcontrol_sleep = 100000
^CSignal caught, exiting!
2015-12-24 04:40:38,288 INFO Ctrl+C: aborting.
[openwebrx] Ctrl+C: aborting

Could it be that I'm using 10.13.37.11 as my server_hostname?  I want this to be accessible across my LAN but not externally.

ha7ilm, 2015-12-26:
Hi!

Does this help?

https://github.com/simonyiszk/openwebrx/commit/0778043eee3414cfc736d4fd0fe62d8badecf3ab

oliverkrystal, 2015-12-30:
Sorry for the wait.  No.  This does not help.

./openwebrx.py 

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] Started rtl_thread: rtl_sdr -s 250000 -f 146730000 -p 0 - | nc -vvl 127.0.0.1 8888
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Ncat: Version 6.47 ( http://nmap.org/ncat )
Ncat: [openwebrx-main] ncat detected, using it instead of rtl_mus:
Using device 0: Generic RTL2832U OEM
Version 6.47 ( http://nmap.org/ncat )
Ncat: Listening on 127.0.0.1:8888
Found Rafael Micro R820T tuner
Traceback (most recent call last):
  File "./openwebrx.py", line 579, in <module>
    main()
  File "./openwebrx.py", line 129, in main
    rtl_mus_cmd = "ncat localhost 8888 | ncat -4l %d -k --send-only --allow 127.0.0.1 " % cfg.iq_server_port
AttributeError: 'module' object has no attribute 'iq_server_port'
[OliverK@zaphodbeeblebrox openwebrx]$ Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sampling at 250000 S/s.
Tuned to 146730000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...

ha7ilm, 2015-12-30:
AttributeError: 'module' object has no attribute 'iq_server_port'

The error message indicates that `iq_server_port = 4951` is missing from `config_webrx.py`, although it should be there:

https://github.com/simonyiszk/openwebrx/blob/0778043eee3414cfc736d4fd0fe62d8badecf3ab/config_webrx.py#L115

oliverkrystal, 2015-12-30:
Getting closer.  From a fresh reboot:

# ==== Server settings ====
web_port=8073
server_hostname="10.13.37.11" # If this contains an incorrect value, the web UI may freeze on load (it can't open websocket
max_clients=5
iq_server_port = 4951

# ==== Web GUI configuration ====
receiver_name=""
receiver_location=""
receiver_qra=""
receiver_asl=200
receiver_ant="Longwire"
receiver_device="RTL-SDR"
receiver_admin="example@example.com"
receiver_gps=(47.000000,19.000000)
photo_height=350
photo_title="Panorama of Budapest from Schönherz Zoltán Dormitory"
photo_desc="""
You can add your own background photo and receiver information.<br />
Receiver is operated by: <a href="mailto:%[RX_ADMIN]">%[RX_ADMIN]</a><br/>
Device: %[RX_DEVICE]<br />
Antenna: %[RX_ANT]<br />
Website: <a href="http://localhost" target="_blank">http://localhost</a>
"""

# ==== sdr.hu listing ====
# If you want your ham receiver to be listed publicly on sdr.hu, then take the following steps:
# 1. Register at: http://sdr.hu/register
# 2. You will get an unique key by email. Copy it and paste here:
sdrhu_key = ""
# 3. Set this setting to True to enable listing:
sdrhu_public_listing = False

# ==== DSP/RX settings ====
dsp_plugin="csdr"
fft_fps=9
fft_size=4096
samp_rate = 250000

center_freq = 146730000
rf_gain = 5
ppm = 0 

audio_compression="adpcm" #valid values: "adpcm", "none" 
fft_compression="adpcm" #valid values: "adpcm", "none" 

start_rtl_thread=True 

# ==== I/Q sources (uncomment the appropriate) ====

# >> RTL-SDR via rtl_sdr 

start_rtl_command="rtl_sdr -s {samp_rate} -f {center_freq} -p {ppm} - | nc -vvl 127.0.0.1 8888".format(rf_gain=rf_gain, center_freq=center_freq, samp_rate=samp_rate, ppm=ppm)
format_conversion="csdr convert_u8_f"

#start_rtl_command="hackrf_transfer -s {samp_rate} -f {center_freq} -g {rf_gain} -l16 -a0 -r hackrf_pipe & cat hackrf_pipe | nc -vvl 127.0.0.1 8888".format(rf_gain=rf_gain, center_freq=center_freq, samp_rate=samp_rate, ppm=ppm)
#format_conversion="csdr convert_i8_f"
## To use a HackRF, first run "mkfifo hackrf_pipe" in the OpenWebRX directory.
## You should also use the csdr git repo from here: 
##   git clone https://github.com/sgentle/csdr
##   git checkout origin/signed_char

# >> Sound card SDR (needs ALSA)
#I did not have the chance to properly test it.
#samp_rate = 96000
#start_rtl_command="arecord -f S16_LE -r {samp_rate} -c2 - | nc -vvl 127.0.0.1 8888".format(samp_rate=samp_rate)
#format_conversion="csdr convert_i16_f | csdr gain_ff 30"

# >> RTL_SDR via rtl_tcp
#start_rtl_command="rtl_tcp -s {samp_rate} -f {center_freq} -g {rf_gain} -P {ppm} -p 8888".format(rf_gain=rf_gain, center_freq=center_freq, samp_rate=samp_rate, ppm=ppm)
#format_conversion="csdr convert_u8_f"

# >> /dev/urandom test signal source
#samp_rate = 2400000
#start_rtl_command="cat /dev/urandom | (pv -qL `python -c 'print int({samp_rate} * 2.2)'` 2>&1) | nc -vvl 127.0.0.1  8888".format(rf_gain=rf_gain, center_freq=center_freq, samp_rate=samp_rate)
#format_conversion="csdr convert_u8_f"

#You can use other SDR hardware as well, by giving your own command that outputs the I/Q samples...

shown_center_freq = center_freq #you can change this if you use an upconverter

client_audio_buffer_size = 4 
#increasing client_audio_buffer_size will:
# - also increase the latency 
# - decrease the chance of audio underruns
[OliverK@zaphodbeeblebrox openwebrx]$ ./openwebrx.py 

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] Started rtl_thread: rtl_sdr -s 250000 -f 146730000 -p 0 - | nc -vvl 127.0.0.1 8888
Ncat: Version 7.00 ( https://nmap.org/ncat )
Ncat: Version 7.00 ( https://nmap.org/ncat )
Ncat: Listening on 127.0.0.1:8888
[openwebrx-main] ncat detected, using it instead of rtl_mus:
ncat localhost 8888 | ncat -4l 4951 -k --send-only --allow 127.0.0.1 
[openwebrx-main] Started rtl_mus.
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Ncat: Connection from 127.0.0.1.
Ncat: Connection from 127.0.0.1:52126.
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sampling at 250000 S/s.
Tuned to 146730000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = sleep 1; nc -v 127.0.0.1 4951 | csdr setbuf 11264 | csdr through | csdr convert_u8_f | csdr fft_cc 4096 27777 | csdr logpower_cf -70 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-main] Starting HTTP server.
Ncat: Version 7.00 ( https://nmap.org/ncat )
Ncat: Connected to 127.0.0.1:4951.
function name given in argument 1 does not exist.

function name given in argument 1 does not exist.

[openwebrx-spectrum] Spectrum thread started.
write: Broken pipe

ha7ilm, 2015-12-30:
function name given in argument 1 does not exist.

This indicates that you are running the dev branch of openwebrx, and not the dev branch of csdr.
Switch to the dev branch of csdr, then recompile it (`make && sudo make install`).

oliverkrystal, 2015-12-30:
After switching to the dev branch of csdr openwebrx is now functioning 
as expected.  Thank you for your time and patience in helping me to get 
this to run.

ha7ilm, 2015-12-30:
Great! Have fun with it! :-)

closed 2015-12-30 by ha7ilm