ha7ilm/openwebrx #15: Setting shift for use with Up-Converter?

bogstandard, 2015-12-29:
Hi,

I'm currently using an up-converter ([Ham It Up v1.3](http://www.nooelec.com/store/ham-it-up.html)) to receive signals.

The "Shift" I use within SDR# to have these display correctly is -124,991,371 ([screenshot](http://i.imgur.com/pEsiuRn.png)), though the recommended amount is -125,000,000. 

Without this shift signals are only accessible at ~125MHz + *actual signal frequency*, which means the frequencies displayed on my OpenWebRX client UI are ~125MHz above what they really are.

For example: To set the centre frequency of the OpenWebRX to 5505180Hz I have to put 130496551 as center_freq in config.webrx.py, which allows tuning but displays the incorrect frequencies.

Does OpenWebRX have the ability to implement a shift so that it can work nicely with an UpConverter?

I hope this makes sense, any help is very much appreciated.

ha7ilm, 2015-12-29:
Hi,

There is an option called `shown_center_freq` in `config_webrx.py`, <a href="https://github.com/simonyiszk/openwebrx/blob/76350936794dc21da313b842b95fab17eda5372f/config_webrx.py#L111">over here</a>.

For example, in your application:

    center_freq = 130496551
    shown_center_freq = 5505180

bogstandard, 2015-12-30:
Excellent, many thanks.

closed 2015-12-30 by bogstandard