ha7ilm/openwebrx #18: Labels over the frequency scale

ha7ilm, 2016-01-30:
We should be able to label some frequencies. 
Clicking on the label should set frequency and modulation mode.
* The list of labels should be contained in a separate file.

DE8MSH, 2016-03-25:
Hi.

This is the most important enhancement inmo. But not the crappy ones of k*****r. The ones of w****r. ;)

ha7ilm, 2016-03-26:
:smile: In progress. Have already made the webdesign part of it.

jester-xbmc, 2016-03-26:
nice project, and +1 for this

Lonecrowe, 2018-07-05:
I've seen this on other servers. - I don't see it in the documentation here.  Is this a custom addon someone did? Or is it officially in the product now?  Also see other people have added digimodes and such.  Are these all forks?