ha7ilm/openwebrx #19: "WebSocket has closed unexpectedly. Please reload the page." problem

alwayswannasleep, 2016-02-03:
```
OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] Started rtl thread: rtl_sdr -s 250000 -f 145525000 -p 0 - | nc -vvl 127.0.0.1 8888
[openwebrx-main] Started rtl_mus.
Listening on [127.0.0.1] (family 0, port 8888)
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM

rtl_mus: Multi-User I/Q Data Server for RTL-SDR v0.22, made at HA5KFU Amateur Radio Club (http://ha5kfu.hu)
    code by Andras Retzler, HA7ILM <randras@sdr.hu>
    distributed under GNU GPL v3

2016-02-03 22:52:04,689 INFO Server is UP
2016-02-03 22:52:04,692 INFO Server listening on port: 4951
Connection from [127.0.0.1] port 8888 [tcp/*] accepted (family 2, sport 37714)
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sampling at 250000 S/s.
Tuned to 145525000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-main] Starting HTTP server.
 [openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = sleep 1; nc -v localhost 4951 | csdr convert_u8_f | csdr flowcontrol 3000000 10 | csdr fft_cc 4096 27777 | csdr logpower_cf -70 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-spectrum] Spectrum thread started.
2016-02-03 22:52:05,783 INFO rtl_tcp host connection estabilished
2016-02-03 22:52:06,673 INFO client accepted: 0@127.0.0.1:56654  users now: 1
Connection to localhost 4951 port [tcp/*] succeeded!
Illegal instruction (core dumped)
Illegal instruction (core dumped)
2016-02-03 22:52:07,107 INFO client disconnected: 0@127.0.0.1
37.229.155.161 - - [03/Feb/2016 22:52:10] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client 37.229.155.161#4d004fd79a71ee9dfdc168280757680a :: client added. Clients now: 1
37.229.155.161 - - [03/Feb/2016 22:52:10] "GET /openwebrx.css HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:10] "GET /sdr.js HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:10] "GET /openwebrx.js HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:10] "GET /gfx/openwebrx-top-photo.jpg HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:10] "GET /gfx/openwebrx-top-logo.png HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:11] "GET /gfx/font-expletus-sans/ExpletusSans-Medium.ttf HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:11] "GET /gfx/openwebrx-ha5kfu-top-logo.png HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:11] "GET /gfx/openwebrx-rx-details-arrow-up.png HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:11] "GET /gfx/openwebrx-avatar.png HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:11] "GET /gfx/openwebrx-avatar-background.png HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:11] "GET /gfx/openwebrx-rx-details-arrow.png HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:11] "GET /gfx/openwebrx-background-cool-blue.png HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:11] "GET /gfx/openwebrx-scale-background.png HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:11] "GET /gfx/openwebrx-panel-log.png HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:11] "GET /gfx/openwebrx-panel-receiver.png HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:11] "GET /gfx/openwebrx-panel-status.png HTTP/1.1" 200 -
37.229.155.161 - - [03/Feb/2016 22:52:11] "GET /favicon.ico HTTP/1.1" 200 -
[openwebrx-dsp-plugin:csdr] Command = nc -v localhost 4951 | csdr convert_u8_f | csdr flowcontrol 3000000 10 | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_140350216572584_shift | csdr fir_decimate_cc 22 0.0068178 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_140350216572584_bpf 0.0281615770483 HAMMING | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.03065759637 | csdr deemphasis_nfm_ff 11025 | csdr fastagc_ff | csdr convert_f_i16 | csdr encode_ima_adpcm_i16_u8
csdr: fifo control mode on
fir_decimate_cc: taps_length = 587
csdr: fifo control mode on
fractional_decimator_ff: window = HAMMING
fractional_decimator_ff: taps_length = 133
shift_addition_cc: reinitialized to -0
[openwebrx-spectrum] client spectrum queue full, closing it.
[openwebrx-httpd] client 37.229.155.161#4d004fd79a71ee9dfdc168280757680a :: client being closed.
[openwebrx-httpd:ws] client closed by other thread
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140350216572584_bpf
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140350216572584_bpf
[openwebrx-httpd] client cannot be closed:  <class '__main__.ClientNotFoundException'>
  File "./openwebrx.py", line 448, in do_GET
    id_to_close=get_client_by_id(myclient.id,False)
  File "./openwebrx.py", line 251, in get_client_by_id
    raise ClientNotFoundException

```
I'm using Ubuntu 14.04 LTS. For installation used [this guide](https://gist.github.com/ha7ilm/944f2c2f7666256bff83#file-openwebrx-quick-setup-sh). When I open chrome, it states `WebSocket has closed unexpectedly. Please reload the page.`. What I can to do? I have already attempted the nc -p\ncat edits to configure_webrx.py. Thanks.

ha7ilm, 2016-02-03:
Seems like `csdr` fails:

    Illegal instruction (core dumped)
    Illegal instruction (core dumped)

Maybe it is a problem with compiler flags. Could you provide more information about your hardware (PC/embedded)?

Could you also post the output of the following:

    cat /proc/cpuinfo

...and the output of:

    cd csdr
    make

alwayswannasleep, 2016-02-04:
```
pavel@pavel-System-Product-Name:~$ cat /proc/cpuinfo
processor       : 0
vendor_id       : AuthenticAMD
cpu family      : 16
model           : 6
model name      : AMD Athlon(tm) II X2 250 Processor
stepping        : 2
microcode       : 0x100009f
cpu MHz         : 1800.000
cache size      : 1024 KB
physical id     : 0
siblings        : 2
core id         : 0
cpu cores       : 2
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 5
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm 3dnowext 3dnow constant_tsc rep_good nopl nonstop_tsc extd_apicid pni monitor cx16 popcnt lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt hw_pstate npt lbrv svm_lock nrip_save vmmcall
bugs            : tlb_mmatch apic_c1e fxsave_leak
bogomips        : 5999.82
TLB size        : 1024 4K pages
clflush size    : 64
cache_alignment : 64
address sizes   : 48 bits physical, 48 bits virtual
power management: ts ttp tm stc 100mhzsteps hwpstate

processor       : 1
vendor_id       : AuthenticAMD
cpu family      : 16
model           : 6
model name      : AMD Athlon(tm) II X2 250 Processor
stepping        : 2
microcode       : 0x100009f
cpu MHz         : 800.000
cache size      : 1024 KB
physical id     : 0
siblings        : 2
core id         : 1
cpu cores       : 2
apicid          : 1
initial apicid  : 1
fpu             : yes
fpu_exception   : yes
cpuid level     : 5
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm 3dnowext 3dnow constant_tsc rep_good nopl nonstop_tsc extd_apicid pni monitor cx16 popcnt lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt hw_pstate npt lbrv svm_lock nrip_save vmmcall
bugs            : tlb_mmatch apic_c1e fxsave_leak
bogomips        : 5999.82
TLB size        : 1024 4K pages
clflush size    : 64
cache_alignment : 64
address sizes   : 48 bits physical, 48 bits virtual
power management: ts ttp tm stc 100mhzsteps hwpstate
```
```

pavel@pavel-System-Product-Name:~$ cd csdr/
pavel@pavel-System-Product-Name:~$ sudo make

rm -f dumpvect*.vect
NOTE: you may have to manually edit Makefile to optimize for your CPU (especially if you compile on ARM, please edit PARAMS_NEON).
Auto-detected optimization parameters: -msse -msse2 -msse4 -mfpmath=sse

c99 -O3 -ffast-math -fdump-tree-vect-details -dumpbase dumpvect -msse -msse2  -msse4   -mfpmath=sse  fft_fftw.c libcsdr_wrapper.c  -g -lm -lrt -lfftw3f -DUSE_FFTW -DLIBCSDR_GPL -DUSE_IMA_ADPCM -Wno-unused-result -fpic -shared -o libcsdr.so
./parsevect dumpvect*.vect

[... 59 lines cut ...]
libcsdr.c:790: note: LOOP VECTORIZED.  logpower_cf: pass 3
libcsdr.c:788: note: not vectorized: relevant stmt not supported: _33 = __builtin_log10f (_32);  logpower_cf: pass 2
libcsdr.c:786: note: LOOP VECTORIZED.  logpower_cf: pass 1
libcsdr.c:805: note: LOOP VECTORIZED.  convert_u8_f
libcsdr.c:810: note: LOOP VECTORIZED.  convert_s8_f
libcsdr.c:815: note: LOOP VECTORIZED.  convert_i16_f
libcsdr.c:820: note: LOOP VECTORIZED.  convert_f_u8
libcsdr.c:827: note: LOOP VECTORIZED.  convert_f_s8
libcsdr.c:837: note: LOOP VECTORIZED.  convert_f_i16
libcsdr.c:844: note: LOOP VECTORIZED.  trivial_vectorize: should pass :-)
libcsdr_gpl.c:50: note: not vectorized: number of iterations cannot be computed.  shift_addition_cc: normalize starting_phase
libcsdr_gpl.c:49: note: not vectorized: number of iterations cannot be computed.  shift_addition_cc: normalize starting_phase
libcsdr_gpl.c:37: note: not vectorized: unsupported use in stmt.  shift_addition_cc: work
libcsdr_gpl.c:77: note: not vectorized: number of iterations cannot be computed.  shift_addition_cc: work
libcsdr_gpl.c:77: note: not vectorized: Bad inner loop.  shift_addition_cc: work
libcsdr_gpl.c:84: note: not vectorized: number of iterations cannot be computed.  shift_addition_cc: normalize phase
libcsdr_gpl.c:131: note: not vectorized: number of iterations cannot be computed.  shift_addition_cc: normalize starting_phase
libcsdr_gpl.c:130: note: not vectorized: number of iterations cannot be computed.  shift_addition_cc: normalize starting_phase
libcsdr_gpl.c:115: note: not vectorized: number of iterations cannot be computed.  shift_addition_cc: work
libcsdr_gpl.c:171: note: not vectorized: control flow in loop.  agc_ff
ima_adpcm.c:157: note: not vectorized: not suitable for gather load step_42 = _stepSizeTable[state$index_36];
ima_adpcm.c:168: note: not vectorized: not suitable for gather load step_34 = _stepSizeTable[state$index_30];
c99 -O3 -ffast-math -fdump-tree-vect-details -dumpbase dumpvect -msse -msse2  -msse4   -mfpmath=sse  csdr.c -g -lm -lrt -lfftw3f -DUSE_FFTW -DLIBCSDR_GPL -DUSE_IMA_ADPCM -L. -lcsdr -Wno-unused-result -o csdr

```

bogstandard, 2016-02-05:
I'm also facing similar issues on my AMD processor machine. After the localhost is declared open the next line is "illegal instruction".

I am running the very same OS and software as my Intel machine and I only get this error on the AMD Processor machine.

Are you able to say what would be the correct flags to have on the csdr makefile for an AMD Processor?

ha7ilm, 2016-02-17:
A temporary solution for this problem would be removing the `$(call cpufeature,sse4,-msse4)` from the `Makefile` of `csdr`, and then recompingind and reinstalling (`make && sudo make install`) `csdr`. 

Does it solve the problem for you?

Excalibur201010, 2016-02-18:
For me this solution is helped, Thanks

alwayswannasleep, 2016-02-18:
Thanks, that solved a problem for me too.

ha7ilm, 2019-01-24:
Fredrik, 

It is rather related to this bug: https://github.com/simonyiszk/csdr/issues/40

Andras

ha7ilm, 2019-01-24:
@cfenell I've proposed a fix for your problem. Try to update to the latest `csdr`, and please give feedback at simonyiszk/csdr#40 if it worked.