ha7ilm/openwebrx #21: Possible to link to RX with frequency and mode in an URL

sm3sgp, 2016-02-07:
I would be nice to be able to directly via the URL be able to point to an RX including frequency and mode.
Example:
http://openwebrx.host.com/?tune=3638.5lsb

Additionally be able to copy the URL with frequency and mode when listening to a station on an openwebrx.

ha7ilm, 2016-02-07:
Already implemented in the dev branch:

https://github.com/simonyiszk/openwebrx/tree/dev

> OpenWebRX now supports URLs like: http://localhost:8073/#freq=145555000,mod=usb

(I'm sorry about that I'm not progressing fast with releasing new versions. Many improvements are already in the dev branch, for example a NEON-optimized DDC.)

sm3sgp, 2016-02-07:
Ok, great! I will probably add some more "issues". I will check the dev branch before I post something.

ha7ilm, 2016-02-07:
No problem, feel free to add more issues / bugs / ideas!

closed 2016-02-07 by ha7ilm