ha7ilm/openwebrx #23: No Audio

Mauro-go, 2016-02-17:
Hello Mauro are IV3SCP
sorry for my English.....
it seems that everything works, I see the signals, but do not feel anything.
I'm using raspberry 2.
What could be the problem?

`OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
Ncat: Version 6.00 ( http://nmap.org/ncat )
[openwebrx-main] Started rtl_thread: rtl_sdr -s 2048000 -f 145525000 -p 0 -g 0 -| ncat -4l 4951 -k --send-only --allow 127.0.0.1
Found 1 device(s):
  0:  Generic, RTL2832U, SN: 77771111153705700

Using device 0: Generic RTL2832U
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2048000 S/s.
Tuned to 145525000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = sleep 1; ncat -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 227555 | csdr logpower_cf -70 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-main] Starting HTTP server.
[openwebrx-spectrum] Spectrum thread started.
Ncat: Version 6.00 ( http://nmap.org/ncat )
Ncat: Connected to 127.0.0.1:4951.
`

ha7ilm, 2016-02-17:
Hi Mauro!

Thanks for the report. Which browser do you use?

If you're using Google Chrome, could you open up the Developer Console (Ctrl+Shift+J) on the OpenWebRX page, and also post here any errors (in red), if there are any?

73!

Andras, HA7ILM

ha7ilm, 2016-02-17:
Well, are you trying to open the page with the built-in browser on Raspbian Jessie, or with Google Chrome from another computer?

Mauro-go, 2016-02-17:
Hello,
I always felt remotely, both firefox and chrome on Windows 10 and does not work.
Now I tried with firefox and windows xp and everything works properly.

Mauro-go, 2016-02-21:
These are errors with chrome and windows 10

audio_calculate_resampling() :: 16, 12000
sdr.js:140 Cannot enlarge memory arrays in asm.js. Either (1) compile with -s TOTAL_MEMORY=X with X higher than the current value 16777216, or (2) set Module.TOTAL_MEMORY before the program runs.
sdr.js:143 Cannot enlarge memory arrays in asm.js. Either (1) compile with -s TOTAL_MEMORY=X with X higher than the current value 16777216, or (2) set Module.TOTAL_MEMORY before the program runs

Mauro-go, 2016-02-21:
Uncaught abort() at Error
    at stackTrace (http://192.168.0.72:8073/sdr.js:1141:17)
    at abort (http://192.168.0.72:8073/sdr.js:11365:27)
    at enlargeMemory (http://192.168.0.72:8073/sdr.js:1166:5)
    at Function.Runtime.dynamicAlloc [as alloc] (http://192.168.0.72:8073/sdr.js:564:41)
    at _sbrk (http://192.168.0.72:8073/sdr.js:5635:26)
    at Object.bc [as _malloc] (http://192.168.0.72:8073/sdr.js:9543:29)
    at Object.asm$.malloc (http://192.168.0.72:8073/sdr.js:11407:26)
    at new sdrjs.RationalResamplerFF (http://192.168.0.72:8073/sdr.js:11579:31)
    at audio_preinit (http://192.168.0.72:8073/openwebrx.js:1447:20)
    at WebSocket.on_ws_recv (http://192.168.0.72:8073/openwebrx.js:1070:7)abort @ sdr.js:11365enlargeMemory @ sdr.js:1166Runtime.dynamicAlloc @ sdr.js:564_sbrk @ sdr.js:5635bc @ sdr.js:9543asm$.malloc @ sdr.js:11407sdrjs.RationalResamplerFF @ sdr.js:11579audio_preinit @ openwebrx.js:1447on_ws_recv @ openwebrx.js:1070

ha7ilm, 2016-02-21:
Thanks! These help much.

on6zg, 2016-05-23:
I believe this issue is still there ? I can't get any audio from any of the RTL-SDR receivers on the website. The kiwisdr ones are working perfectly

regards Jan

ha7ilm, 2016-05-23:
I think it only happens if the default sampling rate of your audio device is 192 ksps or higher (12000*16). It will be fixed anyway. I think recompiling `sdr.js` with `emcc -s TOTAL_MEMORY=33554432` would fix it.

ha7ilm, 2016-07-24:
It should be fixed now, at commit: 62f49cbf4c066cd792b77307b63ca0c542e0a92c