ha7ilm/openwebrx #32: Audio support on iOS iPad like on WebSDR

DE8MSH, 2016-03-25:
Hi András.

Audio support on iOS iPad like on Web**R would be cool in OpenWebRX! :)

ha7ilm, 2016-03-26:
Hi Marco,

Audio now works for me, tested on an iPad 2 in Safari.

ha7ilm, 2016-03-26:
Currently we miss the touch support for panning and zooming the waterfall display. 
I will also implement that.

ha7ilm, 2016-03-27:
I've just tested it on an iPhone 5S as well. The audio worked in Safari, but the UI doesn't fit the small screen (as expected).

DE8MSH, 2016-04-01:
So I'll give it a try. Thank you fer nfo.

ha7ilm, 2016-04-02:
I forgot to commit a file to git repo (big play button shown when OpenWebRX starts on iOS). Now fixed.

jimrochny, 2018-12-29:
Hello, there is no audio when running openwebrx on iOS, either on the iPad or an iPhone. However, if I have a Buetooth device connected, I can hear the audio over the Bluetooth device.

tylermcvaney, 2019-09-20:
@jimrochny I can confirm that this is still the case. I have tested multiple Apple devices and the audio only works with Bluetooth devices.