ha7ilm/openwebrx #36: zoom_levels_count

DH4CK, 2016-03-29:
Hi Andras,

I think the zoom_levels_count=5 in the openwebrx.js should more be like 20 instead of 5. I tested it and it makes it more smooth to zoom and easy to use and I find 5 steps makes it hard to use. Maybe there is also a way that makes it possible to access this variable through the config_webrx.py?

Hauke (ex DO5HHC - now DH4CK)

mara28ba, 2016-04-01:
I have found it here
/openwebrx/htdocs/opemwebrx.js
edit openwebrx.js at section /SCALE ROUTINES/

zoom_max_level_hps=13; //Hz/pixel  --> affect zoom level,default is 33, smaller nuber=higher zoom
zoom_levels_count=8;  --> affect zoom level counts ,default is 5 ,higher number more level counts

ka7u, 2016-04-17:
mara28ba, your suggested values really improve the CW operation. Thank you.