ha7ilm/openwebrx #37: If it is enabled in the config, let user control the parameters of the receiver hardware

ha7ilm, 2016-03-30:
These parameters would include:
* Center frequency
* Gain
* Auto gain
* etc...

This was useful for private receivers.

LDighera, 2016-05-09:
I am pleased to see this enhancement being addressed now.

Thank you for sharing all your hard work.  Very much appreciated.

kc9hzc, 2016-08-31:
I too am pleased!  These would be welcome options for a private receiver such as mine.  Frequency changes from within the browser session would be at the top of my wish list.  I saw mention of the ability to include '#freq=123456789,mod=fm' within the URL but this does not seem to work for me now.

I love this project, thanks for all of your hard work!!

silviuvulcan, 2017-03-01:
Hi. Any chance this is still a thing? I like the idea of a private receiver I can access remotely via vpn...

ha7ilm, 2017-03-01:
Yes, it's not forgotten.

Giorgiofox, 2017-03-22:
I agree, very very useful to set center frequency from web client for personal server! Please add it as soon as possible, and many thanks for this work!