ha7ilm/openwebrx #40: Mobile version website

Luceoria, 2016-04-17:
Would be nice to have one - i.e. as in WebSDR. Full version is a bit laggy for a phone/tablet.

vielmetti, 2016-10-26:
I've noticed this too - audio appears to come in fine when I connect to the SDR on my Android phone, but the UI doesn't cooperate to put the controls in a reasonable place.

bclswl0827, 2019-01-13:
That's what I'm going to do next...

bclswl0827, 2019-08-15:
https://github.com/simonyiszk/openwebrx/issues/132

9M2PJU, 2019-08-19:
> #132

tried to use you htdocs but websocket error

bclswl0827, 2019-08-19:
Make sure your WebSocket is the path at /ws/
If you are using HTTPS ( via Nginx/Apache or under Cloudflare), please mannually modify the field `ws://` to `wss://` in `openwort-mobi.js` and `openwebrx.js`
This code passed the test on my own and my friend’s device.
BTW, now you're able to preview this on [s.ibcl.us/webview/m.wrx](https://s.ibcl.us/webview/m.wrx)