ha7ilm/openwebrx #41: Waterfall keeps accumulating, propose a limit on the scroll-back

nickoe, 2016-04-18:
Waterfall keeps accumulating, propose a limit on the scroll-back.

It seems that waterfall images are only added to the page, this is nice such that you get a nice scroll-able scroll-back, but I think  a limit, preferable settable in the radio settings or similar should be implemented not to get out of memory after long while running the interface.

nickoe, 2016-04-21:
Hi @ha7ilm,

I have tried to fix this issue and have come up the below patch. I am not too familiar with javascrpt/html5, but I gave it a try. It seems to fix the memory leak problem. Initially I only had the `canvases.shift()`, but that only seemed to remove the view and not the data itself from memory. I have not made a pull request because I think this is an ugly hack, and I hope you could use it as input to implement a better solution maybe? Maybe there is a better way to actually remove the canvases from memory than using the `getElementById()` method.
```
diff --git a/htdocs/openwebrx.js b/htdocs/openwebrx.js
index 0055b13..211b48b 100644
--- a/htdocs/openwebrx.js
+++ b/htdocs/openwebrx.js
@@ -1692,6 +1692,15 @@ function add_canvas()
        new_canvas.addEventListener("mousedown", canvas_mousedown, false);
        new_canvas.addEventListener("wheel",canvas_mousewheel, false);
        canvases.push(new_canvas);
+       if (canvases.length > 30)
+       {
+               //Remove the canvases elemnt
+               canvases.shift();
+               //Remove the actual image ilement?
+               var list = document.getElementById("webrx-canvas-container");
+               // Magic numbver 5 is apparently the bottombost canvas object
+               list.removeChild(list.childNodes[5]);
+       }
 }

 function init_canvas_container()

```

DH4CK, 2016-05-08:
Hi nickoe,

good idea, I had the same problem with crashing of the page after a certain time.
I now set limit to "canvases.length > 20" and "canvas_default_height = 100" which reduces jumping of the scrollbar (because length is recduced) a little bit - no crashes yet.
A solution to have no jumping of the scrollbar might be to set a certain length of the webpage from beginning and have an additional element which hides the canaves before they get deleted (something similar to the the scale element) but at the bottom of the page, I guess.
Cool would be also an control element for the users to set these limits.

73s, Hauke, DH4CK