ha7ilm/openwebrx #42: Hangs one or twice every day

daniel5457h, 2016-06-20:
Hi,

I'm trying to set up a public SDR for listening CB comms near Barcelona, Spain and help the local CB enthusiasts listening each other when the QRM is high (URL http://z-ha-dum.dyndns.org:8080). However, it hangs once or twice every day, no errors are given on the log. I tried on a Rasbberry Pi 3 running Raspbian and in a NUC 6I3 running Ubuntu 16.04. Same thing.

When this happens, new clients reach this in the OpenWebRX log: "WebSocket opened to ws://z-ha-dum.dyndns.org:8080/ws/", but they never see the "Web Audio API succesfully initialized, sample rate: 44100 sps" that appears when it's running ok. No audio and no waterfall. 

So, any hints? Or maybe, is there any way I can increase the debug information / logging so we can properly locate the issue here?

By the way, our of despair, I'll just settle happily with a patch for a forced exit of OpenWebRX everytime this happens, I can have systemd restarting it as may times as it has to.

Best regards,
Daniel, EA3HRU.

ha7ilm, 2016-06-20:
Hi Daniel,

Thanks for reporting that. This kind of bug is hard to find without direct access to the system.

However, the first question is to see if it is OpenWebRX that hangs, or it is the receiver (via the `rtl_sdr` tool). The RTL-SDR can randomly fail for various reasons (but typically because of insufficient power supply or getting overheated).

Could you add `csdr through | csdr fifo` in the chain at the config?

This is the original line:

    start_rtl_command="rtl_sdr -s {samp_rate} -f {center_freq} -p {ppm} -g {rf_gain} -".format(rf_gain=rf_gain, center_freq=center_freq, samp_rate=samp_rate, ppm=ppm)

This is the modified line:

    start_rtl_command="rtl_sdr -s {samp_rate} -f {center_freq} -p {ppm} -g {rf_gain} - | csdr through | csdr fifo".format(rf_gain=rf_gain, center_freq=center_freq, samp_rate=samp_rate, ppm=ppm)

* With `csdr through`, you will be able to monitor the speed of the I/Q sample stream. 
* `csdr fifo` will always take all the possible input, even if the other end (in the OpenWebRX script) failed. If it happens, it will emit a lot of: `circular buffer full`, so don't pipe stderr into a log file.

It would help to see what you get at the console with these new settings while the system hangs.

VY 73!

Andras, HA7ILM

daniel5457h, 2016-06-21:
Thanks András,

Still printing numbers since last night, like

through: 4800061 bytes/s 55228864
through: 4800061 bytes/s 55230016
through: 4800061 bytes/s 55231232
through: 4800061 bytes/s 55232384
through: 4800061 bytes/s 55233536
through: 4800060 bytes/s 55234688
through: 4800061 bytes/s 55235904
through: 4800061 bytes/s 55237056
through: 4800061 bytes/s 55238208
through: 4800061 bytes/s 55239424
through: 4800061 bytes/s 55240576
through: 4800061 bytes/s 55241728
through: 4800061 bytes/s 55242944

The web is not accessible tough, I'm getting a "Sorry, the receiver is inactive due to internal error", I guess that's normal with this modification.

I'll let it run until tomorrow.

Thanks again for your help,
Daniel.

ha7ilm, 2016-06-21:
Hi Daniel,

Thanks! 

Since then, I have tried that configuration myself, and realized that it won't work with the `csdr fifo`. The correct command is:

    start_rtl_command="rtl_sdr -s {samp_rate} -f {center_freq} -p {ppm} -g {rf_gain} - | csdr through".format(rf_gain=rf_gain, center_freq=center_freq, samp_rate=samp_rate, ppm=ppm)

Then the web interface should also work.

Andras

ha7ilm, 2016-06-21:
In commit 4a79c6762 I have made some changes so that OpenWebRX will show some more verbose debug information on the USR1 signal.

If the hang happens again, we could get more information about the status of the software if you execute this from another terminal:

    killall -USR1 openwebrx python python2

daniel5457h, 2016-06-21:
Perfect András, I've done both changes, now the web is accessible again and USR1 signal prints some stuff. Let's wait now!

Daniel.

daniel5457h, 2016-06-22:
Hi,

I got a hang a little while ago. Here is the log on the console.

through: 4800056 bytes/s 84033856
through: 4800056 bytes/s 84035008

**Here I log in normally, all is ok**

192.168.1.8 - - [22/Jun/2016 09:20:18] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client 192.168.1.8#4f21a0daddfbc0fd8cbc9c22988ff3a0 :: client added. Clients now: 3
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /sdr.js HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /openwebrx.js HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /openwebrx.css HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /favicon.ico HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /openwebrx.js HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /sdr.js HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /openwebrx.css HTTP/1.1" 200 -
through: 4800056 bytes/s 84036160
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-top-photo.jpg HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-top-logo.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-ha5kfu-top-logo.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-avatar-background.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-avatar.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-rx-details-arrow-up.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-rx-details-arrow.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-panel-status.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-panel-log.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-panel-receiver.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-squelch-button.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-waterfall-auto.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-speaker.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-waterfall-default.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-zoom-in.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-zoom-out.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-zoom-in-total.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-zoom-out-total.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-play-button.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-scale-background.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /gfx/openwebrx-background-cool-blue.png HTTP/1.1" 200 -
192.168.1.8 - - [22/Jun/2016 09:20:18] "GET /favicon.ico HTTP/1.1" 200 -
[openwebrx-httpd:ws,2] command: SET output_rate=11025 action=start
[openwebrx-dsp-plugin:csdr] Command = ncat -v 127.0.0.1 4951 | csdr convert_u8_f | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_140094674991008_shift | csdr fir_decimate_cc 217 0.0006911875 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_140094674991008_bpf 0.0289357084727 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_140094674991008_squelch --outfifo /tmp/openwebrx_pipe_140094674991008_smeter 5 1 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.00308390023 | csdr deemphasis_nfm_ff 11025 | csdr fastagc_ff 1024 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
csdr: fifo control mode on
csdr: fifo control mode on
fir_decimate_cc: taps_length = 5787
csdr: fifo control mode on
fractional_decimator_ff: window = HAMMING
fractional_decimator_ff: taps_length = 133
squelch_and_power_cc: initial squelch level is 0
shift_addition_cc: reinitialized to -0
bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 139) + (input_size = 374) - 1
(overlap_length = 138) = taps_length - 1
Ncat: Version 7.01 ( https://nmap.org/ncat )
Ncat: Connected to 127.0.0.1:4951.
bandpass_fir_fft_cc: filter initialized, low_cut = -0.361696, high_cut = 0.361696
through: 4800056 bytes/s 84037376
through: 4800056 bytes/s 84038528
through: 4800056 bytes/s 84039680
[openwebrx-httpd:ws,2] command: SET audio_rate=44100
[openwebrx-httpd:ws] invalid parameter
[openwebrx-httpd:ws,2] command: SET mod=ssb low_cut=150 high_cut=3000 offset_freq=0
[openwebrx-dsp-plugin:csdr] Command = ncat -v 127.0.0.1 4951 | csdr convert_u8_f | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_140094674991008_shift | csdr fir_decimate_cc 217 0.0006911875 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_140094674991008_bpf 0.0289357084727 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_140094674991008_squelch --outfifo /tmp/openwebrx_pipe_140094674991008_smeter 5 1 | csdr realpart_cf | csdr fractional_decimator_ff 1.00308390023 | csdr agc_ff | csdr limit_ff | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
fractional_decimator_ff: window = HAMMING
fractional_decimator_ff: taps_length = 133
csdr: fifo control mode on
csdr: fifo control mode on
csdr: fifo control mode on
fir_decimate_cc: taps_length = 5787
bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 139) + (input_size = 374) - 1
[... 1118 lines cut ...]
[2016-06-22T09:32:15.466524] GET /gfx/openwebrx-speaker.png from 192.168.1.8
[... cut ...]
[2016-06-22T09:33:06.199976] Stopped streaming to client: 192.168.1.8#e496f0f15d6de07ce3a2230bc09168b3 (users now: 1)
[2016-06-22T09:33:06.200226] Stopped streaming to client: 192.168.1.8#01e7bd0cd1b0183c262d9207a97875b2 (users now: 0)

Any ideas? If we cannot solve the problem, but we manage to detect the condition and patch openwebrx to terminate, I'll be happy enough. I can restart it with systemd. All clients will be disconnected, but they can quickly connect again. Otherwise, they cannot connect until I restart it manually.

Thanks for your help!
Daniel.

ha7ilm, 2016-06-23:
@daniel5457h Thank you very much!

I am commited to solve this bug, as it affects multiple servers.

As of `bcastmsg` is not empty, I suspect that the inner processing loop hangs somewhere in `do_GET`.

I have just updated OpenWebRX so that it will print more relevant debug information (where it is in the loop) for the `-USR1` signal. Normally, the `loopstat` should be 10.

Could you try this new version?
In addition, could you run the following to print information?

    killall -USR1 openwebrx; sleep 0.1; killall -USR1 openwebrx

daniel5457h, 2016-06-24:
Thanks András,

I've updated it yesterday and I'm waiting for a hang to provide you more information.

Best regards and happy weekend!

daniel5457h, 2016-06-26:
Hi again András,

I got a hang, although the behavior this time was slightly different, as new clients had the audio initialized but they could only hear for a fraction of a second. No waterfall.

Here I paste the console output. I've kept the data monitor of csdr.

through: 4800058 bytes/s 167699712
through: 4800058 bytes/s 167700928
through: 4800058 bytes/s 167702080
through: 4800058 bytes/s 167703232
through: 4800058 bytes/s 167704384
through: 4800058 bytes/s 167705600
through: 4800058 bytes/s 167706752
through: 4800058 bytes/s 167707904
through: 4800058 bytes/s 167709120
through: 4800058 bytes/s 167710272
through: 4800058 bytes/s 167711424
through: 4800058 bytes/s 167712640
through: 4800058 bytes/s 167713792
through: 4800058 bytes/s 167714944
through: 4800058 bytes/s 167716160
through: 4800058 bytes/s 167717312
through: 4800058 bytes/s 167718464
through: 4800058 bytes/s 167719680
through: 4800058 bytes/s 167720832
through: 4800058 bytes/s 167721984
through: 4800058 bytes/s 167723136
through: 4800058 bytes/s 167724352
through: 4800058 bytes/s 167725504
through: 4800058 bytes/s 167726656
through: 4800058 bytes/s 167727872
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client 176.83.97.222#2d77d8150223ba2e0b6711dabd089cbc :: client added. Clients now: 3
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /sdr.js HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /openwebrx.css HTTP/1.1" 200 -
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411351920_bpf
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411351920_shift
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411351920_squelch
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411351920_smeter
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411354296_bpf
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411354296_shift
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411354296_squelch
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411354296_smeter
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411354944_bpf
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411354944_shift
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411354944_squelch
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411354944_smeter
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411361192_bpf
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411361192_shift
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411361192_squelch
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_140015411361192_smeter
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/openwebrx-top-photo.jpg HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /openwebrx.js HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/openwebrx-top-logo.png HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/openwebrx-ha5kfu-top-logo.png HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/openwebrx-avatar-background.png HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/openwebrx-avatar.png HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/openwebrx-rx-details-arrow-up.png HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/openwebrx-rx-details-arrow.png HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/openwebrx-panel-status.png HTTP/1.1" 200 -
through: 4800058 bytes/s 167729024
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/openwebrx-speaker.png HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/openwebrx-panel-receiver.png HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/openwebrx-waterfall-auto.png HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/openwebrx-waterfall-default.png HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/openwebrx-squelch-button.png HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:30:59] "GET /gfx/font-expletus-sans/ExpletusSans-Medium.ttf HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:31:00] "GET /gfx/openwebrx-background-cool-blue.png HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:31:00] "GET /gfx/openwebrx-scale-background.png HTTP/1.1" 200 -
176.83.97.222 - - [25/Jun/2016 12:31:00] "GET /gfx/openwebrx-zoom-in.png HTTP/1.1" 200 -
[... 725 lines cut ...]
through: 4800058 bytes/s 168115776
^CSignal caught, exiting!
[openwebrx] Ctrl+C: aborting.
[openwebrx-httpd] client 192.168.1.8#a404db7e0647a7a604c4ce7733d7d30b :: client being closed.
[openwebrx-httpd] client 192.168.1.8#581a1a3b5c600770d7514faff15e32c6 :: client being closed.
[openwebrx-httpd] client 192.168.1.9#66a9f6d8399933e7a86a8cf9788466d8 :: client being closed.
[openwebrx-httpd] client 192.168.1.9#ed944784cbf048ee0e9775589c5ea06b :: client being closed.
[openwebrx-httpd] client 192.168.1.9#ec6a02528344d44b613b26dcaa8cb31d :: client being closed.
[openwebrx-httpd] client 127.0.0.1#07674c0601c2751286df0bde81ddc6ec :: client being closed.

User cancel, exiting...

By the way, I've noticed that when it is operating normally, new clients generate these two lines in the log:

_bandpass_fir_fft_cc: filter initialized, low_cut = -0.331675, high_cut = 0.331675
bandpass_fir_fft_cc: filter initialized, low_cut = 0.0248756, high_cut = 0.248756_

which is normal, as I set as default a frequency not in the middle of the spectrum. However, when hanged, it only shows 

_bandpass_fir_fft_cc: filter initialized, low_cut = -0.331675, high_cut = 0.331675_

I don't know if that's relevant.

Thanks and best regards,
Daniel.

daniel5457h, 2016-06-27:
Hi András,

I got another hang today, Here I paste the console log and the access log. Same behavior as in the previous post.

through: 4800057 bytes/s 185312000
through: 4800057 bytes/s 185313152
through: 4800056 bytes/s 185314368
through: 4800056 bytes/s 185315520
through: 4800056 bytes/s 185316672
through: 4800056 bytes/s 185317888
192.168.1.8 - - [27/Jun/2016 08:37:15] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client 192.168.1.8#fa0f61f06185e88f6611f7fe26a374e2 :: client added. Clients now: 1
192.168.1.8 - - [27/Jun/2016 08:37:15] "GET /sdr.js HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:15] "GET /openwebrx.css HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:15] "GET /openwebrx.js HTTP/1.1" 200 -
through: 4800056 bytes/s 185319040
through: 4800056 bytes/s 185320192
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /favicon.ico HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /sdr.js HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /openwebrx.js HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /openwebrx.css HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-ha5kfu-top-logo.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-top-photo.jpg HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-top-logo.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-avatar.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-avatar-background.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-rx-details-arrow.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-panel-status.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-panel-log.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-panel-receiver.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-speaker.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-squelch-button.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-waterfall-auto.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-waterfall-default.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-zoom-in.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-zoom-out.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-zoom-out-total.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-zoom-in-total.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-play-button.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-scale-background.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-background-cool-blue.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/font-expletus-sans/ExpletusSans-Medium.ttf HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:16] "GET /gfx/openwebrx-rx-details-arrow-up.png HTTP/1.1" 200 -
192.168.1.8 - - [27/Jun/2016 08:37:17] "GET /favicon.ico HTTP/1.1" 200 -
[openwebrx-httpd:ws,0] command: SET output_rate=11025 action=start
[openwebrx-dsp-plugin:csdr] Command = ncat -v 127.0.0.1 4951 | csdr convert_u8_f | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_139625500840016_shift | csdr fir_decimate_cc 217 0.0006911875 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_139625500840016_bpf 0.0289357084727 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_139625500840016_squelch --outfifo /tmp/openwebrx_pipe_139625500840016_smeter 5 1 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.00308390023 | csdr deemphasis_nfm_ff 11025 | csdr fastagc_ff 1024 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
csdr: fifo control mode on
fir_decimate_cc: taps_length = 5787
csdr: fifo control mode on
shift_addition_cc: reinitialized to -0
bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 139) + (input_size = 374) - 1
(overlap_length = 138) = taps_length - 1
fractional_decimator_ff: window = HAMMING
fractional_decimator_ff: taps_length = 133
csdr: fifo control mode on
squelch_and_power_cc: initial squelch level is 0
Ncat: Version 7.01 ( https://nmap.org/ncat )
Ncat: Connected to 127.0.0.1:4951.
bandpass_fir_fft_cc: filter initialized, low_cut = -0.361696, high_cut = 0.361696
through: 4800056 bytes/s 185321344
through: 4800057 bytes/s 185322560
through: 4800056 bytes/s 185323712
through: 4800056 bytes/s 185324864
through: 4800057 bytes/s 185326080
through: 4800057 bytes/s 185327232
through: 4800057 bytes/s 185328384
through: 4800056 bytes/s 185329600
through: 4800057 bytes/s 185330752
through: 4800057 bytes/s 185331904
through: 4800056 bytes/s 185333120
[... 732 lines cut ...]
[2016-06-27T08:38:01.716615] GET /gfx/openwebrx-panel-status.png from 192.168.1.8
[2016-06-27T08:38:01.719168] GET /gfx/openwebrx-panel-log.png from 192.168.1.8
[2016-06-27T08:38:01.721425] GET /gfx/openwebrx-panel-receiver.png from 192.168.1.8
[2016-06-27T08:38:01.726529] GET /gfx/openwebrx-squelch-button.png from 192.168.1.8
[2016-06-27T08:38:01.727857] GET /gfx/openwebrx-waterfall-auto.png from 192.168.1.8
[2016-06-27T08:38:01.730981] GET /gfx/openwebrx-waterfall-default.png from 192.168.1.8
[2016-06-27T08:38:01.732178] GET /gfx/openwebrx-speaker.png from 192.168.1.8
[... cut ...]
[2016-06-27T08:39:30.784300] Stopped streaming to client: 192.168.1.8#fa0f61f06185e88f6611f7fe26a374e2 (users now: 2)
[2016-06-27T08:39:30.784721] Stopped streaming to client: 192.168.1.8#d31fc098b6fe238d12fab58c8d1f9374 (users now: 1)
[2016-06-27T08:39:30.788391] Stopped streaming to client: 192.168.1.8#74945cd6706197f0954d2d0486c24d87 (users now: 0)

From the IP's list connecting to the server, I can see that there are a few prior to the hang that are not actually clients, but seem rather port scanners trying the capabilities of the web server. 

Best regards,
   Daniel.

daniel5457h, 2016-06-30:
Hello András,

Another hang last night. Here I paste the console log.

through: 4800056 bytes/s 261042944
through: 4800057 bytes/s 261044160
through: 4800057 bytes/s 261045312
through: 4800057 bytes/s 261046464
through: 4800057 bytes/s 261047680
through: 4800057 bytes/s 261048832
through: 4800057 bytes/s 261049984
through: 4800056 bytes/s 261051200
through: 4800056 bytes/s 261052352
through: 4800057 bytes/s 261053504
through: 4800056 bytes/s 261054720
through: 4800056 bytes/s 261055872
through: 4800056 bytes/s 261057024
88.10.165.129 - - [29/Jun/2016 22:41:35] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client 88.10.165.129#b316fdb21c9950713137f6f009308693 :: client added. Clients now: 3
88.10.165.129 - - [29/Jun/2016 22:41:35] "GET /sdr.js HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:35] "GET /openwebrx.js HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:35] "GET /openwebrx.css HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:35] "GET /gfx/openwebrx-top-photo.jpg HTTP/1.1" 200 -
through: 4800056 bytes/s 261058176
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/font-expletus-sans/ExpletusSans-Medium.ttf HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-top-logo.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-ha5kfu-top-logo.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-avatar-background.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-avatar.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-rx-details-arrow-up.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-rx-details-arrow.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-panel-status.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-panel-log.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-panel-receiver.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-speaker.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-waterfall-auto.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-squelch-button.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-waterfall-default.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-zoom-in.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-zoom-out.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-zoom-out-total.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-zoom-in-total.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-scale-background.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-play-button.png HTTP/1.1" 200 -
88.10.165.129 - - [29/Jun/2016 22:41:36] "GET /gfx/openwebrx-background-cool-blue.png HTTP/1.1" 200 -
through: 4800057 bytes/s 261059392
[openwebrx-httpd:ws,2] command: SET output_rate=12000 action=start
[openwebrx-dsp-plugin:csdr] Command = ncat -v 127.0.0.1 4951 | csdr convert_u8_f | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_140266477345032_shift | csdr fir_decimate_cc 199 0.00075375 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_140266477345032_bpf 0.0265339966833 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_140266477345032_squelch --outfifo /tmp/openwebrx_pipe_140266477345032_smeter 5 1 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.005 | csdr deemphasis_nfm_ff 11025 | csdr fastagc_ff 1024 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
csdr: fifo control mode on
fir_decimate_cc: taps_length = 5307
csdr: fifo control mode on
shift_addition_cc: reinitialized to -0
csdr: fifo control mode on
fractional_decimator_ff: window = HAMMING
fractional_decimator_ff: taps_length = 133
bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 151) + (input_size = 362) - 1
(overlap_length = 150) = taps_length - 1
squelch_and_power_cc: initial squelch level is 0
Ncat: Version 7.01 ( https://nmap.org/ncat )
Ncat: Connected to 127.0.0.1:4951.
bandpass_fir_fft_cc: filter initialized, low_cut = -0.331675, high_cut = 0.331675
through: 4800056 bytes/s 261060544
through: 4800056 bytes/s 261061696
through: 4800057 bytes/s 261062912
through: 4800057 bytes/s 261064064
through: 4800057 bytes/s 261065216
through: 4800056 bytes/s 261066432
through: 4800057 bytes/s 261067584
through: 4800057 bytes/s 261068736
through: 4800056 bytes/s 261069952
[... 927 lines cut ...]
[2016-06-29T22:45:15.611929] GET /gfx/openwebrx-rx-details-arrow-up.png from 88.10.165.129
[2016-06-29T22:45:15.630990] GET /gfx/openwebrx-rx-details-arrow.png from 88.10.165.129
[2016-06-29T22:45:15.637571] GET /gfx/openwebrx-panel-status.png from 88.10.165.129
[2016-06-29T22:45:15.661083] GET /gfx/openwebrx-panel-log.png from 88.10.165.129
[2016-06-29T22:45:15.686975] GET /gfx/openwebrx-panel-receiver.png from 88.10.165.129
[2016-06-29T22:45:15.734657] GET /gfx/openwebrx-speaker.png from 88.10.165.129
[2016-06-29T22:45:15.740280] GET /gfx/openwebrx-waterfall-auto.png from 88.10.165.129
[2016-06-29T22:45:15.748610] GET /gfx/openwebrx-squelch-button.png from 88.10.165.129
[2016-06-29T22:45:15.765023] GET /gfx/openwebrx-waterfall-default.png from 88.10.165.129
[... cut ...]
[2016-06-29T22:45:17.200554] GET /gfx/openwebrx-background-cool-blue.png from 88.10.165.129
[2016-06-29T22:45:17.435698] GET /ws/aaefe742875f3b398cbab1f8d67a1a4a from 88.10.165.129
[2016-06-29T22:45:17.482142] Started streaming to client: 88.10.165.129#aaefe742875f3b398cbab1f8d67a1a4a (users now: 2)
[2016-06-29T22:45:58.928161] Stopped streaming to client: 192.168.1.9#6faa1ab6ee3e7895027c0b3ab807304a (users now: 1)
[2016-06-29T22:45:58.929161] Stopped streaming to client: 88.10.165.129#aaefe742875f3b398cbab1f8d67a1a4a (users now: 0)

I hope it helps.

Best regards,
Daniel.

JonathanThorpe, 2016-11-05:
Hi András,

I was wondering if there was any further feedback on how we might resolve this issue?

Running the current branch on two Raspberry Pi 3s at different locations and also get the same hangs as described above.

These receivers weren't used often (monitoring of radio transmitter sites), but recently, I wrote a plugin for our monitoring system that creates a WebSocket connection and monitors for a few samples (https://gist.github.com/JonathanThorpe/82274ea39930fcfee86e248a8a3a5a16) to determine if there is carrier.

This increase in connections obviously makes this happen a lot more frequently.

Happy to help with collecting more information, but just wanted to see if this is not an issue that has already been resolved given the quiet since June?

ha7ilm, 2016-11-06:
Currently I am quite sure that this bug was introduced when I replaced `rtl_mus` with `ncat`, and this is indeed a bug in `ncat`. I am working on my own implementation of `ncat` that would use threading, etc.

ha7ilm, 2016-11-06:
Given the fact that a part of users experience this problem, we can suppose that this behaviour might be different across `ncat` versions.

Could you try installing the latest `ncat`, and check if the problem is solved? This requires removing the `nmap` package, and compiling and installing it from source:

```bash
sudo apt-get remove nmap #if nmap is the dependency of anything important on your system, stop here
git clone https://github.com/nmap/nmap
cd nmap
./configure
make -j4
sudo make install
```

JonathanThorpe, 2016-11-06:
Hi ha7ilm,

I've updated two Raspberry Pi 3s with the ncat from the latest nmap branch and will report back in a couple of days.

Thanks!

ha7ilm, 2016-11-06:
@JonathanThorpe Great! Thanks in advance!

JonathanThorpe, 2016-11-06:
Hi @ha7ilm, unfortunately after 4 hours, 40 minutes on one and 4 hours 25 minutes on the other with regular access, both ncat processes seem to be unable to sending output when you connect to it - just like before.

ha7ilm, 2016-11-06:
@JonathanThorpe Thanks for the feedback! I'll fix it in another way then.

JonathanThorpe, 2016-11-06:
@ha7ilm no problem - happy to test anything you come up with.

Looking at the code, is there any reason this needs to use ncat (or equivalent)? Is it not possible to capture the output from stdout from the rtl_sdr process and then have openwebrx.py send the output to each csdr instance? Or would this be harmful to performance?

This is what the output looks like in this state:

```
27638 ?        Ssl    4:32 /usr/bin/python2 /opt/openwebrx/openwebrx.py
27645 ?        S      0:00  \_ /bin/sh -c rtl_sdr -D2 -s 2048000 -f 1069000 -p 0 -g 0 -| ncat -4l 4951 -k --send-only --allow 127.0.0.1
27646 ?        Sl    19:05  |   \_ rtl_sdr -D2 -s 2048000 -f 1069000 -p 0 -g 0 -
27647 ?        S    139:19  |   \_ ncat -4l 4951 -k --send-only --allow 127.0.0.1
27656 ?        S      0:00  \_ /bin/sh -c ncat -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 2880 | csdr logaveragepower_cf -70 4096 79 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
27657 ?        S     24:32  |   \_ ncat -v 127.0.0.1 4951
27658 ?        S     99:22  |   \_ csdr convert_u8_f
27659 ?        R    184:59  |   \_ csdr fft_cc 4096 2880
27660 ?        S     41:31  |   \_ csdr logaveragepower_cf -70 4096 79
27661 ?        S      0:28  |   \_ csdr fft_exchange_sides_ff 4096
27662 ?        S      1:31  |   \_ csdr compress_fft_adpcm_f_u8 4096
31208 ?        Z      0:00  \_ [sh] <defunct>
31841 ?        Z      0:00  \_ [sh] <defunct>
  673 ?        Z      0:00  \_ [sh] <defunct>
 1430 ?        Z      0:00  \_ [sh] <defunct>
 2874 ?        Z      0:00  \_ [sh] <defunct>
 4315 ?        Z      0:00  \_ [sh] <defunct>
 4387 ?        Z      0:00  \_ [sh] <defunct>
 4444 ?        Z      0:00  \_ [sh] <defunct>
 4502 ?        Z      0:00  \_ [sh] <defunct>
 4559 ?        Z      0:00  \_ [sh] <defunct>
 4616 ?        Z      0:00  \_ [sh] <defunct>
 4672 ?        Z      0:00  \_ [sh] <defunct>
 4728 ?        Z      0:00  \_ [sh] <defunct>
 4785 ?        Z      0:00  \_ [sh] <defunct>
 4843 ?        Z      0:00  \_ [sh] <defunct>
 4898 ?        Z      0:00  \_ [sh] <defunct>
 4956 ?        Z      0:00  \_ [sh] <defunct>
 5013 ?        Z      0:00  \_ [sh] <defunct>
```

Definitely ncat as you suggest - if I try to connect, it just fails:
```
pi@openwebrx:~ $ ncat -v 127.0.0.1 4951
Ncat: Version 7.31SVN ( https://nmap.org/ncat )

Ncat: Broken pipe.
```

ha7ilm, 2016-11-06:
bq. Is it not possible to capture the output from stdout from the rtl_sdr process and then have openwebrx.py send the output to each csdr instance? Or would this be harmful to performance?

The former solution (rtl_mus.py) was doing something similar, but the performance was not good, even with `pypy`. `ncat` can do 3 times the performance because it is written in C.

JonathanThorpe, 2016-11-07:
Good point. Was thinking about how to address this myself - perhaps something written in C or Go (might work with multiple threads better) that relies on UNIX sockets or UDP instead of TCP?

JonathanThorpe, 2016-11-07:
Since Unix domain sockets are 1:1, there is actually a tee function on Linux (not to be confused with the utility) - wonder if this would be more efficient than hitting the network stack? https://linux.die.net/man/2/tee

tejeez, 2017-01-16:
The bug seems to be that ncat doesn't close its sockets after use. Looking at /proc/pid_of_ncat/fdinfo/ I found that every connection to the server leaves a file descriptor open even after the connection is closed. After about 100 connections it seems to run out of file descriptors or something and hangs. (Maybe we should also tell ncat developers about this bug...)

I replaced ncat with my own shared memory buffer implementation. This somewhat reduced CPU use and eliminated the hangs. My SHM buffer is nothing really fancy but I could share it anyway and make a pull request if you're interested.

JonathanThorpe, 2017-01-17:
@tejeez - definitely interested if you could please share? :)

ha7ilm, 2017-01-18:
@tejeez Thanks for figuring out the problem with `ncat`! 

In order to solve this issue, I'm also working on my implementation here: 
https://github.com/simonyiszk/csdr/tree/feature/nmux
Made some progress last week, but did not finalize it yet.
We could compare CPU usage between implementations at the end.

In fact, there were some other programs already, which could be used instead of `ncat`. 

**rtlmuxer:** https://github.com/alexander-sholohov/rtlmuxer
This works, but can't use `stdin` as the signal source, which would lead to the use of a superfluous `nc` command and OS socket. It also looks single threaded.

**rtlmux:** https://git.vocti.ca/slepp/rtlmux
This one uses pthreads which is impressive. It does not work if the stream starts with something else than "RTL\0", and I think I had other difficulties in integrating it, which I don't remember anymore. 

I decided to write `nmux` (based on `pthreads`) because I will need a command line program with a very similar structure for the DDC project I'm also working on.

tejeez, 2017-01-18:
@JonathanThorpe Here it is: https://github.com/tejeez/shmbuffer
Modification to OpenWebRX: https://github.com/tejeez/openwebrx/commit/35836c84519244162a5dac373bf410509c933bdb

It "seems to work" but has some known issues so it's probably not ready to merged into openwebrx.

ha7ilm, 2017-01-18:
@tejeez Thanks for sharing that! I'll have a look at it.

JonathanThorpe, 2017-01-18:
Thanks @tejeez - I've just installed this on two Raspberry Pis being used for monitoring which would previously stop responding a few times a day.

One thing I noticed is that I wasn't able to run OpenWebRX as a non-root user once I switched to the shm implementation - this could be my setup however which I will look at later.

I'll let you know how it goes.

ha7ilm, 2017-01-19:
I've just finished my fix, which can be installed like this:

```bash
git clone https://github.com/simonyiszk/csdr
cd csdr
git fetch
git checkout feature/nmux
make
sudo make install
cd ..
git clone https://github.com/simonyiszk/openwebrx
cd openwebrx
git fetch
git checkout feature/nmux
```
Now start OpenWebRX with:

    python2 openwebrx.py

----

@tejeez I think your implementation is superior in speed, achieved with much less code. 

I did some comparison, and these were my results:

### shmbuffer at 2.4 Msps, 10 clients
```
Tasks: 496 total,   1 running, 495 sleeping,   0 stopped,   0 zombie
KiB Mem:   8169328 total,  6403312 used,  1766016 free,     9220 buffers
KiB Swap:  4003836 total,  2864096 used,  1139740 free.  1008124 cached Mem
27917 pcfl      20   0  133252 131536 131448 S   1,0  1,6   0:00.75 shmread
28102 pcfl      20   0  133252 131532 131448 S   0,7  1,6   0:00.46 shmread
28287 pcfl      20   0  133252 131532 131448 S   0,7  1,6   0:00.45 shmread
28333 pcfl      20   0  133252 131532 131448 S   0,7  1,6   0:00.46 shmread
28380 pcfl      20   0  133252 131536 131448 S   0,7  1,6   0:00.44 shmread
27915 pcfl      20   0  133252 131272 131184 S   0,3  1,6   0:00.40 shmwrite
27967 pcfl      20   0  133252 131532 131448 S   0,3  1,6   0:00.52 shmread
28012 pcfl      20   0  133252 131532 131448 S   0,3  1,6   0:00.50 shmread
28056 pcfl      20   0  133252 131536 131448 S   0,3  1,6   0:00.47 shmread
28147 pcfl      20   0  133252 131536 131448 S   0,3  1,6   0:00.48 shmread
28191 pcfl      20   0  133252 131532 131448 S   0,3  1,6   0:00.45 shmread
28243 pcfl      20   0  133252 131536 131448 S   0,3  1,6   0:00.45 shmread
```

### nmux at 2.4 Msps, 10 clients
```
Tasks: 494 total,   3 running, 491 sleeping,   0 stopped,   0 zombie
KiB Mem:   8169328 total,  7539228 used,   630100 free,    54552 buffers
KiB Swap:  4003836 total,  2767756 used,  1236080 free.  1468388 cached Mem
 1658 pcfl      20   0  152716  52684    964 S   5,6  0,6   0:04.84 nmux
 2159 pcfl      20   0    2640    708    596 S   1,3  0,0   0:00.94 nc
 1669 pcfl      20   0    2640    704    596 S   1,0  0,0   0:01.22 nc
 1719 pcfl      20   0    2640    708    596 S   1,0  0,0   0:01.07 nc
 1771 pcfl      20   0    2640    708    596 S   1,0  0,0   0:00.99 nc
 1821 pcfl      20   0    2640    704    596 S   1,0  0,0   0:00.99 nc
 1868 pcfl      20   0    2640    708    596 S   1,0  0,0   0:00.97 nc
 1915 pcfl      20   0    2640    704    596 S   1,0  0,0   0:00.96 nc
 1965 pcfl      20   0    2640    704    596 S   1,0  0,0   0:00.97 nc
 2079 pcfl      20   0    2640    708    596 S   1,0  0,0   0:00.94 nc
 2254 pcfl      20   0    2640    704    596 S   1,0  0,0   0:00.84 nc
 2021 pcfl      20   0    2640    708    596 S   0,7  0,0   0:00.95 nc
```

One thing missing right now is that we need to allow multiple OpenWebRX instances running on the same machine. I'll also be thinking about the other issues you've mentioned in the README. Anyway, I'd be happy to include it into OpenWebRX, because it is fast.

JonathanThorpe, 2017-01-20:
@tejeez More than 24 hours running with constant automation connecting to OpenWebRX every 5 minutes and it's working really well.

ha7ilm, 2017-02-08:
After some tests, I merged my solution. This is because I'm going to restructure the program in a way that the TCP conections will be used only for low data rate. However, @teejez's method could be used between the `rtl_sdr` tool and the next process, which would be a TCP server that does the channelization as well.

closed 2019-01-24 by ha7ilm