ha7ilm/openwebrx #43: Unexpected "client being closed" each few minutes. Broken pipe

raulmerlin, 2016-06-22:
First of all congratulations for this project, as far the best web sdr server of much that I tested before.

I'm having a problem with the server online at http://www.merlitec.com:18080,
after some minutes after opening a client, it drops down with this server output:

[openwebrx-httpd] GET /status/ from 152.66.211.30
[openwebrx-httpd] GET /status/ from 152.66.211.30
[openwebrx-httpd] GET /status/ from 152.66.211.30
[openwebrx-httpd] GET /status/ from 152.66.211.30 // STATUS REQUESTS FROM OPENWEBRX
**[openwebrx-httpd] client 138.4.36.3#3868f6e9b5e38755998b86834f39cb84 :: client being closed.**
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_1949045104_bpf
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_1949045104_shift
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_1949045104_squelch
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_1949045104_smeter
----------------------------------------
Exception happened during processing of request from ('138.4.36.3', 49191)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 599, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 657, in __init__
    self.finish()
  File "/usr/lib/python2.7/SocketServer.py", line 716, in finish
    self.wfile.close()
  File "/usr/lib/python2.7/socket.py", line 279, in close
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 303, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
**error: [Errno 32] Broken pipe**
----------------------------------------

I have good internet connection in the server and client side so, why could be happening?

Thanks indeed and best regards

ha7ilm, 2016-07-14:
Hi Raul,

Thank you for running a public receiver!

> I'm having a problem with the server online at http://www.merlitec.com:18080, after some minutes after opening a client, it drops down with this server output:

If you stop the waterfall like this (in the Google Chrome developer console, opened with Ctrl+Shift+J):

    waterfall_setup_done=0

...does this still happen?
If so, can you see any error messages in the developer console?

Which browser version are you running, on which platform?

73!

Andras

raulmerlin, 2016-07-15:
Hi Andras!

You are right asking about the browser. It only happens running Chrome, with Firefox I have no problem and the client can be opened for hours* without any issue.
I will try with the Chrome console as you said, trying to reach the Chrome reason.

*As said in another issue, having the waterfall for hours consumes a lot of RAM on the client, but it is easy to solve and not crytical for my use case :)

Thanks for your interest and help

73!

raulmerlin, 2016-07-19:
Hello again!

Bad news, I tried and the console shows nothing, simply it gets closed. I took a screenshot:

[http://www.merlitec.com/comp/openwebrxchrome.png](url)

At the moment I am really happy using Firefox, but I will have and eye on it I can get the root problem.

Thanks again, best regards!

ha7ilm, 2016-07-24:
Well, when the browser itself crashes, it is hard to debug: we should not be able to crash Chrome from Javascript. 

Maybe an update to Chrome will solve it, or you could try it with a fresh build of Chromium.

ha7ilm, 2016-07-24:
Some years ago my Chrome also crashed from OpenWebRX. The solution was to upgrade to the latest version. 

(Unfortunately, web apps that use a lot of relatively new features like Web Audio API, WebSockets, WebGL, etc. are more likely to crash the browser than the ones that are only displaying a regular page with text and images.)