ha7ilm/openwebrx #44: No Audio on OS X

mehdideveloper, 2016-08-02:
Hi
I'm trying to run the latest openwebrx with RTL-SDR and I've no errors and I can see the spectrum, but no audio!
The only audio-related log in Chrome's developer tools is this: 

> audio_calculate_resampling() :: 4, 11025

Also here's the first lines of log when I run openwebrx:

> [openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
Ncat: Version 7.12 ( https://nmap.org/ncat )
[openwebrx-main] Started rtl_thread: rtl_sdr -s 1800000 -f 88000000 -p 0 -g 5 -| ncat -4l 4951 -k --send-only --allow 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Using device 0: Generic RTL2832U OEM
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = ncat -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 200000 | csdr logpower_cf -70 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-spectrum] Spectrum thread started.
Ncat: Version 7.12 ( https://nmap.org/ncat )
Ncat: Connected to 127.0.0.1:4951.
Found Rafael Micro R820T tuner
Sampling at 1800000 S/s.
Tuned to 88000000 Hz.
Tuner gain set to 3.70 dB.
Reading samples in async mode...
[openwebrx-main] Starting HTTP server.

I've tried changing "audio_compression" to "none" and also bigger values for "client_audio_buffer_size" without success!

When I close my browser, these errors are shown in the terminal:

> [openwebrx-httpd] error in dsp.stop()
[openwebrx-httpd] client 127.0.0.1#20559ac5c768cd7410e1a4d1bdd53dc1 :: client being closed.
[openwebrx] close_client dsp.stop() :: error - <type 'exceptions.AttributeError'> dsp_plugin instance has no attribute 'process'
  File "openwebrx.py", line 378, in close_client
    clients[i].dsp.stop()
  File "/Users/mehdi/Documents/Radio_Electronics/Source/openwebrx/plugins/dsp/csdr/plugin.py", line 215, in stop
    os.killpg(os.getpgid(self.process.pid), signal.SIGTERM)
Exception happened during processing of request from ('127.0.0.1', 59977)
Traceback (most recent call last):
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 599, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 657, in __init__
    self.finish()
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 716, in finish
    self.wfile.close()
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/socket.py", line 283, in close
    self.flush()
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/socket.py", line 307, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
**error: [Errno 32] Broken pipe**

My system info:
OS X 10.11.6
Chrome 52.0.2743.82

Any ideas?

ha7ilm, 2016-08-02:
Hi Mehdi,

If the spectrum works then `ncat˛` works at least.

Now I would need a line from the console output, the command that OpenWebRX starts when a new client attaches (includes `fir_decimate_cc˙, etc.)

mehdideveloper, 2016-08-03:
Hi
Unfortunately I don't see "fir_decimate_cc", neither in Web Developer Console, nor in the Terminal output!
The only log that I see when a client is attached:

> [openwebrx-httpd] client 127.0.0.1#c463f561225cc7631b4fb10f6e599477 :: client added. Clients now: 1
127.0.0.1 - - [03/Aug/2016 11:42:08] "GET /openwebrx.css HTTP/1.1" 200 -
127.0.0.1 - - [03/Aug/2016 11:42:08] "GET /openwebrx.js HTTP/1.1" 200 -
127.0.0.1 - - [03/Aug/2016 11:42:08] "GET /sdr.js HTTP/1.1" 200 -
........

ha7ilm, 2016-08-05:
I'm looking for something like this:

    [openwebrx-dsp-plugin:csdr] Command = ncat -v 127.0.0.1 4951 | csdr convert_u8_f | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_3059136684_shift | csdr fir_decimate_cc 22 0.0068178 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_3059136684_bpf 0.0281615770483 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_3059136684_squelch --outfifo /tmp/openwebrx_pipe_3059136684_smeter 5 1 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.03065759637 | csdr deemphasis_nfm_ff 11025 | csdr fastagc_ff 1024 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8

If this is stuck somewhere, you will get no audio.

mehdideveloper, 2016-08-06:
Well the only similar thing that I have in the log is this:

> [openwebrx-dsp-plugin:csdr] Command = ncat -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 200000 | csdr logpower_cf -70 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096

There is no other line that starts with "[openwebrx-dsp-plugin:csdr]"

ha7ilm, 2016-08-06:
Strange. Then the processes to generate audio from the I/Q data do not even get opened.

ha7ilm, 2016-08-06:
In addition, `dsp.stop()` should be called when the client disconnects.

Could you share the whole console output with me?
If you do not want to paste it here, you can send it to my e-mail address, which can be found on my GitHub profile.

mehdideveloper, 2016-08-06:
Thanks. I emailed you the whole log

ha7ilm, 2016-08-11:
Hi Mehdi,

Thank you for the log! It seems like `dsp.start()` is not executed at all... Unfortunately, I don't think I can debug this without a Mac.

mehdideveloper, 2016-10-26:
Hey Simonyi
It has passed a while and I haven't been able to solve this issue.
Could you please check this once again? (maybe this time you can find or think of some hints)

* I'm on Mac OS X 10.12.1
* I've compiled the latest csdr (although by changing some parameters in Makefile)
* csdr-fm runs fine and plays from the attached dongle
* I can see the spectrum (without any errors in the browser console, or python shell)

Any ideas?
Thanks