ha7ilm/openwebrx #45: Poor reception?

ways, 2016-08-19:
Hi

I'm running this on a raspberry pi 2 (have also tried on 1) with raspbian and an rtl-sdr dongle. When running on my laptop with gqrx the dongle works fine, but when running on the pi with openwebrx I get the waterfall and noise, but no reception. Can't even hear local radio stations. Any idea why?

ha7ilm, 2016-08-19:
Hi @ways!

Make sure that you are using the same gain setting (and sample rate) for OpenWebRX and Gqrx. 

For OpenWebRX, the default setting is manual gain around 5 dB (`rf_gain=5` in `config_webrx.py`). You can set it to auto gain by `rf_gain=0`. 

The default sample rate is 250 ksps, which will allow you to see 250 kHz bandwidth on the waterfall diagram.

Best regards,

Andras

ways, 2016-08-23:
Ok, `rf_gain=0` got me some more noise in the graph instead of just white noise. But not much better. Limiting the bandwidth also did something, but still unusable.

ways, 2016-08-23:
Is the dev branch better?

ha7ilm, 2016-08-23:
No, the `dev` branch is outdated now.

ha7ilm, 2016-08-23:
Maybe you could check what happens if you run `rtl_tcp` on the Pi, and connect to it via `gqrx` on the laptop. Does it have better performance (with the same settings) or the same as OpenWebRX?

ha7ilm, 2016-08-23:
Also you could try to run OpenWebRX on the laptop and see if the results are different than running it on the Pi. 

In theory, the algorithms behind OpenWebRX should not perform worse than other SDR software in terms of reception quality. I think it is more likely that some noise is introduced by the RPi power supply, or something.

leastonce, 2016-08-23:
I run an airspy on a pi 3 and it's pretty decent at 2.5Mbps. The audio gets a bit choppy when changing frequency but it's no worse than gqrx on my Mac.

ways, 2016-08-24:
I'll see if I can get hold of a Pi3. if not, I'll try it on a laptop.

Techinput, 2016-11-25:
I've got my Pi3 and my RTL2832U & HackRF but both give a very weak (not hearable) signal, I've tweaked with the gain for both radio's but no success for even "loud" local repeater/broadcast signals.

ha7ilm, 2016-11-25:
Is the noise weak as well, or it only happens when you tune to a strong signal?

Techinput, 2016-11-25:
Spectrum is the same, doesn't matter where. Known strong signals are very tiny visible.

Techinput, 2016-11-29:
[image]
This is the signal from my HackRF/PortaPack so with a gain of 40 on OpenWebRX I see the signal however I cannot have the correct modulation (which should be NFM/FM for this local repeater). It gives sound but nothing remote to voice?

ha7ilm, 2016-11-29:
@Techinput Do you hear the noise if you tune to anywhere else?

Techinput, 2016-11-29:
I hear the normal noise and I hear the other "noise" for the station that I see