ha7ilm/openwebrx #47: recording?

vielmetti, 2016-09-27:
Hi and thanks for this software.

I am wondering if there are capabilities and plans for making recordings using this system. The WedSDR software at U Twente, for example, has this:

[image]

[image]

a simple one-button recorder that stores an audio file on the system for download.

I know that there are various ways to enable recording by doing a virtual patch of the received audio into recording software, but thought it would be worth raising the question for a built-in feature.

thanks again

ha7ilm, 2016-09-27:
Hi @vielmetti,

> thought it would be worth raising the question for a built-in feature.

Thanks for the idea. 
Maybe, later. 
I've been working on multiple things in parallel for some time and I'd rather have to finish them.

30rj076, 2019-08-30:
Any news since then... 27 Sep 2016???

bclswl0827, 2019-08-30:
Unfortunately, I am busy with the exam now. If there is time, I am willing to try.