ha7ilm/openwebrx #48: Is it possible to increase the bandwidth? Coverage more of a band?

milesburton, 2016-10-05:
Afternoon,

Open Web RX is a fantastic application. I'd like to use to use it to cover the entire UK 2M band.

Is it possible to increase the bandwidth?

DH4CK, 2016-10-05:
Hi,

sure it is possible. But it depends on your sdr radio interface and your computers horsepower.
I am using rtlsdr with 820t tuner and it is no problem to cover the entire 2 m band from 144 to 146 MHz.
You have to set the samplerate to 2000000 (2 MHz) and center frequency to 145000000 (145 MHz). You can set the samplerate even higher, but in my experience with values of more than 2.5 MHz some samples might get dropped on the pi - it might get choppy and 2 MHz is just what you need.

Here are some lines of my config_webrx.py:

```
dsp_plugin="csdr"
fft_fps=5
fft_size=8192
samp_rate = 2000000
center_freq = 145000000
rf_gain = 33 #in dB. For an RTL-SDR, rf_gain=0 will set the tuner to auto gain mode, else it will be in manual gain mode.

```
you can see I also use an fft_size of 8192 for a finer resolution than the default 4096, but also reduced the fft_fps to 5 to reduce the bandwidth per client. its now around 200 kBit/s for one client, which is okay.

you can visit my sdr here: [hamsatsdr.ddns.net:8073](http://hamsatsdr.ddns.net:8073)

good luck, 73s Hauke DH4CK

milesburton, 2016-10-05:
Excellent, you've just made OpenWebWRX significantly more useful. Hoping to use it for our local club. 73's M0LHA

closed 2016-10-05 by milesburton

ha7ilm, 2016-10-05:
Exactly as @DH4CK said. Thank you, Hauke!