ha7ilm/openwebrx #52: Problem with USRP

stephensworld, 2016-11-13:
Hi,

can someone help me?
I would like to use Ettus B100 with OpenWebRx.

[image]

Thanks
Stephan

ha7ilm, 2016-11-13:
Hi Stephan,

Could you check it with different sample rates?

(Unfortunately I do not know a list of correct sampling rate settings for the USRP1, but you might find it in other software, etc.)

In addition I've made a simple flowgraph for You to check if the osmocom source block works at all: 
[test1.grc.zip](https://github.com/simonyiszk/openwebrx/files/587853/test1.grc.zip)

Cheers,
Andras

stephensworld, 2016-11-13:
Hi Andras,

thank you very much for the answer.
[image]

Same error...

ha7ilm, 2016-11-13:
Okay. Then it might not be the window size of a filter or something. It rather looks the WX GUI window size... :smile: Check the Options block. If you can't find the problematic part, post the screenshot here.

ha7ilm, 2016-11-13:
If there is a "window size" option over there, you should enter a value like that:

    800, 800

ha7ilm, 2016-11-13:
Okay, that's the way to fix it:

[image]

stephensworld, 2016-11-13:
Thank you very much :-)

stephensworld, 2016-11-13:
I get this out:
`python osmocom_source.py
linux; GNU C++ version 4.8.2; Boost_105400; UHD_003.005.005-0-unknown

gr-osmosdr 0.1.1 (0.1.1) gnuradio 3.7.2.1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace 
-- USRP-B100 clock control: 10
--   r_counter: 2
--   a_counter: 0
--   b_counter: 20
--   prescaler: 8
--   vco_divider: 5
--   chan_divider: 5
--   vco_rate: 1600.000000MHz
--   chan_rate: 320.000000MHz
--   out_rate: 64.000000MHz
-- 
-- Using subdev spec 'A:AB'.
Press Enter to quit: 
`

and 

`python openwebrx.py 

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
Ncat: Version 6.40 ( http://nmap.org/ncat )
[openwebrx-main] Started rtl_thread: cat /tmp/osmocom_fifo| ncat -4l 4951 -k --send-only --allow 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = ncat -v 127.0.0.1 4951 | csdr fft_cc 4096 2777 | csdr logaveragepower_cf -70 4096 10 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-main] Starting HTTP server.
[openwebrx-spectrum] Spectrum thread started.
Ncat: Version 6.40 ( http://nmap.org/ncat )
Ncat: Connected to 127.0.0.1:4951.

`

stephensworld, 2016-11-13:
Now I can here the noise. I had to change the path in the file_sink to /tmp/osmocom_fifo