ha7ilm/openwebrx #53: Receiver not listed on SDR.hu

misterdevil, 2016-12-07:
Hi i've tried on your openwebrx project and it's awesome. I've registered my receiver but got some problems when i tried to list my receiver on sdr.hu, my receiver cannot reached from public internet.
Here's my settings

On config:
`web_port=8088
server_hostname="222.124.124.12" # If this contains an incorrect value, the web UI may freeze on load (it can't open websocket)
max_clients=5

# ==== sdr.hu listing ====
# If you want your ham receiver to be listed publicly on sdr.hu, then take the following steps:
# 1. Register at: http://sdr.hu/register
# 2. You will get an unique key by email. Copy it and paste here:
sdrhu_key = "copied like email"
# 3. Set this setting to True to enable listing:
sdrhu_public_listing = True
On config:
web_port=8088
server_hostname="222.124.124.12" # If this contains an incorrect value, the web UI may freeze on load (it can't open websocket)
max_clients=5

# ==== sdr.hu listing ====
# If you want your ham receiver to be listed publicly on sdr.hu, then take the following steps:
# 1. Register at: http://sdr.hu/register
# 2. You will get an unique key by email. Copy it and paste here:
sdrhu_key = "copied like email"
# 3. Set this setting to True to enable listing:
sdrhu_public_listing = True
`
On terminal:
`
[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
Ncat: Version 6.40 ( http://nmap.org/ncat )
[openwebrx-main] Started rtl_thread: rtl_sdr -s 2400000 -f 93700000 -p 0 -g 10 -| ncat -4l 4951 -k --send-only --allow 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = ncat -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 8192 5673 | csdr logaveragepower_cf -70 8192 47 | csdr fft_exchange_sides_ff 8192 | csdr compress_fft_adpcm_f_u8 8192
[openwebrx-main] Starting sdr.hu update thread...
[openwebrx-main] Starting HTTP server.
 [openwebrx-spectrum] Spectrum thread started.
Ncat: Version 6.40 ( http://nmap.org/ncat )
Ncat: Connected to 127.0.0.1:4951.
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2400000 S/s.
Tuned to 93700000 Hz.
Tuner gain set to 8.70 dB.
Reading samples in async mode...
[openwebrx-sdrhu] Update succeeded! (Your receiver is soon getting listed on sdr.hu!)`

but after 10-15minutes still not listed on sdr.hu , i try to check http://222.124.124.12:8088/status and it cannot reached from public internet.

where is wrong with my setup? thanks for your help.

mdmbc, 2016-12-07:
Have you forwarded the correct port on your router?

misterdevil, 2016-12-07:
@mdmbc ah i forgot it, i'll try to setup it for now. thanks brother

misterdevil, 2016-12-08:
Hi my sdr has bee listed on sdr.hu but it cannot opened from public ip, only can opened from lan ip. I try to port check the status is "Success!   I can see your service on 222.124.124.12 on port (8078) 
Your ISP/Router/Firewall is not blocking port 8078. " 

is there something need to add? 

maybe @mdmbc can help me ?

mdmbc, 2016-12-08:
I am able to access your SDR using the IP and port above, so it looks like you're configured correctly. Depending on your router, it's likely that you're not going to be able to access things on your local network using your public IP. The traffic never leaves your local network, so isn't going to come back through NAT in your router.

misterdevil, 2016-12-08:
@mdmbc ah i see, i just checked from other network and it can. thank you so much for your help answer mike

closed 2016-12-08 by misterdevil