ha7ilm/openwebrx #58: Question about setting up multiple receivers on the same machine

misterdevil, 2017-01-04:
Hi i've tried to set multiple receivers. I've made copy for openwebrx-1. 
I've set web_port in config_webrx.py different
I've set listening port different too at 4951 for openwebrx and 4952 for openwebrx-1
I use 1 rtl sdr device so I dont edit start_rtl_command=
And then I run python openwebrx.py for each copy, but only 1 that can connect to ncat, the other one refused.

In terminal said:
[openwebrx-main] Starting HTTP server.
[openwebrx-spectrum] Spectrum thread started.
Ncat: Version 6.40 ( http://nmap.org/ncat )
Ncat: Connection refused

Where is wrong with my setting ? Thanks for the help

ha7ilm, 2017-01-04:
Could you post the full terminal output?

misterdevil, 2017-01-04:
[image]
[image]

ha7ilm, 2017-01-04:
You cannot use the same RTL-SDR device with two software instances running simultaneously.

You will need at least two different hardware devices for that.

misterdevil, 2017-01-05:
I see, thanks @ha7ilm