ha7ilm/openwebrx #64: help my

markusgraj, 2017-02-07:
Hello who will help me configure it all I rtl 2838u, I can not configure this thank Marek

ha7ilm, 2017-02-07:
Hi,

Can you tell more about the error messages you get?

markusgraj, 2017-02-07:
sorry,the receiver is inactive due to internal error

ha7ilm, 2017-02-07:
And on the terminal?

markusgraj, 2017-02-07:
vitrtual box

ha7ilm, 2017-02-07:
I mean, the console output of the `openwebrx` command.

markusgraj, 2017-02-07:
give me the email address of your

markusgraj, 2017-02-07:
I will give you the data and check yourself

markusgraj, 2017-02-07:
I sent you a message

ha7ilm, 2017-02-08:
Moved to e-mail.

closed 2017-02-08 by ha7ilm

markusgraj, 2017-02-08:
> Wysłane: 8 lutego 2017 09:03
> Do: simonyiszk/openwebrx
> DW: markusgraj; Author

hi Do I have to have installed python 2.7 on ubuntu

ha7ilm, 2017-02-08:
Yes, you need at least python 2.7.6. I didn't try with older versions.

markusgraj, 2017-02-10:
Thank you for your help, everything works, tell me how to change the picture in openweb

Wysłane: 8 lutego 2017 17:51

ha7ilm, 2017-02-10:
Replace `htdocs/gfx/openwebrx-top-photo.jpg` with your own.

markusgraj, 2017-02-10:
thank you very much

Wysłane: 10 lutego 2017 13:48

markusgraj, 2017-02-12:
Hi I'm part of the problem server cpu 100% I have something set wrong, it is a win 10, cpu intel pentium 2.6, 4GB RAM thank you for helping Marek

ha7ilm, 2017-02-12:
I think it is possible on a single core Intel Pentium that you use up all the CPU time. You can try these tips though:

https://github.com/simonyiszk/openwebrx/wiki/Optimizing-OpenWebRX-for-speed

markusgraj, 2017-02-12:
HI  of my question, my sdr is shifted by 2 khz as you can to correct the thank Marek

Wysłane: 12 lutego 2017 15:52

https://github.com/simonyiszk/openwebrx/wiki/Optimizing-OpenWebRX-for-speed

[https://avatars0.githubusercontent.com/u/5983248?v=3&s=400]<https://github.com/simonyiszk/openwebrx/wiki/Optimizing-OpenWebRX-for-speed>

Optimizing OpenWebRX for speed - GitHub<https://github.com/simonyiszk/openwebrx/wiki/Optimizing-OpenWebRX-for-speed>
github.com
DSP CPU usage can be fine-tuned in plugins/dsp/csdr/plugin.py: you can set transition bandwidths higher (thus degrade filter performance by decreasing the length of ...

ha7ilm, 2017-02-13:
https://github.com/simonyiszk/openwebrx/wiki/Correcting-the-frequency-offset

ha7ilm, 2017-02-13:
...or you can also do: 

    shown_center_freq = center_freq - 2000

markusgraj, 2017-02-14:
Thank you for your help

Wysłane: 13 lutego 2017 08:43

markusgraj, 2017-02-20:
Hello I have a question I signed up on the website with you, if I can rearrange all the data on the program (ha5kfu at my 2os073) and the rest?

ha7ilm, 2017-02-20:
You can do anything with the web interface, as long as you comply with the [license](https://github.com/simonyiszk/openwebrx/blob/master/LICENSE.txt).

You can replace the OpenWebRX and the HA5KFU logo if you want.

To comply with the AGPL license, you need to distribute the modified source code to those who you give access to your receiver.

markusgraj, 2017-02-21:
hi what program you can edit the file WRX

ha7ilm, 2017-02-21:
Any text editor, like `gedit`, `nano`, `vim`.