ha7ilm/openwebrx #65: program cannot find HackRF one

BustedSec, 2017-02-21:
I tried changing the receiver_device to HackRF in the config_webrx.py but get the following error

File "openwebrx.py", line 691, in <module>
    main()
  File "openwebrx.py", line 120, in main
    cfg=__import__("config_webrx" if no_arguments else sys.argv[1])
  File "/home/havok/openwebrx/config_webrx.py", line 48
    receiver_device=""HackRF"

how can I get it to recognize my device? 

Linux mini 4.4.0-63-generic #84-Ubuntu SMP Wed Feb 1 17:20:32 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

BustedSec, 2017-02-21:
I am an idiot - nevermind

closed 2017-02-21 by BustedSec

ha7ilm, 2017-02-21:
First, try to change 

    receiver_device=""HackRF"

to

    receiver_device="HackRF"

Andras