ha7ilm/openwebrx #66: please help with sample settings

BustedSec, 2017-02-21:
to tune in normal FM for testing purposes, what should these settings be for a hackRF in the US

samp_rate = 25000000
#center_freq = 145525000
center_freq =  87900000
rf_gain = 5 #in dB. For an RTL-SDR, rf_gain=0 will set the tuner to auto gain m$
ppm = 0

BustedSec, 2017-02-21:
Trying to tune in local FM radio stations, all I'm getting is static despite a nice waterfall visualizaion of the data. I've tried changing rf_gain to 0 and adjusting the center frequency and sample rate. How can I just tune this thing to local FM and have it play some music?

ha7ilm, 2017-02-21:
You can't. Currently OpenWebRX does not support WFM. 

Look for NFM, AM or SSB transmissions.

ha7ilm, 2017-02-21:
I mean:

* NFM: narrow bandwidth FM, ~10 kHz BW. Used by communications equipment, like handheld transceivers (e.g. ham FM transceivers, PMR).
* WFM: wide-band FM, ~100 kHz BW. Used by broadcast FM stations around 88-108 MHz (at least in Hungary).

RaymondDay61, 2018-04-05:
I get this too. I know it can do it because with another program I can just do this command line and it sounds good.

rtl_fm -M wbfm -f 104.3M | play -r 32k -t raw -e s -b 16 -c 1 -V1 -

Been edit the /root/openwebrx/config_webrx.py and restating and nothing helps

In that file here are some setting that I mostly change.

# samp_rate = 250000
samp_rate = 2400000
center_freq = 91700000
rf_gain = 0 #in dB. For an RTL-SDR, rf_gain=0 will set the tuner to auto gain mode, else it will be in manual gain mode.
ppm = 0

It be neat if could get this working good. I seen others that open theirs up on the web and they work real good.

I can not zoom out any it's just the one screen. I guess that is some settings too. So I can only see like 3 radio stations then.

-Raymond Day

ha7ilm, 2018-04-05:
Yes, this does not work because WFM is currently unsupported.

RaymondDay61, 2018-04-06:
Links a http://sdr.hu/ show openwebrx working with all sorts of wireless can zoom way out and when you click on a line it don't sound like static. Seems like supported on there systems.

ha7ilm, 2018-04-06:
So actually what parameters does the signal have that you want to listen to?
E.g. modulation, bandwidth?

ha7ilm, 2018-04-06:
See: 

* https://github.com/simonyiszk/openwebrx/wiki/Using-RTLSDR-in-direct-sampling-mode-with-OpenWebRX
* https://github.com/simonyiszk/openwebrx/wiki

RaymondDay61, 2018-04-06:
I all ready did that at the 1st link you gave. Maybe I did some thing wrong.

I been to the wiki too. But I have not read it all yet. I will look at it more. Thanks.

-Raymond Day

RaymondDay61, 2018-04-06:
Here is what mine looks like if it don't support FM then why is it a click part on it? I can't do AM because it's only in the little rang of mhz I don't get how to widened that to make it wader. I guess it's some line in the config_webrx.py

A picture is worth a 1000 words so this my show it with out me typing a 1000 words :)

[image]

-Raymond Day

RaymondDay61, 2018-04-06:
Would it be hard to get it to support FM? I guess AM is easier. I guess you now can make a wireless signal just by torching 2 wires together with power to make a spark.  AM adjust the power of the spark. But FM it's the timing of the on and off of the spark. So seems like it would not be that hard to get FM to work in the OpenWebRx not sure.

-Raymond Day

ha7ilm, 2018-04-06:
No, broadcast FM (WFM) does not work currently, while NFM does. (Contact me if you need this feature in a commercial project.)

>But FM it's the timing of the on and off of the spark.

Today we are not using spark gap transmitters anymore. You can find more information about how an FM signal looks like here: https://en.wikipedia.org/wiki/Frequency_modulation

RaymondDay61, 2018-04-10:
I gave up just can't get it to scroll left or right to see other frequencies no setting to widen it out that I can find. All I can do it set the mid number and stop and start it and it starts at that place and maybe can see 2 or 3 other frequencies on one screen that can't move left or right.

All so still sounds like junk static. But know it's is some sound under the static.