ha7ilm/openwebrx #67: No waterfall, no audio in chromium browser in Raspberry pi3

tikiarya, 2017-03-13:
Thanks for such a interesting SDR framework.i am not sure if we have a blog/forum to share difficulties related OpenWebRX. I tried to find solution to my problem but could not get in net. 

My setup : Raspberry Pi 3, OS: RASPBIAN JESSIE WITH PIXEL, Nooelec Nano 2+. My system works well with GQRX with RTL SDR.
I have followed http://blog.sdr.hu/2015/06/30/quick-setup-openwebrx.html line by line for my setup and did not update config_webrx.py , after running openwebrx.py , i have open chromium browser and typed http://localhost:8073 
The interface is coming nicely but there is no waterfall and no audio. There is no failure message in terminal(LXTerminal in Raspberry Pi, i would provide the complete message dump soon). Please help me to get going. Thanks in advance.

ha7ilm, 2017-03-13:
Thank you for your report!

> i am not sure if we have a blog/forum to share difficulties related OpenWebRX. 

This is the good place. 

> There is no failure message in terminal

Someone else mentioned a segmentation fault, so you could look for that among the messages.

Yes, please provide the full console output.

tikiarya, 2017-03-13:
yes appears segmentation fault. here is complete dump 
pi@raspberrypi:~ $ cd openwebrx
pi@raspberrypi:~/openwebrx $ sudo ./openwebrx.py

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] nmux_bufsize = 65536, nmux_bufcnt = 763
[openwebrx-main] Started rtl_thread: rtl_sdr -s 250000 -f 145525000 -p 0 -g 5 -| nmux --bufsize 65536 --bufcnt 763 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
[openwebrx-main] I/Q server started.
nmux: pthread_create() done, clients now: 1
client 0x1432298: started!
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 2777 | csdr logaveragepower_cf -70 4096 10 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-main] Starting HTTP server.
[openwebrx-spectrum] Spectrum thread started.
nmux: pthread_create() done, clients now: 2
client 0x15b24c0: started!
Connection to 127.0.0.1 4951 port [tcp/*] succeeded!
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sampling at 250000 S/s.
Tuned to 145525000 Hz.
Tuner gain set to 3.70 dB.
Reading samples in async mode...
client 0x1432298: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32127.0.0.1 - - [13/Mar/2017 15:46:22] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client 127.0.0.1#c8bf3c896d628056dc61c9e322c8be4f :: client added. Clients now: 1
127.0.0.1 - - [13/Mar/2017 15:46:22] "GET /openwebrx.js HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:22] "GET /openwebrx.css HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:22] "GET /sdr.js HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:22] "GET /gfx/openwebrx-top-photo.jpg HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:22] "GET /gfx/openwebrx-top-logo.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:22] "GET /gfx/openwebrx-ha5kfu-top-logo.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:22] "GET /gfx/openwebrx-avatar-background.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:22] "GET /gfx/openwebrx-avatar.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:22] "GET /gfx/openwebrx-rx-details-arrow-up.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-rx-details-arrow.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-panel-status.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-panel-log.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-panel-receiver.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-speaker.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-waterfall-auto.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-squelch-button.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-waterfall-default.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-zoom-in.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-zoom-out.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-zoom-in-total.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-zoom-out-total.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-play-button.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-background-cool-blue.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/openwebrx-scale-background.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 15:46:23] "GET /gfx/font-expletus-sans/ExpletusSans-Medium.ttf HTTP/1.1" 200 -
[openwebrx-httpd:ws,0] command: SET output_rate=12000 action=start
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_1986862576_shift | csdr fir_decimate_cc 20 0.0075 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_1986862576_bpf 0.0256 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_1986862576_squelch --outfifo /tmp/openwebrx_pipe_1986862576_smeter 5 1 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.04166666667 | csdr deemphasis_nfm_ff 11025 | csdr fastagc_ff 1024 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
csdr: fifo control mode on
fractional_decimator_ff: use_prefilter = 0, num_poly_points = 12, transition_bw = 0.03, window = HAMMING
fractional_decimator_ff: not using taps
Segmentation fault
csdr: fifo control mode on
shift_addition_cc: reinitialized to -0
nmux: pthread_create() done, clients now: 2
Connection to 127.0.0.1 4951 port [tcp/*] succeeded!
bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 157) + (input_size = 356) - 1
(overlap_length = 156) = taps_length - 1
client 0x17b26c8: started!
csdr: fifo control mode on
fir_decimate_cc: taps_length = 533
taps_length = 533
padded_taps_length = 536
taps = 1466018
taps = 1466020
squelch_and_power_cc: initial squelch level is 0
bandpass_fir_fft_cc: filter initialized, low_cut = -0.32, high_cut = 0.32

ha7ilm, 2017-03-13:
This way we would get the source of the segfault:
```
export LD_PRELOAD=libSegFault.so
python2 openwebrx.py
```
Could you try it?

tikiarya, 2017-03-13:
many thanks Andras

ha7ilm, 2017-03-13:
Er... That fixed the segfault? That's not expected. 

The expected output is more data in the terminal, including the address of the program counter when the segfault happened, and a stack trace.

Anyway, I'll try that on a Pi myself. What is the version of your kernel? 
```
uname -a
```

mprilepok, 2017-03-13:
H Andrasi, on Raspberry Pi 3 also the interface is coming nicely but there is no waterfall and no audio. When I execute the commmand export LD_PRELOAD=libSegFault.so firstly and the python2 openwebrx.py, it works.

pi@sdrpi:~/openwebrx $ uname -a
Linux sdrpi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux

Here is the console output:
pi@sdrpi:~ $ cd openwebrx/
pi@sdrpi:~/openwebrx $ python2 openwebrx.py

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] nmux_bufsize = 65536, nmux_bufcnt = 763
[openwebrx-main] Started rtl_thread: rtl_sdr -s 250000 -f 145525000 -p 0 -g 5 -| nmux --bufsize 65536 --bufcnt 763 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000036

Using device 0: Generic RTL2832U OEM
[openwebrx-main] I/Q server started.
nmux: pthread_create() done, clients now: 1
client 0xbb3298: started!
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 2777 | csdr logaveragepower_cf -70 4096 10 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-main] Starting HTTP server.
[openwebrx-spectrum] Spectrum thread started.
nmux: pthread_create() done, clients now: 2
client 0xd334c0: started!
Connection to 127.0.0.1 4951 port [tcp/*] succeeded!
Found Elonics E4000 tuner
Exact sample rate is: 250000.000414 Hz
Sampling at 250000 S/s.
Tuned to 145525000 Hz.
Tuner gain set to 4.00 dB.
Reading samples in async mode...
client 0xbb3298: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32192.168.6.229 - - [13/Mar/2017 21:52:32] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client 192.168.6.229#2875be0403fbeb1ee1a493e4c20234cc :: client added. Clients now: 1
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /openwebrx.css HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /openwebrx.js HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /sdr.js HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-top-photo.jpg HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-top-logo.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-ha5kfu-top-logo.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-avatar-background.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-avatar.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-rx-details-arrow-up.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-rx-details-arrow.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-panel-status.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-panel-log.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-panel-receiver.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-speaker.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-waterfall-auto.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-squelch-button.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-waterfall-default.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-zoom-in.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-zoom-out.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-zoom-in-total.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-zoom-out-total.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-play-button.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/font-expletus-sans/ExpletusSans-Medium.ttf HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-background-cool-blue.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:32] "GET /gfx/openwebrx-scale-background.png HTTP/1.1" 200 -
192.168.6.229 - - [13/Mar/2017 21:52:33] "GET /favicon.ico HTTP/1.1" 200 -
[... 9 lines cut ...]
squelch_and_power_cc: initial squelch level is 0
client 0xf336c8: started!
bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 143) + (input_size = 370) - 1
(overlap_length = 142) = taps_length - 1
taps_length = 587
padded_taps_length = 592
taps = 83b018
taps = 83b020
fractional_decimator_ff: use_prefilter = 0, num_poly_points = 12, transition_bw = 0.03, window = HAMMING
fractional_decimator_ff: not using taps
Segmentation fault
bandpass_fir_fft_cc: filter initialized, low_cut = -0.35202, high_cut = 0.35202
[openwebrx-spectrum] client spectrum queue full, closing it.
[openwebrx-httpd] client 192.168.6.229#2875be0403fbeb1ee1a493e4c20234cc :: client being closed.
[openwebrx-httpd:ws] client closed by other thread
[... cut ...]
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_1987374216_squelch
[openwebrx-dsp-plugin:csdr] stop() :: unlink failed: /tmp/openwebrx_pipe_1987374216_smeter
[openwebrx-httpd] client cannot be closed:  <class '__main__.ClientNotFoundException'>
  File "openwebrx.py", line 586, in do_GET
    id_to_close=get_client_by_id(myclient.id,False)
  File "openwebrx.py", line 344, in get_client_by_id
    raise ClientNotFoundException
client 0xf336c8: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32

tikiarya, 2017-03-13:
the kernel is 4.4.50-v7+ #970 , what i saw after openwebrx.py execution, if i close the program by CNTRL+C in terminal, close the client browser connection and try to open config_webrx.py, it does not open and stucks.
--------------------------------------------------------

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
pi@raspberrypi:~ $ export LD_PRELOAD=libSegFault.so
pi@raspberrypi:~ $ cd openwebrx
pi@raspberrypi:~/openwebrx $ python2 openwebrx.py

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] nmux_bufsize = 65536, nmux_bufcnt = 763
[openwebrx-main] Started rtl_thread: rtl_sdr -s 250000 -f 123525000 -p 0 -g 5 -| nmux --bufsize 65536 --bufcnt 763 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
nmux: pthread_create() done, clients now: 1
client 0xd8c298: started!
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 2777 | csdr logaveragepower_cf -70 4096 10 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-main] Starting HTTP server.
[openwebrx-spectrum] Spectrum thread started.
nmux: pthread_create() done, clients now: 2
client 0xf0c4c0: started!
Connection to 127.0.0.1 4951 port [tcp/*] succeeded!
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sampling at 250000 S/s.
Tuned to 123525000 Hz.
Tuner gain set to 3.70 dB.
Reading samples in async mode...
client 0xd8c298: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32127.0.0.1 - - [13/Mar/2017 20:43:40] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client 127.0.0.1#206448539b606feebec63756801fd29d :: client added. Clients now: 1
127.0.0.1 - - [13/Mar/2017 20:43:40] "GET /openwebrx.js HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:40] "GET /openwebrx.css HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:40] "GET /sdr.js HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:40] "GET /gfx/openwebrx-top-photo.jpg HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:40] "GET /gfx/openwebrx-top-logo.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:40] "GET /gfx/openwebrx-ha5kfu-top-logo.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:40] "GET /gfx/openwebrx-avatar-background.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:40] "GET /gfx/openwebrx-avatar.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-rx-details-arrow-up.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-rx-details-arrow.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-panel-status.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-panel-log.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-panel-receiver.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-speaker.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-waterfall-auto.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-squelch-button.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-waterfall-default.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-zoom-in.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-zoom-out.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-zoom-in-total.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-zoom-out-total.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:41] "GET /gfx/openwebrx-play-button.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:42] "GET /gfx/openwebrx-scale-background.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:42] "GET /gfx/openwebrx-background-cool-blue.png HTTP/1.1" 200 -
127.0.0.1 - - [13/Mar/2017 20:43:42] "GET /gfx/font-expletus-sans/ExpletusSans-Medium.ttf HTTP/1.1" 200 -
[openwebrx-httpd:ws,0] command: SET output_rate=12000 action=start
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_1990009120_shift | csdr fir_decimate_cc 20 0.0075 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_1990009120_bpf 0.0256 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_1990009120_squelch --outfifo /tmp/openwebrx_pipe_1990009120_smeter 5 1 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.04166666667 | csdr deemphasis_nfm_ff 11025 | csdr fastagc_ff 1024 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
nmux: pthread_create() done, clients now: 2
csdr: fifo control mode on
csdr: fifo control mode on
client 0x110c6c8: started!
csdr: fifo control mode on
[... cut ...]
fractional_decimator_ff: use_prefilter = 0, num_poly_points = 12, transition_bw = 0.03, window = HAMMING
fractional_decimator_ff: not using taps
bandpass_fir_fft_cc: filter initialized, low_cut = -0.32, high_cut = 0.32
[openwebrx-httpd:ws,0] command: SET audio_rate=48000
[openwebrx-httpd:ws] invalid parameter
[openwebrx-httpd:ws,0] command: SET mod=nfm low_cut=-4000 high_cut=4000 offset_freq=0
bandpass_fir_fft_cc: filter initialized, low_cut = -0.32, high_cut = 0.32
shift_addition_cc: reinitialized to -0

ludovicmartin, 2017-03-17:
Same issue, same working solution for me on R-Pi3.

Here the full trace:
```
p4:/home/pi/openwebrx# uname -a
Linux p4.home.lepacha.eu 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
p4:/home/pi/openwebrx# export LD_PRELOAD=libSegFault.so
p4:/home/pi/openwebrx# ./openwebrx.py 

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] nmux_bufsize = 61440, nmux_bufcnt = 814
[openwebrx-main] Started rtl_thread: rtl_sdr -s 240000 -f 27100000 -p 0 -g 0 -| nmux --bufsize 61440 --bufcnt 814 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
nmux: pthread_create() done, clients now: 1
client 0x141b298: started!
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 2962 | csdr logaveragepower_cf -70 4096 9 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-main] Starting HTTP server.
[openwebrx-spectrum] Spectrum thread started.
nmux: pthread_create() done, clients now: 2
client 0x15834c0: started!
Connection to 127.0.0.1 4951 port [tcp/*] succeeded!
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 240000 S/s.
Tuned to 27100000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
client 0x141b298: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32192.168.0.11 - - [17/Mar/2017 11:43:37] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client 192.168.0.11#00e6ba58583114fb1fac27c31a91df1c :: client added. Clients now: 1
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /sdr.js HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /openwebrx.js HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /openwebrx.css HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /favicon.ico HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /sdr.js HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /openwebrx.js HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /openwebrx.css HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-top-photo.jpg HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-top-logo.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-ha5kfu-top-logo.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-avatar-background.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-avatar.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-rx-details-arrow-up.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-panel-status.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-rx-details-arrow.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-panel-log.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-panel-receiver.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-speaker.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-squelch-button.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-waterfall-auto.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-waterfall-default.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-zoom-in.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-zoom-out.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-zoom-out-total.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-background-cool-blue.png HTTP/1.1" 200 -
192.168.0.11 - - [17/Mar/2017 11:43:37] "GET /gfx/openwebrx-scale-background.png HTTP/1.1" 200 -
[... 23 lines cut ...]
[openwebrx-httpd:ws,0] command: SET audio_rate=48000
[openwebrx-httpd:ws] invalid parameter
[openwebrx-httpd:ws,0] command: SET mod=ssb low_cut=300 high_cut=3000 offset_freq=0
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_1987207496_shift | csdr fir_decimate_cc 19 0.007894375 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_1987207496_bpf 0.0253344944977 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_1987207496_squelch --outfifo /tmp/openwebrx_pipe_1987207496_smeter 5 1 | csdr realpart_cf | csdr fractional_decimator_ff 1.05258333333 | csdr agc_ff | csdr limit_ff | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
csdr: fifo control mode on
nmux: pthread_create() done, clients now: 3
client 0x1763790: started!
Connection to 127.0.0.1 4951 port [tcp/*] succeeded!
csdr: fifo control mode on
bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 157) + (input_size = 356) - 1
[... cut ...]
csdr: fifo control mode on
fractional_decimator_ff: use_prefilter = 0, num_poly_points = 12, transition_bw = 0.03, window = HAMMING
fractional_decimator_ff: not using taps
squelch_and_power_cc: initial squelch level is 0
bandpass_fir_fft_cc: filter initialized, low_cut = -0.316681, high_cut = 0.316681
client 0x17636c8: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32shift_addition_cc: reinitialized to -0
bandpass_fir_fft_cc: filter initialized, low_cut = 0.0237511, high_cut = 0.237511
```

dc7jzb, 2017-03-17:
Hello Andras,
It is the same like I reported on Google Groups.
On my Raspi 3 it works too, after starting it with your commands.
On the Raspi i didn't see more output after that, but it works.
On the NanoPI, where it doesn't work, I have more Output in the Terminal Session.

Thats why I was a little bit confused after it. ;)

dc7jzb, 2017-03-17:
System is NanoPi Neo with Ubuntu Core 16.04 LTS with Kernel 4.10.3

root@OpenwebRx70:~# export LD_PRELOAD=libSegFault.so
root@OpenwebRx70:~# python2 openwebrx.py
OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] nmux_bufsize = 258048, nmux_bufcnt = 194
[openwebrx-main] Started rtl_thread: rtl_sdr -s 1024000 -f 439000000 -p 36 -g 7 -| nmux --bufsize 258048 --bufcnt 194 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
Found 1 device(s):
[openwebrx-main] I/Q server started.
nmux: pthread_create() done, clients now: 1
client 0x290f8: started!
[openwebrx-main] Starting watchdog threads.
  0:  Realtek, DVB-T Dongle, SN: 00000991

[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 341333 | csdr logpower_cf -70 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-main] Starting sdr.hu update thread...
Using device 0: Dexatek DK DVB-T Dongle (Logilink VG0002A)
[openwebrx-spectrum] Spectrum thread started.
 [openwebrx-main] Starting HTTP server.
nmux: pthread_create() done, clients now: 2
client 0x291e0: started!
Connection to 127.0.0.1 4951 port [tcp/*] succeeded!
Found Fitipower FC0013 tuner
[openwebrx-sdrhu] Update succeeded! (Your receiver is soon getting listed on sdr.hu!)
Sampling at 1024000 S/s.
Tuned to 439000000 Hz.
Tuner gain set to 7.00 dB.
Tuner error set to 36 ppm.
Reading samples in async mode...
client 0x290f8: CS_THREAD_FINISHED, client_goto_source = 2, errno = 3291.66.123.36 - - [17/Mar/2017 19:58:47] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client 91.66.123.36#4e583f8183865aa371c9e5bbf940d308 :: client added. Clients now: 1
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /openwebrx.js HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /openwebrx.css HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /sdr.js HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-top-photo.jpg HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-ha5kfu-top-logo.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-top-logo.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-panel-status.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-avatar-background.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-panel-receiver.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-waterfall-default.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-rx-details-arrow-up.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-waterfall-auto.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-squelch-button.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-panel-log.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-zoom-in-total.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-avatar.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-zoom-out.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-zoom-out-total.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:48] "GET /gfx/openwebrx-rx-details-arrow.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:49] "GET /gfx/openwebrx-zoom-in.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:49] "GET /gfx/openwebrx-play-button.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:49] "GET /gfx/openwebrx-speaker.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:49] "GET /gfx/openwebrx-scale-background.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:49] "GET /gfx/openwebrx-background-cool-blue.png HTTP/1.1" 200 -
91.66.123.36 - - [17/Mar/2017 19:58:49] "GET /gfx/font-expletus-sans/ExpletusSans-Medium.ttf HTTP/1.1" 200 -
[openwebrx-httpd:ws,0] command: SET squelch_level=0.0000012589254117941661
[openwebrx-httpd:ws,0] command: SET output_rate=11025 action=start
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_3061538728_shift | csdr fir_decimate_cc 92 0.00163037109375 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_3061538728_bpf 0.0287511230907 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_3061538728_squelch --outfifo /tmp/openwebrx_pipe_3061538728_smeter 5 1 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.00952380952 | csdr deemphasis_nfm_ff 11025 | csdr fastagc_ff 1024 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
nmux: pthread_create() done, clients now: 2
[... 57 lines cut ...]
b6f35000-b6f36000 rw-p 00147000 b3:01 4687       /usr/lib/arm-linux-gnueabihf/libfftw3f.so.3.4.4
b6f36000-b6f3b000 r-xp 00000000 b3:01 2091       /lib/arm-linux-gnueabihf/librt-2.23.so
[... cut ...]
b6ffe000-b6fff000 r--p 00017000 b3:01 1968       /lib/arm-linux-gnueabihf/ld-2.23.so
b6fff000-b7000000 rw-p 00018000 b3:01 1968       /lib/arm-linux-gnueabihf/ld-2.23.so
beea5000-beec6000 rw-p 00000000 00:00 0          [stack]
bef80000-bef81000 r-xp 00000000 00:00 0          [sigpage]
bef81000-bef82000 r--p 00000000 00:00 0          [vvar]
bef82000-bef83000 r-xp 00000000 00:00 0          [vdso]
ffff0000-ffff1000 r-xp 00000000 00:00 0          [vectors]
Segmentation fault
bandpass_fir_fft_cc: filter initialized, low_cut = -0.359389, high_cut = 0.359389

m1geo, 2017-03-18:
Same issue here. Exactly as above. de M1GEO :)

ha7ilm, 2017-03-18:
I hope it is fixed by now. I have recently changed the algorithm behind `fractional_decimator_ff, but it seems like it corrupts memory. I now reverted to the old algorithm.

dc7jzb, 2017-03-18:
Updated on my NanoPi Neo, Works like a charm. 8)
Thx.

ludovicmartin, 2017-03-18:
Perfect for me too!
Thx

tikiarya, 2017-03-27:
works like a charm, many thanks

closed 2017-03-27 by tikiarya

zJOHNz, 2017-04-24:
Hello, Andras.
My RPi3 stops with
client 0x7d3298: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32
and "Sorry, the receiver is inactive due to internal error."

zJOHNz, 2017-04-24:
login as: root
root@192.168.0.195's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Apr 24 14:14:48 2017 from 192.168.0.239
root@minibian:~# cd /home/sdr/openwebrx/
root@minibian:/home/sdr/openwebrx# ./openwebrx.py

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] nmux_bufsize = 65536, nmux_bufcnt = 763
[openwebrx-main] Started rtl_thread: rtl_sdr -s 250000 -f 145525000 -p 0 -g 5 -| nmux --bufsize 65536 --bufcnt 763 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
nmux: pthread_create() done, clients now: 1
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
client 0x7d3298: started!
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
 [openwebrx-main] Starting HTTP server.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 2777 | csdr logaveragepower_cf -70 4096 10 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-spectrum] Spectrum thread started.
/bin/sh: 1: nc: not found
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sampling at 250000 S/s.
Tuned to 145525000 Hz.
Tuner gain set to 3.70 dB.
Reading samples in async mode...
client 0x7d3298: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32client 0x7d3298: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32
[openwebrx-check_server] >>>>>>> ERROR: spectrum_thread dsp subprocess failed
192.168.0.239 - - [24/Apr/2017 14:19:43] "GET / HTTP/1.1" 302 -
192.168.0.239 - - [24/Apr/2017 14:19:43] "GET /inactive.html HTTP/1.1" 200 -
192.168.0.239 - - [24/Apr/2017 14:19:43] "GET /gfx/openwebrx-logo-big.png HTTP/1.1" 200 -

zJOHNz, 2017-04-24:
Fresh builded (today) from maser branch csdr and openwebrx

ha7ilm, 2017-04-24:
No, that is the actual error message: `/bin/sh: 1: nc: not found`

Please install netcat.

zJOHNz, 2017-04-24:
After install nc same errno=32

login as: root
root@192.168.0.195's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Apr 24 14:15:50 2017 from 192.168.0.239
root@minibian:~# cd /home/sdr/openwebrx/
root@minibian:/home/sdr/openwebrx# ./openwebrx.py

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE fil                                                                                                        e in the package
________________________________________________________________________________                                                                                                        _________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.p                                                                                                        y"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] nmux_bufsize = 65536, nmux_bufcnt = 763
[openwebrx-main] Started rtl_thread: rtl_sdr -s 250000 -f 145525000 -p 0 -g 5 -|                                                                                                         nmux --bufsize 65536 --bufcnt 763 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
[openwebrx-main] I/Q server started.
nmux: pthread_create() done, clients now: 1
client 0xde298: started!
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f |                                                                                                         csdr fft_cc 4096 2777 | csdr logaveragepower_cf -70 4096 10 | csdr fft_exchange                                                                                                        _sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-main] Starting HTTP server.
[openwebrx-spectrum] Spectrum thread started.
localhost [127.0.0.1] 4951 (?) open
nmux: pthread_create() done, clients now: 2
client 0x25e4c0: started!
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sampling at 250000 S/s.
Tuned to 145525000 Hz.
Tuner gain set to 3.70 dB.
Reading samples in async mode...
client 0xde298: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32

zJOHNz, 2017-04-24:
but Work!
Thank you, Andras!!!

antonjan, 2017-10-20:
Hi
How was this resolved ?

ha7ilm, 2017-10-20:
AFAIK this was not resolved, the command that I suggested (that contained `libSegFault.so`) was only for debugging. Ufortunately. when people tried this command, the segfault did not happen again (so that I don't know what causes it).