ha7ilm/openwebrx #68: Python 3 compatibility?

KM4YRI, 2017-03-14:
Are there any major roadblocks standing in the way of Python 3 compatibility?  

Would you consider taking a pull request if the work to support both 2 and 3 was minimal?  I'd volunteer to do the conversion but wanted to open a dialog before beginning.

ha7ilm, 2017-03-14:
> Are there any major roadblocks standing in the way of Python 3 compatibility?

I have not tried yet.

> Would you consider taking a pull request if the work to support both 2 and 3 was minimal? I'd volunteer to do the conversion but wanted to open a dialog before beginning.

Yes, it is a good idea, if it is possible to keep python 2.7 compatibility, too. Thank you very much for the offer.

KM4YRI, 2017-03-14:
I will look into this.  I have used Python ```six``` before, which is a compatibility layer that allows Python scripts to run in either.  

If you'd like to keep the issue open for a bit, I'll start looking through the code and see how soon I can start on the conversion.  I'll try to make the change soon, as time allows.

ha7ilm, 2017-03-14:
> I have used Python six before

Will users have to install an additional library/tool then?

KM4YRI, 2017-03-14:
it's just a python dependency, e.g. ```import six```.  I looked through your code and surprisingly didn't see any other python dependencies in your code (not a bad thing or a good thing, just an observation).

Most linux and windows users will already have this library, as a high percentage of python packages and scripts use it.  If not, it's as easy as typing ```pip install six```.  I can't really think of any situation where the benefits don't out-weigh the disadvantages here.  I will say confidently that this will be a cakewalk compared to getting your other non-python dependencies satisfied.

ha7ilm, 2017-03-15:
Neil,

Thank you very much for your comments and your effort.
Right now I don't think it is a good idea anymore.

* From the users side, I do not see any added value. Other features (like #37) are asked for again and again every few months, so they are more important.
* From my side, it might increase the probability of bugs in a system that already works, and I'd need to support both version python 2 and 3, which is definitely more work. I think if I will support python 3, I will support it as the only platform (and drop version 2). It seems like there will be a point when I'll have to do that change in the future, but it is not a good idea (based on my plans and available time) to do it now.

KM4YRI, 2017-03-15:
From your comment, I'm assuming you've never tried to write a program that supports Python 2 and 3, which is nothing more than a finite list of syntactical differences that have been abstracted into a well-tested library (six). It's stuff like print foo``` instead of ```print(foo)``` and re-defining functions that BFDL Guido moved on a whim.  That's it.  They don't behave differently in any significant way.  And I'm also guessing you've not had much experience with unit testing, CI platforms, or other tools that are used in modern software development.  At least you're using Github.  Good job there.  I think you're afraid of something that (A) is being done daily by a high percentage of other applications and libraries using the language, and (B) is probably the most benign and easiest-to-test change that would ever happen in your repo, and is probably the easiest thing to unittest.  Things that people like me would have or could have added.

But that's OK, a hard-fork it is then.  

Good day sir.

closed 2017-03-15 by KM4YRI