ha7ilm/openwebrx #73: try with arora-browser

szg0000, 2017-03-26:
Arora browser is a cross platform QtWebKit based desktop web browser. I tried openwebrx with arora browser, but with no success. Can you tell me what is missing or what is wrong with arora browser to run openwebrx?

Please take a look to support arora browser.
https://github.com/icefox/arora

ha7ilm, 2017-03-26:
Hi,

Only the latest version of Google Chrome and Firefox are supported at the moment. 
If Aurora browser has a developer console (like Chrome has if you press Ctrl+Shift+I) then you can check if there are any Javascript errors.

Cheers,

Andras

szg0000, 2017-03-26:
Here is a screenshot about the arora console output.
[image]

Can you tell me from the logs, what is the problem?
Thanks,
Gabor

ha7ilm, 2017-03-26:
Gabor,

Thank you for the screenshot.

I think the Arora browser lacks some HTML5 features including the [`AudioBuffer` interface](https://developer.mozilla.org/en-US/docs/Web/API/AudioBuffer) that OpenWebRX currently relies upon. 

Andras