ha7ilm/openwebrx #74: Request for additional support for RTL-SDR and SDRplay

tikiarya, 2017-03-27:
Hello
I could not find a forum where i could request for features.

Friends, i am proposing for a paypal donation based campaign to request Andras to add few features for RTL-SDR and i need your suggestion. There are few requests mentioned in http://blog.sdr.hu/support but i am not sure these are directed toward RTL-SDR and SDRplay.

The proposed request.
1. To add more control in browser like KiwiSDR. For example to control /change frequency,gain,sampling rate 
2. device string , to support direct sampling so that HF can be received in RTL-SDR via direct sampling

Please suggest/comment as appropriate.
Many Thanks
Priyasloka

Luceoria, 2017-03-27:
I believe direct sampling can be set by editing start_rtl_command in config_webrx.py.

pumatrax, 2017-03-27:
You need to use a different rtl-SDR build to use I or Q branch. I believe keenerd has a fork on github for you. You just add a -D x where x is 0,1,2

tikiarya, 2017-03-27:
thanks Pumatrax , i will try that.

In GQRX and GNU RADIO block or even SDR Sharp software we have option to choose Q channel. I believe those must be different implementation.

from config_webrx.py
------------------------------------------------------------------------------------
start_rtl_command="rtl_sdr -s {samp_rate} -f {center_freq} -p {ppm} -g {rf_gain} -".format(rf_gain=rf_gain, center_freq=center_freq, samp_rate=samp_rate, ppm=ppm)
format_conversion="csdr convert_u8_f"
-----------------------------------------------------------------------------------
I was trying to see the option associated rtl_sdr executable 

rtl_sdr -?
rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:   -f frequency_to_tune_to [Hz]
        [-s samplerate (default: 2048000 Hz)]
        [-d device_index (default: 0)]
        [-g gain (default: 0 for auto)]
        [-p ppm_error (default: 0)]
        [-b output_block_size (default: 16 * 16384)]
        [-n number of samples to read (default: 0, infinite)]
        [-S force sync output (default: async)]
        filename (a '-' dumps samples to stdout)

ha7ilm, 2017-03-27:
Hi Priyasloka,

Regarding your second question, I hope this helps: 

https://github.com/simonyiszk/openwebrx/wiki/Using-RTLSDR-in-direct-sampling-mode-with-OpenWebRX

The first one is already being worked at (however, I am still doing the university thus progress very slowly with new features). 

Cheers,

Andras

bennybubble, 2017-05-23:
Hello Andras,

I ran into this piece of art yesterday, installed everything today (lubuntu, rtlsdr) and it worked the first time.
Very impressive !!
I second the plan to (remotely) set a frequency and gain!

Keep up the good work!

-ben

djbr1, 2017-05-28:
Hello Andras,
Just installed openwebrx with sdrplay, works like charm.  Please go ahead with option to set a frequency.
thx
HB9TXB