ha7ilm/openwebrx #75: Disable listing sdr on sdr.hu?

Hiermur, 2017-04-06:
Hello,

i have registered my sdr and i have received the key. All is ok.

When i unwanted listing at sdr.hu, i set sdrhu_public_listing to False.

I also removed the sdrhu_key.

My sdr is still listet.

How can disable temporary listing my sdr?

ha7ilm, 2017-04-06:
Listing will be automatically removed once your receiver is unavailable at the same URL for an hour. 

Contact me if it is still listed after that.

Hiermur, 2017-04-06:
Ok. Thats the reason my SDR is permanent listet and ignoring my settings.

While testing the SDR, it would be whishfull to set the SDR temporary disable from public listening.

73

ha7ilm, 2017-04-06:
If you change the port number in the config, it will be unlisted in
approximately 5 minutes.

Hiermur, 2017-04-07:
Ok. Thank you.

closed 2017-04-07 by Hiermur