ha7ilm/openwebrx #76: airspy_rx nmux: end input stream exitting.

zanco, 2017-04-18:
Hi, I try to change my openwebrx setup to one rtl-sdr and one arispy mini.
I have completely removed the openwebrx and csdr and did start from scratch with the latest openwebrx and csdr. I allready had SoapySDR and the airspy part installed.

The rtl-sdr is running, on port 4952 and works.

the Airspy is supposed to run on port 4951 but I get this message:

`[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] nmux_bufsize = 15003648, nmux_bufcnt = 4
[openwebrx-main] Started rtl_thread: airspy_rx -f403.2 -r /dev/stdout -a2 -g16 -b0| nmux --bufsize 15003648 --bufcnt 4 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
nmux: (main thread/for) end input stream, exiting.
`

If I start airspy_rx -f 403.2 -w it writes an IQ file on the given frequency so that seems to work.

What else could I try ?

Thanks,
Ben

ha7ilm, 2017-04-18:
Ben, 

Could you check if the following works?

    airspy_rx -f403.2 -r /dev/stdout -a2 -g16 -b0 | csdr through > /dev/null

`nmux` should not quit unless the receiver command exits.

Andras

zanco, 2017-04-18:
Hi Andras,

that command exits with no output at all.

Thanks,
Ben

ha7ilm, 2017-04-18:
> that command exits with no output at all.

In that case, it is not `nmux`, simply the receiver command does not work. You could either use rx_tools:

https://github.com/simonyiszk/openwebrx/wiki/Using-SoapySDR-via-rx_tools-with-OpenWebRX

...or try to figure out the command line for `airspy_rx` that writes the I/Q samples to the standard output.

zanco, 2017-04-19:
Quick update:
Hi, rx_tools gave the same result.  However, after adding -d 0 to the line where I start the airspy that does work. Later more, I am @work now.

zanco, 2017-04-23:
Well, I must say that I am impressed....

airspy_rx **-d 0** -f403.2 -r /dev/stdout -a2 -g16 -b0 | csdr through > /dev/null also does work. So both airspy_rx and rx_tools require the pointing to the device, even though there is only one device detected. (while there are 2 other RTL dongles connected).

Keep in mind that other software for Airspy counts the gain in values between 1 and 21 but for these plugins the gain is in dB ! So 21 for the airspy is not the highest gain. I am using 40 (dB) now and it works nicely.

On an i3 with Ubuntu 16.04 and quite high FFT resolution the Airspy Mini runs on 6 MHz bandwidth with only 28 percent processor load. I am happy with that !

Regards,
Ben

closed 2017-04-23 by zanco

ha7ilm, 2017-04-23:
Thanks for the info, I have added the `-d0` to the article in the Wiki.