ha7ilm/openwebrx #77: Audio-Compressor?

Hiermur, 2017-04-28:
Hello,

how can I disable Audio-Compression?

I don't mean Audio-Stream-Compression.

It would be nice, FM-Signals  have a constant noise level.

73, Hiermur

ha7ilm, 2017-04-28:
Dear Hiermur, 

For your own server, you can disable it here, in the config:
https://github.com/simonyiszk/openwebrx/blob/master/config_webrx.py#L80

For publicly available receivers listed on sdr.hu, you cannot change this setting from the web interface.

VY 73!

Andras, HA7ILM

closed 2017-04-28 by ha7ilm

Hiermur, 2017-04-28:
Hi Andras,

sry, you missunderstood me.

I mean the audio hears compressed.

Regardless of the setting of the stream compression.

73, Hiermur

ha7ilm, 2017-04-28:
You can tweak the AGC settings in the code, in `plugins/dsp/csdr/plugin.py`, based on the README of `csdr`.