ha7ilm/openwebrx #78: Default squelch level

sp2ong, 2017-05-03:
I have use OpenWebRX as local with 2m band. How set default squelch level?? 
I would like when who anybody start use OpenWebRX have squelch closed instead open. I have try set default level in htdocs/index.wrx from -150 to -70 but dose not work.

zanco, 2017-05-13:
Hi. If I set the squelch to auto the OpenwebRX PC says:
[openwebrx-httpd:ws,0] command: SET squelch_level=3.162277660168379e-7

So, in plugins/dsp/csdr/plugin.py I edited the line 
		self.squelch_level = 0 to 3.162277660168379e-7

Now, when I start the OpenwebRX it is squelched by default.

Note1: set the squelch on an unused frequency
Note2: make a backup of the plugin.py before editing

Hope this helps ,
73 ' Ben - PE2BZ

JJi-588, 2017-05-13:
I can confirm this to be a working solution.
However, do bear in mind that setting the squelch level this way prevents you from setting the squelch level lower, when/if it is required.