ha7ilm/openwebrx #79: cannot find plugins/dsp/ ?

halbeshuhn, 2017-05-06:
Hi @all,

cannot start the openwebrx on a Raspberry 2 with Debian jessie. rtl_test is ok.

==========================================================
pi@raspberrypi:~ $ /home/pi/script/openwebrx/openwebrx.py

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
Traceback (most recent call last):
  File "/home/pi/script/openwebrx/openwebrx.py", line 691, in <module>
    main()
  File "/home/pi/script/openwebrx/openwebrx.py", line 132, in main
    import_all_plugins("plugins/dsp/")
  File "/home/pi/script/openwebrx/openwebrx.py", line 67, in import_all_plugins
    for subdir in os.listdir(directory):
OSError: [Errno 2] No such file or directory: 'plugins/dsp/'

============================================================

What is the Problem? Please help me!

Thanx
Sebastian

ha7ilm, 2017-05-06:
Hi Sebastian,

1) Try to clone the repository again. It seems like some files are missing.

2) If it does not solve it, could you show us the output of `ls -alR` inside the `openwebrx` directory?

Andras

halbeshuhn, 2017-05-06:
Hallo Andreas,

ich habe das Problem gefunden. Ich muss es mit "python openwebrx.py" starten. Jetzt geht es einwandfrei. Allerdings vermisse ich den Bandplan und eine direkte Eingabe der Frequenzen. Wie kann ich das sichtbar machen. Jetzt kann ich die Frequenz nur ändern mit einem Mausklick in den Wasserfall.

LG
Sebastian

halbeshuhn, 2017-05-06:
Hello, Andreas,

I have found the problem. I must begin it with "python openwebrx.py". Now it goes perfectly. Indeed, I miss the tape plan and a direct input of the frequency. How I can make this visible. Now I can change the frequency only with a mouse click in the waterfall.

LG
Sebastian

ha7ilm, 2017-05-07:
> direct input of the frequency. How I can make this visible.

Unfortunately you cannot, as currently it is not a feature of OpenWebRX.

closed 2017-05-07 by ha7ilm