ha7ilm/openwebrx #80: Dont Work ?!?!?

ameyst, 2017-05-17:
Hello

 it dont run only the promt 

```
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 0000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sampling at 250000 S/s.
Tuned to 438412500 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
Signal caught, exiting!
Short write, samples lost, exiting!

User cancel, exiting...
```
the webpage dont came up 
port is 8073 in the conf

the rtl test says

```
root@websdr:/home/websdr/openwebrx/openwebrx# rtl_test -p
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 0000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
Sampling at 2048000 S/s.
Reporting PPM error measurement every 10 seconds...
Press ^C after a few minutes.
Reading samples in async mode...
lost at least 32 bytes
real sample rate: 2048043 current PPM: 21 cumulative PPM: 21
real sample rate: 2048021 current PPM: 11 cumulative PPM: 16
real sample rate: 2048046 current PPM: 23 cumulative PPM: 18
```
RaspberryPi Model B+ V1.2

pls hlp

ha7ilm, 2017-05-17:
Hi,

Does this work?

    rtl_sdr -s 250000 - | csdr through > /dev/null

You should see this:

```bash
rtl_sdr -s 250000 - | csdr through > /dev/null
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sampling at 250000 S/s.
Tuned to 100000000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
through: 500226 bytes/s, buffer #128
through: 500081 bytes/s, buffer #256
through: 499992 bytes/s, buffer #384
through: 500030 bytes/s, buffer #512
```

Andras

ameyst, 2017-05-17:
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 0000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sampling at 250000 S/s.
Tuned to 100000000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
Signal caught, exiting!
Short write, samples lost, exiting!

User cancel, exiting...
Ungültiger Maschinenbefehl
root@websdr:/home/websdr/openwebrx/openwebrx#

ha7ilm, 2017-05-17:
This is a problem with `librtlsdr`, not OpenWebRX. 

Please try a different sampling rate (`samp_rate` in `config_webrx.py`), possibly 1024000, 2048000 or 2400000. These are known to work.

closed 2017-05-17 by ha7ilm