ha7ilm/openwebrx #84: Problem with R820T2 Stick

dc7jzb, 2017-06-13:
Hello Andras,

I have an Issue with an R820T2 Stick.
It runs fine under SDR# etc. but doesn't start up under OpenwebRX.

Is there an way to get it running ? I can't find any Failure.

`[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] nmux_bufsize = 512000, nmux_bufcnt = 98
[openwebrx-main] Started rtl_thread: rtl_sdr -s 2048000 -f 145000000 -p 0 -g 5 -| nmux --bufsize 512000 --bufcnt 98 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
Illegal instruction
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2048000 S/s.
Tuned to 145000000 Hz.
Tuner gain set to 3.70 dB.
Reading samples in async mode...
Signal caught, exiting!
Short write, samples lost, exiting!

User cancel, exiting...`

dc7jzb, 2017-06-13:
I've forgot:

It is running on an Raspberry Pi 3 with Raspbian Jessie.

73 de Mario, DC7JZB

Calimerorulez, 2017-07-02:
Hi!

Same error here on a Rpi 1...

I think nmux throws the error...

Calimerorulez, 2017-07-02:
Small update. I changed the Makefile for csdr.  The hardware on a Rpi1 is a BCM2835 (info from cat /proc/cpuinfo):

PARAMS_ARM = $(if $(call cpufeature,BCM2835,dummy-text),$(PARAMS_RASPI),$(PARAMS_NEON))

Now OpenWebRX starts, but errors out in the end:

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] nmux_bufsize = 65536, nmux_bufcnt = 763
[openwebrx-main] Started rtl_thread: rtl_sdr -s 250000 -f 145525000 -p 0 -g 5 -| nmux --bufsize 65536 --bufcnt 763 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
Found 1 device(s):
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
nmux: pthread_create() done, clients now: 1
client 0xd05298: started!
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 2777 | csdr logaveragepower_cf -70 4096 10 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-main] Starting HTTP server.
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
[openwebrx-spectrum] Spectrum thread started.
nmux: pthread_create() done, clients now: 2
client 0xe854c0: started!
Connection to 127.0.0.1 4951 port [tcp/*] succeeded!
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sampling at 250000 S/s.
Tuned to 145525000 Hz.
Tuner gain set to 3.70 dB.
Reading samples in async mode...
client 0xd05298: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32

ha7ilm, 2017-07-02:
That's not an error, that's normal:
```
client 0xd05298: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32
```
...but I should say something about that in the message.

krjones, 2017-07-30:
Did this get solved? I'm running a PI2 and getting the same error as OP. Would love to know the solution.

majurius, 2017-12-07:
`[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-main] nmux_bufsize = 602112, nmux_bufcnt = 84
[openwebrx-main] Started rtl_thread: rtl_sdr -s 2400000 -f 144250000 -p 0 -g 5 -| nmux --bufsize 602112 --bufcnt 84 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
Illegal instruction
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Sampling at 2400000 S/s.
Tuned to 144250000 Hz.
Tuner gain set to 3.70 dB.
Reading samples in async mode...
Signal caught, exiting!
Short write, samples lost, exiting!

User cancel, exiting...`

It looked like a nifty software.

jamesmstone, 2018-02-24:
+1 hate to be that guy, but any resolution ?

jamesmstone, 2018-02-24:
I think this issues is the same as https://github.com/simonyiszk/csdr/issues/28

arastel, 2018-02-28:
Hi @ha7ilm, 

I am also trying to run Openwebrx on RPi 3 and encountering `CS_THREAD_FINISHED, client_goto_source = 2, errno = 32` such as many others.
I ran over a lot of issues here but found no solution and have no further stacktrace when executing ` env SEGFAULT_SIGNALS="abrt segv" LD_PRELOAD=/lib/arm-linux-gnueabihf/libSegFau
lt.so ./openwebrx.py`. 

Is there any solution currently for this ?

ha7ilm, 2018-02-28:
@arastel Not yet.