ha7ilm/openwebrx #85: Raspberry Pi: No audio, non waterfall, pending websocket

aldoir, 2017-06-19:
Hi All,

I'm trying to setup OpenWebRX on a Raspberry pi 3 without success.

The same RPi runs another dongle with dump1090, so I believe rtl_sdr stuff is working

I'm able to compile and install csdr as well.

On the openwebrx side, I did edit the config_webrx.py, changed the server_hostname and added a -d 1 to start_rtl_command in order to match my dongle.

By the output, everything looks fine, but I can't get a waterfall or audio on the web interface.

Using Chrome DevTools, all requests are ok (status code 200), except for the WebSocket request, that keeps a 101 Status code and always Pending (0 bytes).

Here is the console output. Any help is appreciated.
`pi@raspberrypi:~/openwebrx $ ./openwebrx.py

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] nmux_bufsize = 65536, nmux_bufcnt = 763
[openwebrx-main] Started rtl_thread: rtl_sdr -s 250000 -f 145525000 -p 0 -g 5 -d 1 -| nmux --bufsize 65536 --bufcnt 763 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
Found 2 device(s):
nmux: listening on 127.0.0.1:4951
  0:  Realtek, RTL2832U, SN: 00001000
  1:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 1: Generic RTL2832U OEM
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
nmux: pthread_create() done, clients now: 1
client 0x3ce298: started!
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 2777 | csdr logaveragepower_cf -70 4096 10 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-main] Starting HTTP server.
[openwebrx-spectrum] Spectrum thread started.
nmux: pthread_create() done, clients now: 2
client 0x54e4c0: started!
Connection to 127.0.0.1 4951 port [tcp/*] succeeded!
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sampling at 250000 S/s.
Tuned to 145525000 Hz.
Tuner gain set to 3.70 dB.
Reading samples in async mode...
x.x.x.x - - [19/Jun/2017 08:34:16] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client x.x.x.x#60323f8b6aeedffe5ac260d4e81b3edb :: client added. Clients now: 1
[(http log removed)]
[openwebrx-httpd:ws,0] command: SET output_rate=11025 action=start
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_1986796040_shift | csdr fir_decimate_cc 22 0.0068178 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_1986796040_bpf 0.0281615770483 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_1986796040_squelch --outfifo /tmp/openwebrx_pipe_1986796040_smeter 5 1 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr old_fractional_decimator_ff 1.03065759637 | csdr deemphasis_nfm_ff 11025 | csdr fastagc_ff 1024 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
nmux: pthread_create() done, clients now: 3
csdr: fifo control mode on
Connection to 127.0.0.1 4951 port [tcp/*] succeeded!
csdr: fifo control mode on
csdr: fifo control mode on
shift_addition_cc: reinitialized to -0
bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 143) + (input_size = 370) - 1
(overlap_length = 142) = taps_length - 1
old_fractional_decimator_ff: window = HAMMING
old_fractional_decimator_ff: taps_length = 133
squelch_and_power_cc: initial squelch level is 0
fir_decimate_cc: taps_length = 587
taps_length = 587
padded_taps_length = 592
taps = 178018
taps = 178020
client 0x74e6c8: started!
bandpass_fir_fft_cc: filter initialized, low_cut = -0.35202, high_cut = 0.35202
`

ha7ilm, 2017-06-21:
Hi!

Could you tell us your accurate browser version?

Andras

aldoir, 2017-06-21:
Google Chrome 58.0.3029.110 (Win 10 64 bits). I can use other public servers with this browser, so I think the issue might be related with my setup/openwebrx version.

ha7ilm, 2017-06-21:
Does the following run?

    rtl_sdr -s 250000 -f 145525000 -p 0 -g 5 -d 1 - | csdr through > /dev/null

aldoir, 2017-06-21:
yep

pi@raspberrypi:~ $ rtl_sdr -s 250000 -f 145525000 -p 0 -g 5 -d 1 - | csdr through > /dev/null
Found 2 device(s):
  0:  Realtek, RTL2832U, SN: 00001000
  1:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 1: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sampling at 250000 S/s.
Tuned to 145525000 Hz.
Tuner gain set to 3.70 dB.
Reading samples in async mode...

aldoir, 2017-06-22:
This is the pending request on Google Chrome. Firefox behaves exactly the same way.

`General
Request URL:ws://172.16.2.40:8073/ws/762bd07bf08d7af3aedfa71c171bb552
Request Method:GET
Status Code:101 Switching Protocols
Response Headers
view source
Connection:Upgrade
CQ-CQ-de:HA5KFU
Sec-WebSocket-Accept:wi9HQYYD11OR1Adg8gw8Q1EHOck=
Upgrade:websocket
Request Headers
view source
Accept-Encoding:gzip, deflate, sdch
Accept-Language:pt-BR,pt;q=0.8,en-US;q=0.6,en;q=0.4
Cache-Control:no-cache
Connection:Upgrade
Cookie:prevActiveTab=%23btn-map; mapLat=-25.08xxx; mapLon=-47.548xx; mapZoom=7
Host:172.16.2.40:8073
Origin:http://172.16.2.40:8073
Pragma:no-cache
Sec-WebSocket-Extensions:permessage-deflate; client_max_window_bits
Sec-WebSocket-Key:78grI0sdXyo8gEUu6keGdQ==
Sec-WebSocket-Version:13
Upgrade:websocket
User-Agent:Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/58.0.3029.110 Safari/537.36`

ha7ilm, 2017-06-22:
If you wait a few seconds after running the command above, do you get any information about the data throughput? If you only get the headers that you posted then the `rtl_sdr` tool does not output any data.

aldoir, 2017-06-22:
No, it doesn't show anything else, only "Reading samples in async mode.." until I stop it.

If I run the same command using the other stick (orange ProStick) I can se the csdr output

Any advice? I can use this dongle fine with rtl_tcp

aldoir, 2017-06-22:
Nevermind, restarted the RPi and it started working :(

closed 2017-06-22 by aldoir