ha7ilm/openwebrx #86: SDR support ... LimeSDR-USB -- probably via SoapySDR

KC7NOA, 2017-06-19:
Looking for a support for LimeSDR-USB -- probably via SoapySDR.

Would be nice to have a web based SDR head to listen to my various SDR's

ha7ilm, 2017-06-20:
Hi,

I think it should be already possible via (SoapyLMS7)[https://github.com/myriadrf/LimeSuite/tree/master/SoapyLMS7], see also:

https://github.com/simonyiszk/openwebrx/wiki/Using-SoapySDR-via-rx_tools-with-OpenWebRX

I don't have a LimeSDR board, can someone confirm?

VegaDeftwing, 2017-07-04:
I've made some progress as related to ha7ilm's link above

`soapy_device_query="driver=lime" #can be a number or can correspond to a driver, e.g. soapy_device_query="driver=hackrf", see `rx_sdr --help`
start_rtl_command="rx_sdr -d {device_query} -F CF32 -s {samp_rate} -f {center_freq} -p {ppm} -g {rf_gain} -".format(device_query=soapy_device_query, rf_gain=rf_gain, center_freq=center_freq, samp_rate=samp_rate, ppm=ppm)
format_conversion=""`

seems to get the device to sucessfull make itself but then:

`OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] nmux_bufsize = 65536, nmux_bufcnt = 763
[openwebrx-main] Started rtl_thread: rx_sdr -d driver=lime -F CF32 -s 250000 -f 145525000 -p 0 -g 0 -| nmux --bufsize 65536 --bufcnt 763 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
nmux: pthread_create() done, clients now: 1
client 0x8040a0: started!
Traceback (most recent call last):
  File "openwebrx.py", line 691, in <module>
    main()
  File "openwebrx.py", line 211, in main
    httpd = MultiThreadHTTPServer(('', cfg.web_port), WebRXHandler)
  File "/usr/lib/python2.7/SocketServer.py", line 417, in __init__
    self.server_bind()
  File "/usr/lib/python2.7/BaseHTTPServer.py", line 108, in server_bind
    SocketServer.TCPServer.server_bind(self)
  File "/usr/lib/python2.7/SocketServer.py", line 431, in server_bind
    self.socket.bind(self.server_address)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 13] Permission denied
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr fft_cc 4096 2777 | csdr logaveragepower_cf -70 4096 10 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-spectrum] Spectrum thread started.
localhost [127.0.0.1] 4951 (pwgwims) open
nmux: pthread_create() done, clients now: 2
client 0x9c4580: started!
Found 1 device(s):
  0: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 3.0] 9060B00542519, media=USB 3.0, module=STREAM, name=LimeSDR-USB, serial=0009060B00542519, 

verbose_device_search(driver=lime)
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060B00542519'
[INFO] Estimated reference clock 30.7196 MHz
[INFO] Selected reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] LMS7002M cache /home/vega/.limesuite/LMS7002M_cache_values.db
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Enable
[INFO] Rx Filter calibrated from cache
[INFO] Tx Filter calibrated from cache
[INFO] Rx Filter calibrated from cache
[INFO] Tx Filter calibrated from cache
Using device LimeSDR-USB: boardSerialNumber=0x9060b00542519 firmwareVersion=3 hardwareVersion=4 protocolVersion=1 
Found 6 antenna(s): NONE LNAH LNAL LNAW LB1 LB2 
Found 3 gain(s): LNA TIA PGA 
Found 2 frequencies: RF BB 
Found 2 sample rates: 100000 65000000 
Found 0 bandwidths: 
Using output format: CF32 (input format CS16)
Sampling at 250000 S/s.
SetFrequency using cache values vco:0, csw:181
Tuned to 145525000 Hz.
Tuner gain set to 0.00 dB.
Reading samples in sync mode...
############################################################
Rx calibration using RSSI INTERNAL ON BOARD loopback
Rx ch.A @ 145.525 MHz, BW: 30 MHz, RF input: LNAL, PGA: 12, LNA: 1, TIA: 1
Rx calibration: using cached values
Rx calibration values found in cache:
   | DC  | GAIN | PHASE
---+-----+------+------
I: |   0 | 2047 | 0
Q: |   0 | 2047 |
############################################################
nmux: (main thread/for) end input stream, exiting.`

Anyone have any ideas?

ha7ilm, 2017-07-04:
```
socket.error: [Errno 13] Permission denied
```
Try to close any other instance of OpenWebRX or any other process listening on TCP port 8073. 

You can see possible problematic processes by running: `netstat -tulpn | grep 8073`

VegaDeftwing, 2017-07-04:
Whoops, my bad, I killed the instance and re-ran it- here is the proper "error log":

```
OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-import] Found plugin: plugins.dsp.csdr.plugin
[openwebrx-main] nmux_bufsize = 65536, nmux_bufcnt = 763
[openwebrx-main] Started rtl_thread: rx_sdr -d driver=lime -F CF32 -s 250000 -f 145525000 -p 0 -g 0 -| nmux --bufsize 65536 --bufcnt 763 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
nmux: pthread_create() done, clients now: 1
client 0x19700a0: started!
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr fft_cc 4096 2777 | csdr logaveragepower_cf -70 4096 10 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-main] Starting HTTP server.
[openwebrx-spectrum] Spectrum thread started.
localhost [127.0.0.1] 4951 (pwgwims) open
nmux: pthread_create() done, clients now: 2
client 0x1b30580: started!
Found 1 device(s):
  0: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 3.0] 9060B00542519, media=USB 3.0, module=STREAM, name=LimeSDR-USB, serial=0009060B00542519, 

verbose_device_search(driver=lime)
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060B00542519'
[INFO] Estimated reference clock 30.7196 MHz
[INFO] Selected reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] LMS7002M cache /home/vega/.limesuite/LMS7002M_cache_values.db
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Enable
[INFO] Rx Filter calibrated from cache
[INFO] Tx Filter calibrated from cache
[INFO] Rx Filter calibrated from cache
[INFO] Tx Filter calibrated from cache
Using device LimeSDR-USB: boardSerialNumber=0x9060b00542519 firmwareVersion=3 hardwareVersion=4 protocolVersion=1 
Found 6 antenna(s): NONE LNAH LNAL LNAW LB1 LB2 
Found 3 gain(s): LNA TIA PGA 
Found 2 frequencies: RF BB 
Found 2 sample rates: 100000 65000000 
Found 0 bandwidths: 
Using output format: CF32 (input format CS16)
Sampling at 250000 S/s.
SetFrequency using cache values vco:0, csw:181
Tuned to 145525000 Hz.
Tuner gain set to 0.00 dB.
Reading samples in sync mode...
############################################################
Rx calibration using RSSI INTERNAL ON BOARD loopback
Rx ch.A @ 145.525 MHz, BW: 30 MHz, RF input: LNAL, PGA: 12, LNA: 1, TIA: 1
Rx calibration: using cached values
Rx calibration values found in cache:
   | DC  | GAIN | PHASE
---+-----+------+------
I: |   0 | 2047 | 0
Q: |   0 | 2047 |
############################################################
nmux: (main thread/for) end input stream, exiting.
[openwebrx-check_server] >>>>>>> ERROR: spectrum_thread dsp subprocess failed
127.0.0.1 - - [03/Jul/2017 21:41:42] "GET / HTTP/1.1" 302 -
127.0.0.1 - - [03/Jul/2017 21:41:42] "GET /inactive.html HTTP/1.1" 200 -
127.0.0.1 - - [03/Jul/2017 21:41:42] "GET /gfx/openwebrx-logo-big.png HTTP/1.1" 200 -
127.0.0.1 - - [03/Jul/2017 21:41:42] "GET /favicon.ico HTTP/1.1" 200 -

```

ha7ilm, 2017-07-04:
And does the `start_rtl_command` actually work? You can check it by:

    rx_sdr -d driver=lime -F CF32 -s 250000 -f 145525000 -p 0 -g 0 - | csdr through > /dev/null

VegaDeftwing, 2017-07-04:
It appears to:

```
Found 1 device(s):
  0: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 3.0] 9060B00542519, media=USB 3.0, module=STREAM, name=LimeSDR-USB, serial=0009060B00542519, 

verbose_device_search(driver=lime)
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060B00542519'
[INFO] Estimated reference clock 30.7196 MHz
[INFO] Selected reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] LMS7002M cache /home/vega/.limesuite/LMS7002M_cache_values.db
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Enable
[INFO] Rx Filter calibrated from cache
[INFO] Tx Filter calibrated from cache
[INFO] Rx Filter calibrated from cache
[INFO] Tx Filter calibrated from cache
Using device LimeSDR-USB: boardSerialNumber=0x9060b00542519 firmwareVersion=3 hardwareVersion=4 protocolVersion=1 
Found 6 antenna(s): NONE LNAH LNAL LNAW LB1 LB2 
Found 3 gain(s): LNA TIA PGA 
Found 2 frequencies: RF BB 
Found 2 sample rates: 100000 65000000 
Found 0 bandwidths: 
Using output format: CF32 (input format CS16)
Sampling at 250000 S/s.
SetFrequency using cache values vco:0, csw:181
Tuned to 145525000 Hz.
Tuner gain set to 0.00 dB.
Reading samples in sync mode...
############################################################
Rx calibration using RSSI INTERNAL ON BOARD loopback
Rx ch.A @ 145.525 MHz, BW: 30 MHz, RF input: LNAL, PGA: 12, LNA: 1, TIA: 1
Rx calibration: using cached values
Rx calibration values found in cache:
   | DC  | GAIN | PHASE
---+-----+------+------
I: |   0 | 2047 | 0
Q: |   0 | 2047 |
############################################################

```

ha7ilm, 2017-07-04:
You get `inactive.html` if OpenWebRX could not acquire any samples from the SDR hardware for some time.

ha7ilm, 2017-07-04:
Yes, but do you see the data throughput printed by `csdr through`? If not, there is no data output.
Check something like `cat /dev/urandom | csdr through > /dev/null` for reference.

VegaDeftwing, 2017-07-04:
No, it just exits immediately after printing the output I posted above.

ha7ilm, 2017-07-04:
OK, then you should try to make that command output the I/Q data somehow to make it work with OpenWebRX.

VegaDeftwing, 2017-07-04:
Alright. I'll look into it over the next few days, thanks for helping me find the source of the issue. SDRs are insane with how many deps and libs and whatnots go into each thing, so, it's taking some time to learn. Thank you so much. I'll report back in a few days if I get anywhere.

ha7ilm, 2017-07-04:
Strange, because it should work at that point I think. The LimeSDR libraries are reported to be loaded.

In case of no luck with `rx_sdr`, you could possibly try `SoapySDRUtil` from my fork, see here:

https://github.com/ha7ilm/SoapySDR/

https://github.com/pothosware/SoapySDR/issues/43

Alan-D-James, 2018-02-05:
I have read in several places that there are people who are having trouble getting their limeSDR USB to connect to their computers. I have spent three months now trying to get my limeSDR to work. In the last week, I finally managed to get a glimmer of life from the thing. Lime suite has reported that a westbridge device is available, but it won't connect. After a bit more searching I found that a jumper must be connected to the fx3 boot (some instructions on the Myriad pages say one should jumper J17, others J13) (on my device, it says fx3 boot adjacent to a header just beside the USB connector).
Now aside from the fact that the Myriad pages are a mess, with no proper indication of which header is either J17 or J13 (in both cases there are blocks of four headers, so which is which?) it would be  a service to all limeSDR owners if Myriad placed drawings and not photos showing where these points are on a card.
The final straw for me was finally discovering that the jumper that is intended to take the device out of boot mode was not even supplied when I received the device. I have since been to a couple of computer shops here in Aalborg, Denmark, but I have yet to find a jumper. Three months wasted because Myriad didn't install the jumper before delivery. (I haven't worked on the unit continuously because I have regularly given up in frustration).
Myriad would also do everyone a favour if they collected the various software packages into one place, instead of forcing people to discover that Cypress USB sofware is needed etc., rather than relying on the good offices of many well meaning enthusiasts.

KC7NOA, 2018-05-15:
Mine worked straight out of the box

Gymnae, 2018-09-26:
rx_tools seem to be discontinued and are not supporting new devices like AirSpyHF, while Soapy does. IS your fork an alternaitve to rx_tools, @ha7ilm ?

ha7ilm, 2018-09-26:
Maybe. I have not maintained it since then. Maybe the current version of `SoapySDRUtil` is better then mine was at that time.

Gymnae, 2018-09-26:
Thanks for your quick answer. I don't think the current version can stream like you need it.