ha7ilm/openwebrx #88: New Version throws error / didn't work

dg8ygz, 2017-07-16:
Hi
i've got WebSDR on an VMWARE virtual box running for a while, without any big problems. It's an 64bit machine with debian wheezy. Had to cut a few CPU-Options (sse3/4), to get it running.

Downloaded the latest Version just today and tried to compile it.
CSDR throws a few red lines and errors, but this didn't matter before...
When i try to start the application i got this:

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
Traceback (most recent call last):
  File "./openwebrx.py", line 737, in <module>
    main()
  File "./openwebrx.py", line 157, in main
    while nmux_bufsize * nmux_bufcnt < cfg.nmux_memory * 1e6: nmux_bufcnt += 1
AttributeError: 'module' object has no attribute 'nmux_memory'

Any hints ?
I rolled back with openwebrx.pyto and csdr to the previous version, wich is running fine..

ha7ilm, 2017-07-16:
I think you're getting the error because you're trying to use the same config file for the new version that you used with the old version. There are new options in the config file for the new features, so you need to use the latest from that, too. You can use the `diff` tool to see the differences, or also possibly `git diff`.

dg8ygz, 2017-07-16:
ahh thanks, i had a quick look in the new config-file and it said 2015 in its head. I thought mostly were the same ;-)    will give it new try and report. Thanks for the fast reply and good luck for the future

dg8ygz, 2017-07-16:
yepp that config-file did the trick, sorry for the disruption.
now i will do some tests =)

73s de dg8ygz

closed 2017-07-17 by dg8ygz