ha7ilm/openwebrx #89: Pass values to webrx.py

stevegpgh, 2017-07-29:
I wanted to pass the center frequency and bandwidth to OpenWebRx via a script that request those values from the owner at start up.  I am running Raspian on a Pi3 using a RTL-SDR and have implemented it using direct sampling mode.  But because of the 2.4 bandwidth limit, if I want to switch bands I need to close OpenWebRX, edit the config_webrx, save it then restart OpenWebRX.  I thought a script to handle an input of the desired frequency and relaying that to the config file would accomplish that.  My issue is on getting the values from the script and implementing them in the config file.  

I assume capturing the values in the shell script would be pretty easy:
read -p "Enter frequency in Mhz: " a
f=a*1000000

ha7ilm, 2017-08-10:
Thanks for the thoughts on that.

I am working on center frequency change capability, so that you will be
able to change that from the UI (and also other hardware parameters like
gain and PPM).

stevegpgh, 2017-08-10:
That's great news.  
If you are looking for additional items at any point (though I am sure you are busy enough as is):1.  From the UI, a way to key enter the center frequency (rather than dragging)2.  Enabling client use on iOS devices3.  Handling multiple RTL-SDRs in Q-Branch (for HF)4.  If max users = 1, then possibly allow the user to change the center frequency to any band (though that may disagree with your objective?)5.  An easier way to set the default range setting for the colors of the waterfall
I appreciate all the work you have put into this application!
Steve

DanielCraig, 2017-10-04:
How about an admin panel or login, to designate the one allowed to set center freq?

ha7ilm, 2017-10-04:
@DanielCraig That is being worked on.