ha7ilm/openwebrx #95: Frequency error

daniel5457h, 2017-09-24:
Hi,

With the last version, I've a frequency display error around 105kHz @ 27MHz in the waterfall and demodulator (mostly, everywhere). I've tested with an older version on the same computer and all is working ok. Played around the ppm adjustment and no luck.

Receiver is an RTL-SDR.

Thanks,
Daniel.

ha7ilm, 2017-10-25:
Was it solved?

daniel5457h, 2017-10-25:
Hi,

Yes, my mistake, I put the default frequency where the mixer lo frequency should be.

By the way, my receiver is not longer listed in sdr.hu... The keys and the flag are correctly set... Was this a change from your part on the sdr.hu policy or a bug/mistake that needs investigation?

Thanks,
Daniel.

ha7ilm, 2017-10-25:
> By the way, my receiver is not longer listed in sdr.hu... The keys and the flag are correctly set... Was this a change from your part on the sdr.hu policy or a bug/mistake that needs investigation?

No. Which receiver is that? What happens if you run `python sdrhu.py`?

daniel5457h, 2017-10-25:
I get a nice
[openwebrx-sdrhu] Update succeeded! (Your receiver is soon getting
listed on sdr.hu!)
but no listing.
The receiver is somewhat "EA3HRU blah blah", address http://z-ha-dum.dy
ndns.org:8080 It's accessible from the outside.
Best regards,

ha7ilm, 2017-10-25:
Yes, it looks okay. Did you modify the settings recently?

ha7ilm, 2017-10-25:
Your avatar is too large. Make it under 10.000 bytes.

closed 2017-10-25 by ha7ilm