ha7ilm/openwebrx #98: Update center_frequency while running

krippendorf, 2017-10-06:
Hi, I'm  integrating FlexRadio Dax IQ output (https://twitter.com/HB9FXQ/status/916367406456549377) Is there a way to update the center_frequency value while running, e.g. reading from an ENV variable?

I get raw IQ from the radio while operating, and can query/notify the current center Freq. of the stream. Now I'd like to sync the OpenWebRx displayed scale with the current TRX state. Any Idea?

ha7ilm, 2017-10-07:
>https://twitter.com/HB9FXQ/status/916367406456549377)

Cool!

Well, I'm working on being able to change the center frequency while running from the web UI, but that's the reverse direction compared to what you want. This is not implemented in OpenWebRX yet, but it was possible to do via a named pipe, I think. An approach is:

* adding named pipe read here: https://github.com/simonyiszk/openwebrx/blob/master/openwebrx.py#L477
* sending a WebSocket message to the client: `ws.send("MSG center_freq=100000000")` 
* the client should made update the scale on getting such a message if the watefall is already initialized: https://github.com/simonyiszk/openwebrx/blob/master/htdocs/openwebrx.js#L1209 (call `mkscale`)

This solution will work only for a single client, because multiple client threads won't work with the same named pipe. For multiple clients you need to broadcast to all of their threads.

krippendorf, 2017-10-07:
Thx for the pointers. I'll keep that in mind + for now will put openwebrx into a docker container and restart the container when the Flexradio panadapter changes frequency

mo8Zomo0, 2018-01-29:
Hi!

Yes, it would be great if an option could be added "allow frequency to be changed by web interface" and have the frequency field in the "Receiver" box editable. 

For me a pure single user / client mode is fine, but I do not see a reason why a multiple client (like two computer for the same user) or even that everybody can set (or maybe with some authentication) the frequency.

Lonecrowe, 2018-07-05:
Wouldn't be possible due to hardware limitations for multiple people to change center freq.  But that being said having a single user online it would be nice.  

I was hoping for a proposed "admin" mode (different port?) where there were some live functions that the admin could change in real time.  So if you are there others are listening in (friends) and they request a change you can change the center freq.  

I wouldn't want to get into ACL's but maybe the first person who connects gets control UNLESS the admin is online.  If the first person logs they lose control the center resets.  Next in line has control.  

I support this option.  I mean we can just shut it down edit the config and restart it but would be so much nicer to be able to move it.

ofadam, 2019-01-23:
Any update on moving the center_frequency from the web UI? I'd love to have that feature!