ha7ilm/openwebrx #99: Use with real radios?

kathiedart, 2017-10-24:
Can somehow I connect this to a radio which uses hamlib/rigctl/soundcard access and provide speaking?

If not, what would be the steps?

Cheers

ha7ilm, 2017-10-25:
What do you want to achieve exactly with that? In theory, it is possible to use a traditional SSB receiver with 3 kHz bandwidth as an input signal source for OpenWebRX, but that doesn't make much sense.

ha7ilm, 2017-10-25:
Or if you mean that, an SDR receiver hardware(+software) is a real radio. The list of supported hardware is here: 

https://github.com/simonyiszk/openwebrx/wiki

Anyway, I think I don't understand the question yet.

kathiedart, 2017-10-25:
Yes, what I mean is the 3 kHz bandwidth audio source, which is tunable via software.

ha7ilm, 2017-10-25:
You could start out from this: https://github.com/simonyiszk/openwebrx/wiki/Using-OpenWebRX-with-the-audio-card-as-a-VLF-receiver

The web UI currently doesn't support changing the center frequency for SDR hardware either, so you have to implement yourself if you want such functionality. 

Anyway, OpenWebRX was designed to let multiple people access wide bandwidth receivers, so there are probabily other better solutions for remote controlling a typical rig. You are basically looking for something like this, but with audio streaming, I think: https://github.com/ntoulasd/Webrig

kathiedart, 2017-10-25:
Okay - thanks for the info, I haven't thought other interfaces look very nice before - I'll see if that works well enough for me.

Also changed the title for anyone who's searching if they have the same query.

Cheers

closed 2017-10-25 by kathiedart