ha7ilm/openwebrx #101: OpenWebRX "make" error on  CPU A64 Quad-core Cortex-A53 64bit

R8QAAE, 2017-11-16:
Hi ALL !

I try to run the project on Orange Pi Win , CPU A64 Quad-core Cortex-A53 64bit, and I get a compilation error: 

orangepi@Orangepi:~/csdr$ make
NOTE: you may have to manually edit Makefile to optimize for your CPU (especially if you compile on ARM, please edit PARAMS_NEON).
Auto-detected optimization parameters: -mfloat-abi=hard -march=armv7-a -mtune=cortex-a8 -mfpu=neon -mvectorize-with-neon-quad -funsafe-math-optimizations -Wformat=0 -DNEON_OPTS

rm -f dumpvect*.vect
gcc -std=gnu99 -O3 -ffast-math -fdump-tree-vect-details -dumpbase dumpvect -mfloat-abi=hard -march=armv7-a -mtune=cortex-a8 -mfpu=neon -mvectorize-with-neon-quad -funsafe-math-optimizations -Wformat=0 -DNEON_OPTS fft_fftw.c libcsdr_wrapper.c  -g -lm -lrt -lfftw3f -DUSE_FFTW -DLIBCSDR_GPL -DUSE_IMA_ADPCM -Wno-unused-result -fpic -shared -Wl,-soname,libcsdr.so.0.15 -o libcsdr.so.0.15
gcc: error: unrecognized command line option '-mfloat-abi=hard'
gcc: error: unrecognized command line option '-mfpu=neon'
gcc: error: unrecognized command line option '-mvectorize-with-neon-quad'
Makefile:52: recipe for target 'libcsdr.so' failed
make: *** [libcsdr.so] Error 1

USING SYSTEMS 
Ubuntu 16.04.3 LTS aarch64, or ARMBIAN (same mistake):

Tell me please, how should I modify the MakeFile for this CPU ?
__
With greetings from Russia,  R8QAAE !

R8QAAE, 2017-11-18:
I found ))
For this processor, the directives are as follows:
-march = armv8-a + crc + crypto -mtune = cortex-a53

FeatherGrass2019, 2019-08-15:
Can you tell me what the directives for Raspberry Pi 4 are?the processor is Broadcom BCM2711, Quad core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz
thanks..